openthread/openthread #4918: Two Leader node upon one mesh when calling otBecomeLeader()

dophanngoc, 2020-05-01:
I am deploying openthread network and want to set node permanently as Leader,
I increased Leaderweight of the node and called otBecomeLeader() to force set it become Leader,
It takes the Leader role back from the second FTD and the second FTD downgrade to Router when deploying with 2 FTD devices properly but 
The problem is when i set the network with more than 2 FTD devices, 
the second FTD doesn't downgrade to Router and what i see is the mesh keep 2 Leader existing.

I think i can force set it downgrade by calling otBecomeRouter() but i need to use UDP controlling message.
Is there simply way to fix this issues?

:nRF52840

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2020-05-01:
The Thread Specification does not currently support this use case.

> Rule 1: A non-singleton Thread Network Partition always has higher priority than a singleton Thread Network Partition.

Indeed, OpenThread is currently conformant to the specification.

https://github.com/openthread/openthread/blob/7b0cb984455731bbaf2eb51188baf5d225a2ee41/src/core/thread/mle_router.cpp#L1111-L1135

Supporting your use case would require a Thread specification change. Can you provide more information about your use case and motivation for forcing a device as Leader?

dophanngoc, 2020-05-07:
Thank you for the answering,

I am pretty sure my use case has a little difference from Thread specification.
The device i want to set as permanent Leader always needs Commissioner permission,
but according to Thread Specification, only Leader has the permission.
That is why i want to force that device to be a Leader.
Hope you can give me some advice with this use case.

jwhui, 2020-05-07:
@dophanngoc , OpenThread supports an on-mesh commissione in the Leader, Router, or End Device role. Can you point to where in the Thread Specification that the Commissioner must also be a Thread Leader?

dophanngoc, 2020-05-07:
I apologize for my misunderstanding about this point, 
as you told above indeed OpenThread supports an on-mesh commissioner in all Device role.
But in my implementation, as **securities reason** that i was able to satisfy,
 I need to static Leader device.

jwhui, 2020-05-07:
@dophanngoc , as previously mentioned, the Thread Specification does not currently support forcing a single device to be the Leader. The Thread Specification prioritizes partitions that have more than one router-capable device over partitions that only have a single router-capable device. For that reason, it is generally not possible for a device to assume the Leader role at all times.

dophanngoc, 2020-05-12:
@jwhui ok, Thank you, i understand your mean. Actually i don't expect keep node to be Leader permanently can handle my issues. 
↓
Suppose that i have much more FTD devices on-mesh, and i want to keep a node as Recipient, in other word all CoAP messages always to be sent to the Node. But what i know is when i enable DHCP on-mesh, all the FTD device has same RLOC16 type (no way to distinguish between them).
So how can i do that?
Moreover, even for some reason the Node breakdown or shutdown from the mesh, when it re-merge to the partition i want it must be a Recipient of the mesh.

jwhui, 2020-05-13:
@dophanngoc , would using a stable IPv6 address work for your use case? Can you use the Mesh-Local EID (ML-EID) (see https://openthread.io/guides/thread-primer/ipv6-addressing)?

dophanngoc, 2020-05-29:
@jwhui Hi Jonathan,
sorry for replying lately, i restart with this issue with a hypothesis like following:
I have a NCP node (M) started alreadly and has commissioner right, every node want to join to the mesh need to join to M, then sending COAP messages to M every certain time.
* Do i need to use ML-EID in this case?
* If i want to get ML-EID of M, how to get it?
* All node need to send COAP messages to M,  what  i need is attach ML-EID of M to socket of COAP frame? or another way.?

jwhui, 2020-06-01:
> @jwhui Hi Jonathan,
> sorry for replying lately, i restart with this issue with a hypothesis like following:
> I have a NCP node (M) started alreadly and has commissioner right, every node want to join [...]

Using the ML-EID is not required in this case. However, given that your communication is local to the mesh (i.e. does not require communicating with IPv6 end points external to the Thread network) and you prefer a stable IPv6 address, the ML-EID seems well-suited.

> * If i want to get ML-EID of M, how to get it?

Discovering the location of services is a common problem in an IP-based network. You could use standard protocols like DNS. You could also implement your own proprietary mechanism (e.g. multicast a message from M to allow other nodes to discover it).

> * All node need to send COAP messages to M,  what  i need is attach ML-EID of M to socket of COAP frame? or another way.?

To receive CoAP messages on M, you would need to bind a CoAP endpoint to a UDP port on M. You can bind to either the ML-EID address specifically or the IPv6 Unspecified Address if you don't want to be bound to a given address.

dophanngoc, 2020-06-05:
@jwhui 

> Discovering the location of services is a common problem in an IP-based network. You could use standard protocols like DNS. You could also implement your own proprietary mechanism (e.g. multicast a message from M to allow other nodes to discover it).
Thank you so much, I would like to deploy DNS because it is best option in my user case but building a DNS server seems hard implementation. :) 

> To receive CoAP messages on M, you would need to bind a CoAP endpoint to a UDP port on M. You can bind to either the ML-EID address specifically or the IPv6 Unspecified Address if you don't want to be bound to a given address.

According to RFC 4291 (IP Version 6 Addressing Architecture), The address 0:0:0:0:0:0:0:0 is called the unspecified address. It means that i can bind a COAP endpoint to a UDP port on M by using 0:0:0:0:0:0:0:0.?

jwhui, 2020-06-05:
> According to RFC 4291 (IP Version 6 Addressing Architecture), The address 0:0:0:0:0:0:0:0 is called the unspecified address. It means that i can bind a COAP endpoint to a UDP port on M by using 0:0:0:0:0:0:0:0.?

Yes. See Section 1.2 of https://www.tldp.org/HOWTO/Linux+IPv6-HOWTO/ch03s01.html

dophanngoc, 2020-06-08:
@jwhui 
> Using the ML-EID is not required in this case. However, given that your communication is local to the mesh (i.e. does not require communicating with IPv6 end points external to the Thread network) and you prefer a stable IPv6 address, the ML-EID seems well-suited.

You said that 
> Using the ML-EID is not required in this case. 

So if i do not use ML-EID, what i should to use in this case.

jwhui, 2020-06-10:
> > Using the ML-EID is not required in this case.
> 
> So if i do not use ML-EID, what i should to use in this case.

In general, you can use any IPv6 address of appropriate scope. If you assigned other global IPv6 addresses using an on-mesh prefix, that would work as well. However, mesh-local addressing is convenient if you only care to communicate between devices in the same Thread network.

jwhui, 2020-06-30:
Closing stale issue.

closed 2020-06-30 by jwhui