openthread/openthread #4925: EFR32MG1P RCP-only Makefile Changes

gabekassel, 2020-05-04:
**Goal:** Support RCP on EFR32MG1P (256K flash/32K RAM)

**Status:** I've successfully built an RCP image that runs by making the following (somewhat hacky) changes:

1. Duplicated efr32mg12 platform example in all pertinent places
2. Copied then edited efr32mg12.ld to efr32mg1.ld, changed RAM/FLASH quantities to match part - note that diff'ing the .ld from gecko sdk shows no other changes beyond the nvm3 changes recently made
3. (Hacky bit) Compilation was failing with RAM overrun error due to building mtd in examples/apps/ncp/ even if I removed --enable-mtd from the efr32mg1 (copied from mg12) makefile. I simply deleted all references to mtd from examples/apps/ncp/Makefile.am - I'm sure there's a more correct way to do this

**Request/Desire:** Community support to correct add efr32mg1p as a supported platform for RCP-only given footprint limitations

**Additional context** Somewhat messy set of files tracked in a branch/fork [here](https://github.com/openthread/openthread/compare/master...gabekassel:efr32mg1p?expand=1)

syabyr, 2020-05-23:
I have just ported to  efr32mg1p  ikea tradfri module:
https://github.com/syabyr/openthread

gabekassel, 2020-05-23:
> I have just ported to  efr32mg1p  ikea tradfri module:
> 
> https://github.com/syabyr/openthread
> 
> 

@syabyr do you know if this works with/expects gecko bootloader? I’ve gotten mg1p to work on BRD4151 but not custom design with bootloader (NCP). I’ll check out your diff later this weekend. 

Thanks.

gabekassel, 2020-05-24:
@syabyr I looked through your changes and they're very close to what I did. One question: why do you create the `EFR32MG1_TRADFRI/config` folder? I don't see those files referenced anywhere, but have suspected we need to include retargetserial somewhere.

Have you been able to test/exercise this yet?

Thanks

gabekassel, 2020-05-26:
> @syabyr I looked through your changes and they're very close to what I did. One question: why do you create the `EFR32MG1_TRADFRI/config` folder? I don't see those files referenced anywhere, but [...]

Seems that should get copied to silabs folder under hardware/kit. 

 Curious if Tradfri uses gecko bootloader.

syabyr, 2020-05-27:
I`m new to openthread and have bing testing them these days

gabekassel, 2020-05-28:
@syabyr I tested your branch and compared to mine both on BRD4151A and a custom design. In both cases it works as a baseline. Would you be willing to open a pull request for these changes?

It would be great to get efr32mg1p into upstream so Silabs engineers and other maintainers bring it along as changes are made to the similar mg12/mg13 parts.

Thanks

syabyr, 2020-05-29:
tradfri module have a 256KB spi flash on board,i`m considering using it as nvm3 storage to sotre none volatile data

gabekassel, 2020-05-30:
nvm3 works for me. Can you post your full formation steps? Did you dataset init new first?

syabyr, 2020-05-30:
I fallowed this guide:
https://codelabs.developers.google.com/codelabs/openthread-hardware/

version
OPENTHREAD/20191113-00810-g6686a2f0-dirty; EFR32; May 30 2020 10:58:46
Done
> dataset init new
Done
> dataset
Active Timestamp: 1
Channel: 16
Channel Mask: 07fff800
Ext PAN ID: b0caece132c2b5fc
Mesh Local Prefix: fd62:1d03:85e8:698c/64
Master Key: 950e100d6f48302ad030651f9cc280e7
Network Name: OpenThread-b9dc
PAN ID: 0xb9dc
PSKc: 676be00f794f465eb298564ff0aab6dd
Security Policy: 0, onrcb
Done
> dataset commit active
Error 1: Failed
>

jwhui, 2020-07-09:
Resolved by #5191

closed 2020-07-09 by jwhui