openthread/openthread #4948: Modifying of Platform Abstraction Layer APIs

testter2, 2020-05-11:
Hello everybody,

In this link https://openthread.io/guides/porting/implement-platform-abstraction-layer-apis 
Random should be replaced by Entropy as Random has been moved from platform abstraction to become an open thread module. 

Thanks,

jwhui, 2020-05-11:
@testter2 , thanks for pointing this out. We'll work to update the docs. Thanks.

Vyrastas, 2020-05-12:
Looks like the Flash one needs to be updated too, I assume we should use /include/openthread/platform/flash.h now?

jwhui, 2020-05-12:
> Looks like the Flash one needs to be updated too, I assume we should use /include/openthread/platform/flash.h now?

@Vyrastas , platforms need only implement one of:

1. https://github.com/openthread/openthread/blob/master/include/openthread/platform/flash.h

2. https://github.com/openthread/openthread/blob/master/include/openthread/platform/settings.h

The choice is made with:
https://github.com/openthread/openthread/blob/b4a5df49e64c5f13333f5251619d71235a60be38/src/core/config/openthread-core-default-config.h#L344

testter2, 2020-05-13:
> Looks like the Flash one needs to be updated too, I assume we should use /include/openthread/platform/flash.h now?

I also think so as the current link in the website has the old path for flash.h file before moving it the platform.

Vyrastas, 2020-05-14:
Thanks, I'll get them updated.

Vyrastas, 2020-05-19:
This has been fixed.

closed 2020-05-19 by Vyrastas