openthread/openthread #4984: SubMac: on enable go from SLEEP state to SLEEP state

thcu-gp, 2020-05-19:
https://github.com/openthread/openthread/blob/6adfa966e0c9399061630d9c9d461a5c48d4eafd/src/core/mac/sub_mac.cpp#L144

according to the info in nclude/openthread/platform/radio.h
calling `Enable` should already put you in the `SLEEP` state, there seems to be no reason to
explicitly call the `Sleep` function to achieve that

abtink, 2020-05-19:
@thcu-gp , you are correct that this is technically redundant.

However, the explicit call to `Sleep()` helps ensure correct start/behavior even if the radio platform implementation does not follow the requirement from `radio.h` (does not start in sleep after `Enable()`). We expect that it would not add any complications.

Are you aware of any issues/problems with this call to `Sleep()`?

thcu-gp, 2020-05-19:
I saw the issue triggering an assert on a local test setup, when the call to `Sleep` returned an error code from the platform code (since the device was already sleeping)
Fixed it locally to be more robust.

Flagged it anyways.

abtink, 2020-05-19:
Thanks @thcu-gp .

I think if the radio is in sleep state already, the expected behavior is that calling `otPlatRadioSleep()` again should just return `OT_ERROR_NONE`. 

We would expect same with `otPlatRadioReceive` (if already in rx state). I know the OT code during operation would call `Receive()` when radio may already be in rx state (e.g., after a tx done, or before starting a new frame tx). I don't know if we have such a pattern for sleep (other than situation you mentioned after enable) but it may be used in future.

closed 2020-06-05 by jwhui