openthread/openthread #4995: Support for Fanstel BT840X

sjlongland, 2020-05-22:
Fanstel make a number of FCC-approved modules and dongles built on the Nordic nRF52840 system on chip.

There are a few differences in how the chip is wired up which makes stock OpenThread incompatible as-is, and 3 of the devices (the BT840X, BT840XE and USB840X) additionally incorporate an analogue front-end PA/LNA IC for increased range.

It would be great if OpenThread supported these.  We've been maintaining a fork of OpenThread which implements support for the BT840X and related modules, and so far have been using the USB840X dongles quite successfully in the field as NCPs with this firmware.

Our fork is at https://github.com/vrtsystems/openthread/tree/feature/fanstel-bt840x … the [README](https://github.com/vrtsystems/openthread/tree/feature/fanstel-bt840x/examples/platforms/nrf528xx/nrf52840#support-for-the-fanstel-bt840x-bt840xe-and-usb840x) for the nRF52840 has been updated with information on how to support these devices.

Changes are:
* allowing the RX and TX pins on standard nRF52840 builds to be moved around to avoid clashes with specific boards
* support for the SKY66112 front-end IC used in the BT840X family of modules, with configuration options for moving pin assignments around on other boards
* a `BOARD=xxxx` option in the `Makefile` so these options can be set to sane values for particular well-known board configurations
* aforementioned README update.

jwhui, 2020-05-26:
@LuDuda , any thoughts about adding support for Fanstel BT840X?

LuDuda, 2020-05-26:
Hello @sjlongland 

Thank you for bringing this up. This looks like a good idea. 

As for now we have decided to officially support nRF21540 Front-End-Module in OT only. We still would like to support nRF21540 by default, but it is OK to revert support for SKY.

In the nRF5x we provide production-ready support that is tested on our DKs. I would like to have a clear separation when adding support for the additional boards (which preferably only modifies some defines).

In case you would like to contribute this support, I can help with review. My recommendation would be to do it in two phases.

1. Add support for SKY66112 so that it is possible to use either nRF21540 or SKY66112 by selecting appropriate defines. Note that Radio Driver already supports both FEMs (`simple_gpio` and `three_pin_gpio` options). The main problem is that selection between them is done by selecting appropriate files, rather than selecting appropriate defines. And so it might be challenging to add dynamic configuration options without touching the `configure.ac` file. I'll check if we can improve it internally.

2. Add support for a custom board - modification of Makefile (as you did) with the proper defines. I can imagine an additional folder `boards` that would include board-specific sub-Makefiles with the configuration that overwrites the default one, or something similar.

One more time thank you for this, and spending time on good documentation!

sjlongland, 2020-05-26:
No problems there… I'll have a look at how the nRF21540 is done and see what can be done to weave the SKY66112 in with it.

A separate `boards/` directory sounds like a cleaner way to do things, I'll have to get some new test hardware (including maybe nRF21540 DK… I have two plain nRF52840 boards but will have to look around for the other) but should be able to try re-factoring what I have to accommodate both.

jwhui, 2020-07-15:
@sjlongland , any updates on this?

sjlongland, 2020-07-15:
Hi, been flat out with other projects at the moment, plus the nRF21540 was unavailable through my usual suppliers (I tried Mouser, RS and Digikey), so I'm leery about changes that might break the nRF21540 support.

sjlongland, 2020-08-21:
@LuDuda Hi, do you know a distributor that's selling the nRF21540 DK (PCA10112) boards?  I've looked around periodically over the last few months, and not seen any.  Google helpfully points me to sites in Denmark that sell the bare nRF21540 chip when I try looking.

Obviously, they exist somewhere, as other projects like the Zephyr project also support them.  Any clues as to where?

jwhui, 2020-11-04:
@sjlongland , any updates on this issue?

sjlongland, 2020-11-04:
@jwhui Still looking around for where to source the nRF21540 DKs… without those, it's hard to make changes without possibly breaking this platform accidentally.

sjlongland, 2020-11-09:
Okay, got a reply back… looks like these dev boards are expected to start appearing in distributor catalogues in January, so maybe early next year, I'll get started on this.

Thinking being, I'll order a nRF21540-DK, then get stuck into the source code to see if I can cleanly insert the support for the USB840X in there.  By that stage, I should then have both boards to test that both are operational.

jwhui, 2020-11-09:
> Okay, got a reply back… looks like these dev boards are expected to start appearing in distributor catalogues in January, so maybe early next year, I'll get started on this.
>
> Thinking being, I'll [...]

@sjlongland , sounds good. Thanks for the update!

Vijay-Karthick, 2021-11-16:
@sjlongland 
Did you make any progress on this? I'm looking at procuring BT840XE for RCP with otbr-agent. I've successfully used BT840F but that doesn't have the SKY based PA. I was curious if you have any updates, as your repository is not updated over a long time!

Kind regards,
Vijay

sjlongland, 2021-11-16:
On Mon, 15 Nov 2021 17:21:04 -0800
Vijay Karthick Baskar ***@***.***> wrote:

> @sjlongland 
> Did you make any progress on this? I'm looking at procuring BT840XE for RCP with otbr-agent. I've successfully used BT840F but that doesn't have the SKY based PA. I was curious if you have any updates, as your repository is not updated over a long time!

Sadly no, I was assured early last year that there would be available
stock for the Nordic board around, but to date, I've still not been
able to procure one.

If someone can recommend an Australian distributor for me to talk to,
I'll investigate further.
-- 
Stuart Longland (aka Redhatter, VK4MSL)

I haven't lost my mind...
  ...it's backed up on a tape somewhere.

Vijay-Karthick, 2021-11-16:
Thanks for quick reply, @sjlongland .

I just placed orders for BU840XE. I'll keep you posted on the progress.

marcorussi, 2022-06-03:
Hello, I am also interested in this topic. Any update?

Vijay-Karthick, 2022-06-03:
@marcorussi 
I wasn't able to get ot-rcp firmware running on BU-840XE following instructions in [this link](https://github.com/vrtsystems/openthread/tree/feature/fanstel-bt840x/examples/platforms/nrf528xx/nrf52840#support-for-the-fanstel-bt840x-bt840xe-and-usb840x).  I had to clone the openthread repo, edit the pin definitions to match the power amplifier pins according to BU-840XE datasheet and make the project.

However, this is still an USB based solution. I'd like to use a pre-certified otbr which has an ot-rcp device integrated and exposes an Ethernet interface - so I can write my business logic on another PC and communicate to this otbr interface purely for interfacing to Thread network.

sjlongland, 2022-06-03:
Well, I had another look this morning and found these:

* https://au.mouser.com/ProductDetail/Nordic-Semiconductor/nRF21540-DB?qs=pBJMDPsKWf0OWgaWWBwo%252BA%3D%3D
* https://au.mouser.com/ProductDetail/Nordic-Semiconductor/nRF21540-DK?qs=pBJMDPsKWf1XO%2FKzCuNBZg%3D%3D

The latter is non-stocked… but the fact is it exists on the market now, so I can possibly get some test hardware that I can use for testing both the Nordic nRF52840DK and Fanstel BT840X to ensure both boards function with the code modifications.

So maybe I can schedule some time to work on this.

rcrouthamel, 2022-06-07:
@Vijay-Karthick @marcorussi This product https://github.com/farmjenny/FarmJenny_LTE_Border_Router_HAT includes the Fanstel BT840XE.  The standard firmware (built from an openthread fork here https://github.com/farmjenny/openthread/tree/3000133 ) implements RCP over SPI and includes PA support inspired by @sjlongland .  This can be paired with a Raspberry Pi for a certified implementation.

marcorussi, 2022-06-07:
Thank you very much @rcrouthamel

Vijay-Karthick, 2022-06-07:
Thank you @rcrouthamel and @sjlongland

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui