openthread/openthread #4997: Process transmit security on acks

testter2, 2020-05-24:
Hello everybody,

In issue#4557 the function otPlatRadioTxAckStarted is called from radio.c whenever transmitting ack or enh-ack and this function calls the function HandleTransmitAckStarted which after that calls ProcessTransmitSecurity in mac.cpp. Does that mean the acks in thread1.2 will be secured and the Security Enabled field in the frame control should be set? also how will it be guaranteed to send the ack within the 192 us specified in the standard?

Thanks.

Irving-cl, 2020-05-25:
Hi @testter2 ,

* For the first question, `ProcessTransmitSecurity` would first check if Security Enabled field is set and only do the job when the bit is set. And whether to set the Security Enabled field in ack is determined by something else. 
Now we use radio driver to generate ack, the acks are not generated and sent by OpenThread core. But they might be updated before entirely sent out. For example, for `SSED`, we want to get an accurate CSL IE and append them to the ACK we sent. So we get the time exactly when the frame is to be sent. And at this moment there would be an hardware interrupt to call  `otPlatRadioTxAckStared` to update some content in the ack frame prepared by the radio driver. Since something in the frame has changed, we also need to process transmit security at this time.
So the radio driver actually generates the acks. In terms of whether to enable security for the ack frame, in nordic drivers: https://github.com/openthread/openthread/blob/master/third_party/NordicSemiconductor/drivers/radio/mac_features/ack_generator/nrf_802154_enh_ack_generator.c#L73
```
static void fcf_security_enabled_set(const uint8_t * p_frame)
{
    m_ack_data[SECURITY_ENABLED_OFFSET] |=
        (p_frame[SECURITY_ENABLED_OFFSET] & SECURITY_ENABLED_BIT);
}
```
The nordic driver uses the same security enabled bit with the received frame. So if the received frame enabled security, the ack enables security.

* For the second problem, as I mentioned above, the radio driver should ensure that the ack is sent within the 192 us.

testter2, 2020-05-25:
Hi @Irving-cl ,

Thanks for your support, in the thread specification there is the following:
"The CSL Receiver SHOULD be ready to receive **earlier** than the time specified in 6 CSL Phase and SHOULD stay ready to receive until **after** the time specified in CSL 7 Phase, according to the implementation and accuracy expectations."
 Can you illustrate to me how much time earlier and after? and where is it handled in the code? and shouldn't the PHY startup time be a parameter in the earlier time?

Thanks,

Irving-cl, 2020-05-26:
Hi @testter2 ,

We can take a look at the code in `sub_mac.cpp`:https://github.com/openthread/openthread/pull/4557/files#diff-b177f17d083567cb616a9b08cd905d43R793
```
case kCslSample:
{
    mCslState = kCslSleep;
    IgnoreError(Sleep());
    // OT_US_PER_TEN_SYMBOLS: computing CSL Phase using floor division.
    mCslTimer.StartAt(mCslSampleTime, mCslPeriod * OT_US_PER_TEN_SYMBOLS - OT_US_PER_TEN_SYMBOLS);
    otLogDebgMac("CSL sleep %u", mCslTimer.GetNow().GetValue());
    break;
}
```
This code handles when a CSL receiver stops its sampling and ready to sleep. At this moment it also sets the timer for its next wake-up. As you can see, we set the time point to `CslSampleTime + CslPeriod - TEN_SYMBOLS`(omit the unit conversion here). `CslSampleTime` is the time point when CSL receiver did sampling last time. And `CslSampleTIme + CslPeriod` is the time point when the CSL receiver would do its next sampling. This time point would also be calculated in radio driver: https://github.com/openthread/openthread/pull/4557/files#diff-244447b0b448a98c0e93f693ae85f98bR350 and passed to CSL transmitter via `phase` in CSL IE. So CSL transmitter would think that CSL receiver will do the next sampling at this time point. And we let CSL receiver starts sampling **OT_US_PER_TEN_SYMBOLS** ahead of the time that CSL transmitter think it would do. So the CSL receiver would do the sampling **OT_US_PER_TEN_SYMBOLS** earlier. The sampling would end at `CslSampleTime + kCslSampleWindow`:
```
mCslTimer.StartAt(mCslSampleTime, kCslSampleWindow * OT_US_PER_TEN_SYMBOLS);
```
So CSL receiver would stop the sampling `kCslSampleWindow` after the time that CSL transmitter thinks CSL receiver starts its sampling.

testter2, 2020-05-26:
Thanks a lot @Irving-cl for your support.

closed 2020-05-26 by jwhui