openthread/openthread #5020: [MeshCoP] Flood of MGMT_ACTIVE_SET.req from peer devices to leader

MatthewPennock, 2020-05-28:
**Describe the bug** A clear and concise description of what the bug is.

A leader will reject MGMT_ACTIVE_SET.req from peer devices if it crashes during the handling of a prior request. Peer devices don't know how to handle rejection, and will initiate new transactions forever.

The additional MeshCoP traffic impacts the application layer's performance.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id 
    d281be6aefe518f501719b39007fda8905c03eb6
2. IEEE 802.15.4 hardware platform 
    nrf52840 development kits with ot-cli-ftd images built from this repository
3. Build steps
     ```
    make -f examples/Makefile-nrf52840 USB=1 DEBUG=1 FULL_LOGS=1 LINK_RAW=1 COMMISSIONER=1 JOINER=1 BORDER_ROUTER=1 BORDER_AGENT=1
    ```
4. Network topology 
    Single leader, single peer (router)

It's hard to restart at exactly the right time. [This branch](https://github.com/MatthewPennock/openthread/tree/meshcop_mgmt_active_set_flood) helps reproductions with a `prime` command which makes devices restart while handling MGMT_ACTIVE_SET.req commands.

1. Form a network with two devices:
    To leader:
    ```
    dataset init new
    dataset channel 25
    dataset masterkey 00112233445566778899aabbccddeeff
    dataset commit active
    ifconfig up
    thread start
    ```
    To peer:
    ```
    dataset masterkey 00112233445566778899aabbccddeeff
    dataset active commit
    ifconfig up
    thread start
    ```
2. Inspect datasets (optional, useful to compare against after a reset)
    To leader:
    ````
    > state
    leader
    Done
    > dataset active
    Active Timestamp: 1
    Channel: 25
    Channel Mask: 07fff800
    Ext PAN ID: 742c6bb30e104f47
    Mesh Local Prefix: fde7:613b:9f35:f257/64
    Master Key: 00112233445566778899aabbccddeeff
    Network Name: OpenThread-9cca
    PAN ID: 0x9cca
    PSKc: 64605993860134ac689f783ab7be7c09
    Security Policy: 0, onrcb
    Done
    ````
    To peer:
    ````
    > state
    router
    Done
    > dataset active
    Active Timestamp: 1
    Channel: 25
    Channel Mask: 07fff800
    Ext PAN ID: 742c6bb30e104f47
    Mesh Local Prefix: fde7:613b:9f35:f257/64
    Master Key: 00112233445566778899aabbccddeeff
    Network Name: OpenThread-9cca
    PAN ID: 0x9cca
    PSKc: 64605993860134ac689f783ab7be7c09
    Security Policy: 0, onrcb
    Done
    ````
3. Roll back leader's timestamp:
    To leader:
    ```
    dataset activetimestamp 0
    prime
    dataset commit active
    ```
4. Have the leader rejoin the network:
    To leader:
    ```
    ifconfig up
    thread start
    ```
Step #3 forces all devices to send the leader a MGMT_ACTIVE_SET.req. The leader will restart while handling the first, but after committing the peer's more recent dataset (I believe). The `StateTlv::kAccept` response is lost in a queue, so our peers are unaware their petition succeeded.

**Expected behavior** A clear and concise description of what you expected to happen.

A leader could encourage peers to seek out new data sets by increment the MLE data version (as suggested by Jonathan Hui). However, my ultra naive implementation has only made the problem worse! 

**Console/log output** If applicable, add console/log output to help explain your problem.

1. Leader's SEGGER RTT log output: 
[leader_rtt.log](https://github.com/openthread/openthread/files/4697976/leader_rtt.log)

2. Peer's SEGGER RTT log output: 
[router_rtt.log](https://github.com/openthread/openthread/files/4697977/router_rtt.log)

3. Wireshark capture: 
[flood.zip](https://github.com/openthread/openthread/files/4697982/flood.zip)
    Link Key: 00112233445566778899aabbccddeeff

**Additional context** Add any other context about the problem here.

The RTT logs in this instance show a more complete picture as the leader becomes trapped in a rapid-fire conversation with itself. I hope to follow up very shortly with a reproduction using a previous commit to master (before new MeshCoP transactions were only initiated after CoAP's complete 93 second timeout) and without my increment of data version.

Thank you,
Matt

MatthewPennock, 2020-05-28:
PR #4981 masks the issue by rate-limiting the flood. This branch (https://github.com/MatthewPennock/openthread/tree/meshcop_mgmt_active_set_flood) diverges from `master` on May 13th (prior to rate limiting) and does not attempt to increment data version. The linked branch is where we first observed the behaviour.

Following the steps above, here are logs and a wireshark capture with observable network traffic:

1. Leader: [leader_rtt.log](https://github.com/openthread/openthread/files/4698180/leader_rtt.log)
2. Router: [router_rtt.log](https://github.com/openthread/openthread/files/4698187/router_rtt.log)
3. Wireshark: [flood_two.zip](https://github.com/openthread/openthread/files/4698189/flood_two.zip)

I find this set of logs much more helpful for observing the issue.

jwhui, 2020-05-28:
@MatthewPennock , thanks for this detailed report! I'll take a look and get back to you.

jwhui, 2020-05-28:
@MatthewPennock , thank you again for submitting this issue. I was able to reproduce the issue and work towards a proposed resolution.

The primary issue I see in this scenario is when the leader resets and reattaches to the network, it retrieves the old active dataset from the neighboring router. After that, the leader is stuck in an endless cycle of trying to update its own network data with the one stored in non-volatile. Of course, because the active timestamp values between the MGMT_ACTIVE_SET.req message and the dataset stored in non-volatile, no update actually occurs and the process repeats endlessly.

I have submitted a proposed resolution in #5020. There, the leader is prohibited from accepting an active dataset in an MLE Data Response from a neighboring device. The leader is authoritative when propagating new datasets via MLE. Instead, the leader simply increments its network data versions (stable and full) to ensure that its active dataset is propagated to the rest of the network.

Please review #5020 and let me know if this resolves your issue.

MatthewPennock, 2020-05-29:
I left a comment on #5021 that the proposed branch quiets the flood. Thank you for the detailed explanation and fix!

PR #5021 (2020-05-30): [mle] fix bug in leader active/pending dataset synchronization
The leader is the authoritative source when propagating active/pending
operational datasets. When the leader reboots, it is possible for the
leader to become out-of-sync. In particular, after the leader resets,
it restores its network data and active/pending operational datasets
using MLE Data Request/Response to retrieve them from a neighboring
device. If the neighboring device has an older active/pending
operational dataset, the leader will retrieve the older datasets and
never attempt to propagate the newer datasets that it may  [... cut]
closed 2020-05-30 by jwhui