openthread/openthread #5022: on-mesh GUA unreachable when manually configured

simonlingoogle, 2020-05-29:
**Describe the bug** 
When an on-mesh GUA prefix is configured without SLAAC and DHCPv6, a node with manually configured GUA is not reachable from another node that is not configured with GUA of this prefix. 

**To Reproduce**

1. Git commit id: 75a382d292b2271c0a6e202324550091753467f0
2. IEEE 802.15.4 hardware platform: OpenThread simulation 
3. Build steps
4. Network topology
Leader---Router

```bash
$ otns -bin ./cmake-build-debug/examples/apps/cli/
> node 1
Done
node 1> prefix add 2001::/64 pros med  # SLAAC not enabled
Done
node 1> netdataregister
Done
node 1> ipaddr
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:2000
fdde:ad00:beef:0:fd5d:286f:6cbb:7a39
fe80:0:0:0:b445:ce82:946a:1dea
node 1> node 2
Done
node 2> ipaddr add 2001::a:b:c:d  # manually config GUA on node 2
Done
node 2> ipaddr
2001:0:0:0:a:b:c:d
fdde:ad00:beef:0:0:ff:fe00:1400
fdde:ad00:beef:0:fa0b:36d5:6a8e:8db1
fe80:0:0:0:5c0e:5ba4:de3f:dcd9
Done
node 2> exit
Done
> ping 1 "2001::a:b:c:d" # ping unreachable
Done
> exit
```

**Expected behavior**
According to Spec. 5.15.7, node 1 should send address query for destination `2001::a:b:c:d`, because the prefix is on-mesh, and then sends ping request successfully. 
The source address for the ping request should be node 1's MLEID. 

**Console/log output**
[current.pcap.zip](https://github.com/openthread/openthread/files/4700450/current.pcap.zip)

However, if node 1 is also manually configured with a GUA of the same on-mesh prefix, it can do address query and send ping request successfully. 

Thoughts on this issue? 
@jwhui @librasungirl

jwhui, 2020-05-29:
@simonlingoogle , thanks for raising this issue!

This is because we do not currently check IPv6 destination addresses against the set of on-mesh prefixes.
https://github.com/openthread/openthread/blob/46305be08269a3f1ec2a3364d565270e090b2dd7/src/core/net/ip6.cpp#L1469-L1483

I'll submit a proposed fix soon.

jwhui, 2020-05-29:
Submitted #5024, please review.

simonlingoogle, 2020-05-30:
Thanks @jwhui . I will verify the fix ASAP.

simonlingoogle, 2020-06-01:
verified that this PR works: [current.pcap.zip](https://github.com/openthread/openthread/files/4708986/current.pcap.zip)

PR #5024 (2020-06-01): [ip6] forward any packet with on-mesh destination
Existing implementation only looks at addresses configured on the
network interface to determine what prefixes are on-link (on-mesh for
Thread). When the network interface does not have an assigned address
for a given on-mesh prefix, any packet with that on-mesh prefix will
not be forwarded to the Thread network.

This commit also checks the destination address against Thread's set
of on-mesh prefixes, independent of whether addresses out are assigned
out of the on-mesh prefix.

Resolves #5022
closed 2020-06-01 by jwhui