openthread/openthread #5041: [mac frame] enhanced ack should not the security mode/counter from rx frame

abtink, 2020-06-02:
Recently PR https://github.com/openthread/openthread/pull/4884 added code for generation of enhanced ack in `Mac::Frame`.

When generating a enhanced ack, the current implementation uses the security frame counter and key id from the received frame (the frame being acked) for the secured enhanced ack frame.
https://github.com/openthread/openthread/blob/0565f44a4c538074f2bf3e8ccd7b92aef41efdd8/src/core/mac/mac_frame.cpp#L1169-L1179

Note that "seq number" on ack is set to match the received frame (which is correct and expected behavior) but I think using security frame counter from received frame for the enhanced ack is not correct and it should be tied to sender/device and not the received frame (i.e. sec frame counter should be incremented for each frame sent by device whethere it is a data frame or an enahnced ack frame).

FYI: @bukepo , @brianljt , @Irving-cl, @jwhui

Irving-cl, 2020-06-03:
Yes, that's true. I discussed with @brianljt . I will change this after PR#4976(which moves frame counter to subMac) gets merged.

brianljt, 2020-06-03:
Yes, it is a known issue and is raised here https://github.com/openthread/openthread/pull/4990#discussion_r428402209

Current code does not generate any Enh-Ack so that it is fine to keep it as is. We can address this after PR #4976 is merged.

brianljt, 2020-06-22:
Only frame counter should be using Enh-ACK sender's transmit frame counter. Other security related fields(key ID and key ID mode) should be same as received frame.

> _**Spec 15.4-2015 Section 7.3.3 Ack frame format**_
If the Security Enabled field in the Frame Control field is set to one, the Enh-Ack frame is protected using the procedure described in 9.2.1. The SecurityLevel, KeyIdMode, KeySource and KeyIndex shall be set to match the corresponding fields of the frame that is being acknowledged.

jwhui, 2020-06-22:
Resolved by #5118

closed 2020-06-22 by jwhui