openthread/openthread #5047: [cli] ‘prefix’ output in Thread Simulation is incorrect

dsyx, 2020-06-04:
```sh
$ lsb_release -a
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.4 LTS
Release:        18.04
Codename:       bionic
```

During the [Simulating a Thread network with OpenThread](https://codelabs.developers.google.com/codelabs/openthread-simulation-posix/#3), the following problems were found:

Node 1:

```
> dataset
Active Timestamp: 1
Channel: 16
Channel Mask: 07fff800
Ext PAN ID: b5588264daa2d0b0
Mesh Local Prefix: fd19:729d:437f:5e4a/64
Master Key: cbea31686e38387f719e485508ef2076
Network Name: OpenThread-0d0f
PAN ID: 0x0d0f
PSKc: 660bdfad83906944ea48e911d22c0513
Security Policy: 0, onrcb
Done
> ipaddr
fd19:729d:437f:5e4a:0:ff:fe00:fc00
fd19:729d:437f:5e4a:0:ff:fe00:4000
fd19:729d:437f:5e4a:96e1:8a68:553:341b
fe80:0:0:0:3ce4:9f73:f3dd:eaea
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
|  6 | 0x1800 |       63 |         0 |     3 |      3 |   5 | d6e7ad0df57f31b7 |
| 16 | 0x4000 |       63 |         0 |     0 |      0 |   0 | 3ee49f73f3ddeaea |

Done
> prefix
Done
```

I expect that ‘prefix’ should return `Mesh Local Prefix: fd19:729d:437f:5e4a/64`, but it does not.

jwhui, 2020-06-04:
@dsyx , thanks for raising this issue.

As noted in the [prefix command documentation](https://github.com/openthread/openthread/tree/master/src/cli#prefix), the `prefix` command only lists the on-mesh prefixes contained within the Thread Network Data. In other words, prefixes that are configured/registered individually by one or more devices.

The Mesh Local Prefix is special in that it is part of the Active Operational Dataset and stored persistently on every device commissioned for the given Thread network. As a result, the Mesh Local Prefix available via the `dataset` command.

Hope that helps.

dsyx, 2020-06-05:
Thank you for your answer.

Do you mean that `prefix` will only list GUA Prefix, not ULA Prefix?

But I think the `prefix` command may be more friendly if it can list all types of prefixes.

Because for almost every item in `dataset`, there is a corresponding single command, such as: `channel`, `panid`, `masterkey` ...

According to general thinking, people want to get the `Mesh Local Prefix` in `dataset` and they should enter `prefix` naturally.

Would it be better if the output of `prefix` was designed as follows:

```
> prefix
ULA Prefix:
     Prefix 1
GUA Prefix:
     Prefix 2
     ...
```

jwhui, 2020-06-05:
Thread uses IPv6 prefixes in different ways:

1. **Mesh-Local**: required to support Thread control protocols required to form and maintain the mesh. The Mesh Local Prefix is configured via Operational Datasets and stored in non-volatile memory.

1. **On-Mesh**: that are analogous to [RFC4861](https://tools.ietf.org/html/rfc4861) "on-link" prefixes. These are configured via Thread Network Data by individual devices on the mesh. On-mesh prefix may be [RFC4193](https://tools.ietf.org/html/rfc4193) unique-local prefixes or globally routable prefixes.

1. **Routes**: that are analogous to [RFC4191](https://tools.ietf.org/html/rfc4191) more-specific routes. These are configured via Thread Network Data by individual devices on mesh. Routes may be [RFC4193](https://tools.ietf.org/html/rfc4193) unique-local prefixes or globally routable prefixes.

The `prefix` command is used to configure and manage "on-mesh" prefixes only. The `prefix` command may be used to list on-mesh prefixes already configured and add/remove on-mesh prefixes. I think it would be equally confusing to say that the `prefix` command can be used to retrieve information from an Operational Dataset, but only the `dataset` command can be used to set information in the Operational Dataset.

Note that the individual `channel`, `panid`, `masterkey` commands were created before Operational Datasets were introduced in Thread 1.1. The preferred method for managing network configuration parameters is now via the `dataset` command. We have updated all our docs, guides, and code labs to use the `dataset` command when forming/joining networks.

One area where individual `channel`, `panid`, etc. commands are useful is for the sniffer tool. However, a sniffer tool is not really operating as a Thread device.

closed 2020-06-05 by dsyx