openthread/openthread #5060: [data-poll-sender] question about the polling behaviour

LuDuda, 2020-06-05:
During testing, I have experienced strange behavior on Sleepy End Device.

SED has a polling interval set to default 236 seconds. The test consists of only two devices Leader and SED. Both devices constantly send packets to each other (the channel is also quite busy by external interferences). Few times I saw the situation when despite parent had tens of frames queued for SED, SED stopped sending consecutive Data Poll Requests although the FP bit was set to 1.

From the logs it looks like SED was able to fetch let's say N frames, it sends N+1 Data Poll Request, it received the response where FP was set to true. It correctly received the frame, and basically it waited for 236 seconds. I don't see that SED even tried to send the next Data Request indicated by the `-MAC-----: Sending data poll` log or by the logs on the Leader. This situation happens very rarely, and I don't see anything suspicious in the logs. Also before this situation happens there are no Data Poll Request retransmissions indicated by the proper logs as well.

Logs at the end:
```
[DEBG]-MAC-----: Sent data poll, fp:yes
[DEBG]-MAC-----: Finishing operation "TransmitPoll"
[DEBG]-MAC-----: Starting operation "WaitingForData"
[DEBG]-MAC-----: Received frame from short address 0x2400
[DEBG]-MAC-----: Rx security - frame counter 508
[DEBG]-MAC-----: Finishing operation "WaitingForData"
[DEBG]-MAC-----: RadioState: Receive -> Sleep
[DEBG]-MAC-----: Idle mode: Radio sleeping
```

@abtink Do you have any idea, why this happens?

The only suspicious line in the code I found, which may be relevant is [this](https://github.com/openthread/openthread/blob/master/src/core/mac/mac.cpp#L587). This might have sense since I don't see `Request to start operation "TransmitPoll"` log.

abtink, 2020-06-05:
@LuDuda this may be an issue on the sender (parent) side not setting frame pending on the data frame.

Here is the flow:

- Child sends data poll
- Parent acks the data poll and set the fp in ack
   - The log `"Sent data poll, fp:yes"` indicates  ack from parent has `fp` flag set
- Child would then stay in rx mode to receive the frame form parent.
-  If child receives a frame from its parent: 
   - If the received frame does not have `fp` set, then child would go to sleep and send next data poll based on the poll period. 
   - Otherwise (if the received frame has `fp` set), then the child would immdiately send a next data poll.
This is checked here in rx path in Mac layer:
https://github.com/openthread/openthread/blob/2e60748d5b1015a42f579891d0d8ef065f3bca31/src/core/mac/mac.cpp#L1681
And then processed here to trigger an immediate data poll tx in `fp` is set on received frame:
https://github.com/openthread/openthread/blob/2e60748d5b1015a42f579891d0d8ef065f3bca31/src/core/mac/data_poll_sender.cpp#L309-L328
- If the child does not receive any frame from parent, then it's conidered a data poll timeout and the related mechansim to handle timeout gets started (where data polls are tried again)

LuDuda, 2020-06-06:
Thanks, @abtink,

Thank you for your explanation. While I do like the current implementation where SED checks FP in the data message, isn't it against the specification? I thought that SED needs to send additional Data Poll Request to see if there is more data pending on the Parent. I will add some more logs on the Router side then to see if indeed FP is not set in the data message itself.

abtink, 2020-06-06:
> While I do like the current implementation where SED checks FP in the data message, isn't it against the specification? I thought that SED needs to send additional Data Poll Request to see if there is more data pending on the Parent.

@LuDuda , I am not super familiar with details on spec but I don't think it's the case (I would say it would be rather inefficient if this is required since there would always be one extra data poll at end of message exchanges between parent and child). 

From what I remember the 15.4 spec itself delegated when/how polls are sent to the child/SED. I think the use-case was that some SED devices may be very power restricted (e.g., doing energy harvesting) so they should decide when they want to send data poll.

I did check spec a bit (2011-15.4). Here is what I found:

> _On receipt of the acknowledgment frame with the Frame Pending field set to one, a device shall enable its receiver for at most macMaxFrameTotalWaitTime to receive the corresponding data frame from the coordinator. If there is an actual data frame pending within the coordinator for the requesting device, the coordinator shall send the frame to the device using one of the mechanisms described in this subclause. If there is no data frame pending for the requesting device, the coordinator shall send a data frame without requesting acknowledgment to the device containing a zero length payload, indicating that no data are present, using one of the mechanisms described in this subclause._
> 
> _If the requesting device does not receive a data frame from the coordinator within macMaxFrameTotalWaitTime or if the requesting device receives a data frame from the coordinator with a zero length payload, it shall conclude that there are no data pending at the coordinator. If the requesting device does receive a data frame from the coordinator, it shall send an acknowledgment frame, if requested, thus confirming receipt._
> 
> **_If the Frame Pending field of the data frame received from the coordinator is set to one, the device still has more data pending with the coordinator. In this case it may extract the data by sending a new data request command to the coordinator._**
> 

Last paragraph mentions the "FP" in the received frame is used to decide if there are more pending frame but in that case uses "**may** extract". In OT implementation we immediately send a data poll when the received frame indicates "FP" falg.

Interestingly earlier paragraph allows device to conclude there are no pending frames in the case of a timeout.  In OT implementation, we would retx the data poll in such a case.

LuDuda, 2020-06-07:
Hello @abtink it's been a long time since I read the 15.4-2006 specification (Thread 1.1). For some reason, I always thought that reading the FP from data message is something desired but not available without extensions. I believe it's because according to the 802.15.4-2006 specification there is no indication about FP bit value in `MCPS-DATA.ind` structure (in original form).

Anyway, the current implementation is something we want to have :) Thank you for your thoughts.
I'll let you know in case I'll find why I observe such unwanted behaviour.

abtink, 2020-06-08:
Thanks @LuDuda , sounds good.

jwhui, 2020-06-30:
Closing stale issue.

closed 2020-06-30 by jwhui