openthread/openthread #5064: Update openthread.io Pre-Built NCP Firmware

brianljt, 2020-06-08:
With recent change merged (PR #4976), the new posix host driver requires to use the new RCP image. Otherwise, user will encounter Spinel parsing error (like the one here: https://groups.google.com/g/openthread-users/c/RtolmaADSc4).

This issue is opened to track the updates to the prebuilt firmware in openthread.io for both nrf52840 and cc2538.

brianljt, 2020-06-08:
Thanks Yakun (@bukepo) for pointing this issue out.

Vyrastas, 2020-06-08:
The nrf52840 NCP image was built with
BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 USB=1 LINK_RAW=1

And the cc2538 NCP image was built with
BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 LINK_RAW=1

Should I use the same flags for the RCP versions?

brianljt, 2020-06-09:
Hi Jeff. Looks like we only put NCP image on openthread.io (https://openthread.io/platforms/co-processor/firmware). Then I think it is fine to just update the NCP image and use the same flags. No need to put RCP image on openthread.io. Thank you!

bukepo, 2020-06-09:
> The nrf52840 NCP image was built with
> BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 USB=1 LINK_RAW=1
> 
> And the cc2538 NCP image was built with
> BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 LINK_RAW=1
> 
> Should I use the same flags for the RCP versions?

Those flags will not affect RCP build, it's OK to use the same flags.

Vyrastas, 2020-06-09:
Oh I thought you meant to replace the NCP images with RCP.  I'll just update the NCP ones then.

Vyrastas, 2020-06-17:
I've updated the nRF52840 NCP firmware on openthread.io.

CC2538 has been removed (along with all the instructions) as the platform is EOL and downloads of the firmware are extremely low.

closed 2020-06-17 by Vyrastas