openthread/openthread #5073: Commissioning with many units makes NCP sometimes crash and takes for ever to finish

musalk-dk, 2020-06-09:
**Describe the bug** 
Starting wpanctl commissioner joiner-add "*" 1200 PPSSKK
having eg. 8 units mostly succeed after 2-5 min.
having 50 units will take 10 - 30 min with a likely crash of the NCP. and restart of the joining process multiple times (tested with different timeouts)
having 4 units can sometimes be done in 10-15 sec other times a 1 min or 2.

**To Reproduce**
start commissioner on gateway with "*" for EUI64.
also tested with ot-ncp and RCP mode
also tested with ot-commissioner example (generally works better/ faster) but also with possible crashes.
Usually a few units joins in then a long break then another. 
Sometimes a few units will join after another and then a long break.
Sometimes the NCP crashes.

joiner pseudo code:
otLinkSetPanId(OtInst, 0xffff);
otJoinerStart(OtInst, "PPSSKK", NULL, "Test", "device", NULL, "test", NULL, NULL);
while(not joined) call openthread processes
if joiner timeout then wait 10sec and restart joiner.

1. Git commit id
Tried commits about 6 different commits from 1 month ago to 8 month ago no difference.
For both gateway, ncp and the FTD joining devices.

2. IEEE 802.15.4 hardware platform
nrf52840 with SPI to RPi in RCP or NCP mode

3. Build steps
Tested with embedded TLS and without, and a range of full to minimal options.
Default config except SPI pins and OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE 4096

4. Network topology
RPi gateway
4 - 60 units in an office space (almost always a direct link is possible.)

**Expected behavior**
units joining one by one with 5-10 sec in between.

**Additional context**
Many tests have been run. in the begining assuming that we were just too far behind on our commit so we got everything up to date last month and ran a bunch of tests. First assuming that it was too much for the NCP so we tested with ot-ncp using the nrf52840 as RCP, Then tested the ot-commissioner example code which did better for the most part. But still crashing the NCP.
Maybe we are just missing some tweaks of our config and not just run on default config.
But having no idea which config will affect the commissioning. 
Maybe number of message buffers?
However the NCP crashing seams not to be intended even if it runs out of message buffers.

LuDuda, 2020-06-09:
What do you mean by "NCP sometimes crashes"? Do you see some assert? Is this the same for RCP architecture?

musalk-dk, 2020-06-10:
Crash as in sometimes we can do the following sequence to regain function when on SPI:
wpanctl reset
Systemctl restart wpantund
wpanctl reset
Systemctl restart wpantund

First we thought that it was related to the SPI and spinel. So we tried stock ncp firmware on nrf52840-dongle via USB. But a crash there usually results in having to replug the USB or hard reset the nrf.

We had to get on with the application and wanted to try and raise the issue here to see if anyone else had experienced it or could give us pointers for where to look.

We have not yet had it connected to a dev board with debugging. But as the issue persists in all configurations it seams ikely to be related to the commissioner code or mbed tls.
I cannot remember for sure if the rcp crashed or if it was the ot-ncp.

jwhui, 2020-06-15:
@musalk-dk , unfortunately, I don't think we've experienced this behavior so far. Any chance you could hook up a debugger and provide more insight into the issue?

musalk-dk, 2020-06-17:
Will do. We returning to work on this issue next week and I will come with updates

musalk-dk, 2020-06-29:
Just a KeepAlive update. 
As of today we have a setup where the NCP( via USB) crashes after joining the first FTD.
Tomorrow we will hook it up with a debugger to see what goes on.

jwhui, 2020-06-30:
> Just a KeepAlive update.
> As of today we have a setup where the NCP( via USB) crashes after joining the first FTD.
> Tomorrow we will hook it up with a debugger to see what goes on.

@musalk-dk , thanks for the update. Look forward to any more information you can provide.

musalk-dk, 2020-06-30:
Okay. Short'ish update.
**Test 1:**
NCP: pre-compiled firmware from Nordic Semi. (nRF5_SDK_for_Thread_and_Zigbee_v4.1.0_32ce5f8) NCP on nrf52840-dongle over USB.

1. Commissioning using wpanctl possible. (Probably will take a few tries to make it crash again. But did not have the patience)
2. Using otbr-commissioner or the ot-commissioner examples results in:
 - All units getting to OT_JOINER_STATE_CONNECT
 - One FTD joiner gets joined
 - NCP crash by ending up in memmagr fault IRQ. 

But as that firmware is pre compiled and no map file is available there is not much to get from that. Maybe there is a bug with the external heap (i believe they compile it with).

**Test 2**
I pulled the latest openthread and only changed the pins for uart and SPI. Build the NCP with: "make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_FORWARD=1 LINK_RAW=1 USB=0"
Connected via UART to PI.

I can commission the units fine but slowly (56 units in 15min) using wpanctl com... (no crashes) 
-- Am i wrong in expecting that it should be able to do it in a couple of minuts?
Ofcause the NCP has limited resources and it looks like only 6-10 get through then the others timeout and the next bunch gets in when they retry joining and so on...

So trying both the ot-commissioner (compiled and run on ubuntu other than the border router and has worked previously also tried a fresh pull today) or the otbr-commissioner (Nordic rpi image RaspPi_OT_Border_Router_Demo_v4.0.0-1.alpha).
Has only resulted in the units only getting to OT_JOINER_STATE_CONNECT.
in the output from otbr-commissioner i get: DTLS handshake failed: -0x6a80!

**Current crash theories**
I think there are 2 different issues i am having.
One seams to be a Nordic Semi issue with idle sleep and the crypto cell. Some other non commissioning related freezes or extreme slow response from the NCP seams solved after commenting out any entering sleep mode.

The other seams to do with mbedtls, the border agent or the DTSL forwarding. Could it be that whatever triggers the nordic pre-compiled firmware to crash is the same that makes the external commissioners not work? Not resulting in a crash but no one gets joined?

Any ideas or pointers are much appreciated.

jwhui, 2020-06-30:
@musalk-dk , regarding the "Test 2" failure, this sounds similar to #5142 

@LuDuda , and thoughts on the above?

musalk-dk, 2020-07-07:
I checked the #5142 and adding the coaps api seams to have fixed the joining. Not sure if it was the only thing. 
I pulled latest ot-commissioner, openthread rebuild
And commissioning got 58 units joined in less than 2 min. Which I find acceptable. 

But the joy was short lived hence the slow to update this issue.
The ncp drops to end-device randomly.
So I pulled the latest otbr and rebuild
Then I cannot commission external or internal.
So tried Nordic border router stock example image. And the ncp loses contact af 5 min requiring hard reset. Did not have time to find out what is going on here but seams to be another issue.

Conclusion to this issue. 
The trouble we have had with the commissioning and crashing looks to be related to nrf. 
We have disabled mcp sleeping in the example sdk code ncp from Nordic using new ot libs. which resolved som random disfunction/lost contact with the ncp during normal operation.
The stock ncp firmware from Nordic semi. Ends up in mem mangr fault when using external commissioning in the last repeatable test, one got joined after which the men mangr fault. If there is a bug in the nrf heap it would explain alot as different issues persists with different builds or with the change to using the crypto cell. But cannot say for sure and don't have the manpower to investigate.

Now I just need to get it stable again when not commissioning. 
Thanks

closed 2020-07-07 by musalk-dk