openthread/openthread #5090: Simulation instances never connect

Artox, 2020-06-12:
Nodes of the simulation platform never join into a network ... ...

Attempted were both
- docker image (https://codelabs.developers.google.com/codelabs/openthread-simulation/#0)
- build from source (https://codelabs.developers.google.com/codelabs/openthread-simulation-posix/#0)

For sake of simplicity I am only providing the steps for reproducing my problem with the docker instructions.

## steps to reproduce

- `docker pull openthread/codelab_otsim:latest`
- first shell:
   ```
   docker run --name codelab_otsim_ctnr -it --rm --sysctl net.ipv6.conf.all.disable_ipv6=0 --cap-add=net_admin openthread/codelab_otsim bash
   cd ~/src/openthread
   ./output/x86_64-unknown-linux-gnu/bin/ot-cli-ftd 1
   > dataset init new
   > ifconfig up
   > thread start
   > state
   ```
- second shell:
   ```
   docker exec -it codelab_otsim_ctnr bash
   cd ~/src/openthread
   ./output/x86_64-unknown-linux-gnu/bin/ot-cli-ftd 2
   > dataset masterkey <key>
   > dataset panid <panid>
   > dataset commit active
   > ifconfig up
   > thread start
   > state
   ```

## Expected output

- first shell:
   ```
   > state
   leader
   Done
   ```

   - second shell:
   ```
   > state
   child
   Done
   ```

## Actual output

- first shell:
   ```
   > state
   leader
   Done
   ```

   - second shell:
   ```
   > state
   detached
   Done
   ```

## complete console logs

- first shell:
   ```
   docker run --name codelab_otsim_ctnr -it --rm --sysctl net.ipv6.conf.all.disable_ipv6=0 --cap-add=net_admin openthread/codelab_otsim bash
   root@35e4017b3ad6:/# cd ~/src/openthread
   root@35e4017b3ad6:~/src/openthread# ./output/x86_64-unknown-linux-gnu/bin/ot-cli-ftd 1
   
   > dataset init new
   Done
   > dataset
   Active Timestamp: 1
   Channel: 21
   Channel Mask: 07fff800
   Ext PAN ID: 5a736b5ea726ee58
   Mesh Local Prefix: fdd4:c22c:3b11:9d0/64
   Master Key: 19b655d426ad5a62a039d43d5d305fab
   Network Name: OpenThread-1367
   PAN ID: 0x1367
   PSKc: 5ad2ca52d8f9344b93bee0fb4a04d3c2
   Security Policy: 0, onrcb
   Done
   > ifconfig up
   Done
   > thread start
   Done
   > state
   leader
   Done
   > ipaddr
   fdde:ad00:beef:0:0:ff:fe00:fc00
   fdde:ad00:beef:0:0:ff:fe00:2c00
   fdde:ad00:beef:0:9fb6:e74f:1927:2a33
   fe80:0:0:0:74b4:f508:eb53:377b
   Done
   > dataset
   Active Timestamp: 1
   Channel: 21
   Channel Mask: 07fff800
   Ext PAN ID: 5a736b5ea726ee58
   Mesh Local Prefix: fdd4:c22c:3b11:9d0/64
   Master Key: 19b655d426ad5a62a039d43d5d305fab
   Network Name: OpenThread-1367
   PAN ID: 0x1367
   PSKc: 5ad2ca52d8f9344b93bee0fb4a04d3c2
   Security Policy: 0, onrcb
   Done
   > state
   leader
   Done
   ```

- second shell:
   ```
   docker exec -it codelab_otsim_ctnr bash
   root@35e4017b3ad6:/# exit^C
   root@35e4017b3ad6:/# cd ~/src/openthread
   root@35e4017b3ad6:~/src/openthread# ./output/x86_64-unknown-linux-gnu/bin/ot-cli-ftd 2
   
   > dataset masterkey 19b655d426ad5a62a039d43d5d305fab
   Done
   > dataset panid 0x1367
   Done
   > dataset commit active
   Done
   > ifconfig up
   Done
   > thread start
   Done
   > state
   detached
   Done
   > state
   detached
   Done
   ```

jwhui, 2020-06-12:
@Artox , on the "first device", you are missing the command to commit the dataset to the Active Dataset:
```
> dataset init new
Done
> dataset commit active
Done
> ifconfig up
Done
> thread start
Done
```

For some examples, see:

- [src/cli/README_DATASET.md](https://github.com/openthread/openthread/blob/master/src/cli/README_DATASET.md)

- [Simulating a Thread network using OpenThread in Docker](https://codelabs.developers.google.com/codelabs/openthread-simulation/index.html?index=..%2F..index#2)

Hope that helps.

Artox, 2020-06-12:
@jwhui Thank you very much - I did indeed muss this step in the codelab despite that it's there.

I would probably update the relevant codelab section and separate the *dataset commit active* step from the previous dataset init, to make it more visible.

And in fact - after a little patience, I got a network :)

jwhui, 2020-06-12:
@Artox, great to hear you were able to form a network!

> I would probably update the relevant codelab section and separate the _dataset commit active_ step from the previous dataset init, to make it more visible.

@Vyrastas, thoughts on the above?

Vyrastas, 2020-06-17:
Sure, let me take a look... I've got those other revisions in progress for ot-daemon, I'll add this update to that.

Vyrastas, 2020-06-17:
I've updated https://codelabs.developers.google.com/codelabs/openthread-simulation-posix

Docker one should be updated later today.

jwhui, 2020-06-18:
@Vyrastas , thanks!

closed 2020-06-18 by jwhui