openthread/openthread #5092: CSL synchronization

testter2, 2020-06-13:
Hello everybody,

In issue #4557 to overcome the clock drift between the CSL receiver and the CSL transmitter the CSL receiver will enable its receiver earlier than the sampling time by period equals OT_US_PER_TEN_SYMBOLS which is defined as 160 us. 
this is handled in the sub_mac.c 
`case kCslSample:
{
    mCslState = kCslSleep;
    IgnoreError(Sleep());
    // OT_US_PER_TEN_SYMBOLS: computing CSL Phase using floor division.
    mCslTimer.StartAt(mCslSampleTime, mCslPeriod * OT_US_PER_TEN_SYMBOLS - OT_US_PER_TEN_SYMBOLS);
    otLogDebgMac("CSL sleep %u", mCslTimer.GetNow().GetValue());
    break;
}`
My question is if the PHY will take significant time could be more than 160 us to go from sleep state to receive state should that time be added to the 160 us So instead start enabling the receiver at T2 we shall start it at T1?

[image]

Thanks,

Irving-cl, 2020-06-16:
Hi @testter2 , thanks very much for the proposal. This is a very good question.

I can actually not tell if PHY would take significant time(usually more than 160us or not). Take **nrf52840** as an example, when the timer fires, `otPlatRadioReceive` would be called, and then calls `nrf_802154_receive`. I think it takes short time, but still, not sure about this.

However, according to our experience, the transmitter is also likely to do its transmission a little bit late than the  `phase`. And that's why we don't put the `phase` in the middle of receiver's sample window, instead we put it at almost the begining of its sample window:
[image]

So in the above picture, you worry that **T2** will maybe delay a little bit, even go after **T3**. Regarding our experience, the actual transmission time is likely to lies after **T3**, so this might not be a big problem. Of course, it's very welcome if you would like to do some experiments about that and if PHY would take much time.

testter2, 2020-06-18:
Hi @Irving-cl, thanks for your support.

I am suggesting if it's possible you can add an api in the platform to get the PHY start-up time then adding this time to OT_US_PER_TEN_SYMBOLS here
mCslTimer.StartAt(mCslSampleTime, mCslPeriod * OT_US_PER_TEN_SYMBOLS - (OT_US_PER_TEN_SYMBOLS+PHY_START_UP_TIME);
to cancel its effect.

and I have another question in the thread 1.2 specification there is the following:

"The CSL Transmitter provides clock accuracy to the end device via the CSL Accuracy TLV. A  CSL Receiver MAY take this and its own local clock drift into consideration when determining how often to synchronize." 

Could you please tell me where is the part of the code where the CSL timeout will be changed according to the clock accuracy's values?

Thanks,

Irving-cl, 2020-06-19:
Hi @testter2, thanks for the good suggestion.

To be honest, I think I don't know how to measure the time for PHY start-up. I take a look at the apis on nrf platforms and I didn't find anything that seems to do that thing. From my perspective, I don't know what actually happens when PHY start-up. All I can see in the code is just, the receive api is called. Do you have any idea about how can we test this time? Or maybe we can ask people from Nordic what would happen at this moment and will it takes much time. If this is a really short time, I think we can just ignore it.

Regarding the CSL Accuracy, the current code doesn't handle it. The reason seems to be that, on some devices we use, the accuracy is more than 255 ppm which cannot be put in a`CSL accuracy TLV`. And the solution for this just suspended. But we think we'll do this thing. We will include it later.

testter2, 2020-06-29:
Hi @Irving-cl  thanks for your help,

In mac 2015 standard the clock drift between two devices is calculated as follows :
"Drift = [Clock accuracy] × time elapsed since the last synchronizing event"
then the max drift would be calculated as follows:
"maxDrift = [Clock accuracy] × cslTimeout"
is the 160 us (how much earlier the csl receiver starts receiving ) calculated according to the above equation ?

Thanks,

Irving-cl, 2020-06-30:
Hi @testter2 ,

160 us is 10 symbols. In CSL, 10 symbols is used as a basic unit.(Referring IEEE 802.15.4-2015 7.4.2.3 CSL IE).

Regarding the value 160, we can find for **2450** MHz PHY, the symbol rate is **62.5k/s**. So 1 symbol is `1000000 / 62500 = 16` us. 10 symbols is 160us.

testter2, 2020-07-14:
Thanks a lot @Irving-cl  for your help.

closed 2020-08-11 by jwhui