openthread/openthread #5112: Should not send ADDR_QRY.qry for off-mesh destination

simonlingoogle, 2020-06-17:
**Describe the bug**
OT sends `ADDR_QRY.qry` for destination address of `off-mesh` prefix. 

**To Reproduce**
1. Git commit id: master
2. IEEE 802.15.4 hardware platform: OTNS
3. Steps: 
```bash
make -f example/Makefile-simulation OTNS=1

$ otns
> add router x 100 y 100
1
Done
> add router x 100 y 200
2
Done
> node 1 "prefix add 2001::/64 pars med"
Done
> node 1 "netdataregister"
Done
> node 1 "ipaddr"
2001:0:0:0:59eb:2331:ac03:2b7e
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:3c00
fdde:ad00:beef:0:cad2:dcc0:d14d:163c
fe80:0:0:0:bcdf:1bb4:f60f:fbdd
Done
> node 2 "ipaddr"
2001:0:0:0:e5a0:be90:273:71c
fdde:ad00:beef:0:0:ff:fe00:c000
fdde:ad00:beef:0:4228:c5af:aa9c:2e97
fe80:0:0:0:2c81:d84d:47e:820d
Done
> ping 1 "2001:0:0:0:e5a0:be90:273:71c"
Done
> exit
```

Use wireshark to find the `ADDR_QRY.qry` in the pcap: [current.pcap.zip](https://github.com/openthread/openthread/files/4793285/current.pcap.zip)

4. Network topology
```Leader---Router```

**Expected behavior**
OT should not send `ADDR_QRY.qry` for `off-mesh` destination according to `Spec. 5.15.7 Router Behavior`:
```
22 If the IPv6 Destination Address is not on-mesh, then the Router uses the IPv6 Source
23 Address to determine the set of Border Routers that offer routes associated with the same
24 Provisioning Domain as the prefix of the IPv6 Source Address.
```

I also check the case when the `off-mesh` prefix is **NOT** using SLAAC, `ADDR_QRY.qry` is still sent for a fake `off-mesh` destination, as shown in pcap: [current_no_slaac.pcap.zip](https://github.com/openthread/openthread/files/4793320/current.pcap.zip)

Is this expected behavior for off-mesh destinations? 
Thoughts? @jwhui @bukepo @librasungirl

jwhui, 2020-06-17:
@simonlingoogle , thanks for identifying and raising this issue!

When checking if a prefix is on-mesh, it appears we are only checking for the presence of a Border Router TLV and not the Border Router Entry flags:
https://github.com/openthread/openthread/blob/e653478c503d5b13207b01938fa1fa494a8b87d3/src/core/thread/network_data_leader.cpp#L254-L273

I can look at proposing a fix.

simonlingoogle, 2020-06-17:
> When checking if a prefix is on-mesh, it appears we are only checking for the presence of a Border Router TLV and not the Border Router Entry flags:

In the `Ip6::IsOnLink` method, we also have this code:
https://github.com/openthread/openthread/blob/e653478c503d5b13207b01938fa1fa494a8b87d3/src/core/net/ip6.cpp#L1478-L1484

Per my understanding, this code returns `true` if the destination address has the same prefix with any local unicast address, even if the destination address if `off-mesh`. This should not be a correct behavior?

jwhui, 2020-06-17:
> Per my understanding, this code returns `true` if the destination address has the same prefix with any local unicast address, even if the destination address if `off-mesh`. This should not be a correct behavior?

In this case, the prefix should be considered on-link, which is consistent with traditional IPv6 stacks. If the address does not also specify an on-link prefix, it should set the prefix length to /128.

simonlingoogle, 2020-06-17:
> In this case, the prefix should be considered on-link, which is consistent with traditional IPv6 stacks. If the address does not also specify an on-link prefix, it should set the prefix length to [...]

I am not sure if I have fully understood this. Do you mean that for an `off-mesh` prefix with SLAAC on, the node should generate unicast address of this prefix with prefix length set to `/128` rather than the prefix length of the prefix ?

jwhui, 2020-06-17:
> I am not sure if I have fully understood this. Do you mean that for an `off-mesh` prefix with SLAAC on, the node should generate unicast address of this prefix with prefix length set to `/128` [...]

Yes, the prefix in this case specifies the subnet mask and is used for on-link determination. If any other address matching the same prefix should NOT be considered on-link, then the prefix should be set to /128.

simonlingoogle, 2020-06-17:
> Yes, the prefix in this case specifies the subnet mask and is used for on-link determination. If any other address matching the same prefix should NOT be considered on-link, then the prefix should [...]

Then, my question would be how does the node know if the addresses of an `off-mesh` prefix should be considered on-link or not? The prefix in `Network Data` does not have this information.

jwhui, 2020-06-17:
> Then, my question would be how does the node know if the addresses of an `off-mesh` prefix should be considered on-link or not? The prefix in `Network Data` does not have this information.

The Prefix TLV carries sub-TLVs. The Border Router Entry has an on-mesh flag. If any Border Router Entry has the on-mesh flag set, it should set the prefix to /64. Otherwise, it should set to /128.

simonlingoogle, 2020-06-17:
> > Then, my question would be how does the node know if the addresses of an `off-mesh` prefix should be considered on-link or not? The prefix in `Network Data` does not have this information.
> 
> The Prefix TLV carries sub-TLVs. The Border Router Entry has an on-mesh flag. If any Border Router Entry has the on-mesh flag set, it should set the prefix to the specified prefix length. Otherwise, it should set to /128.

Thanks for your explanation. Looking forward to the fix.

PR #5116 (2020-06-24): [network-data] only send ADDR_QRY.qry for on-mesh prefixes
This PR includes the following fixes.

- fix `NetworkData::Leader::IsOnMesh()` to check on-mesh flags.

- fix tests to specify on-mesh flag in prefix 3.

- fix SLAAC to set prefix length to /128 when prefix is not on mesh.

Resolves #5112
closed 2020-06-24 by jwhui