openthread/openthread #5114: [cmake] on macOS, include mbedtls-config.h causes lots of errors

jwhui, 2020-06-17:
**Describe the bug** A clear and concise description of what the bug is.

Errors printed when building on macOS.

But introduced with #5084.

**To Reproduce** Information to reproduce the behavior, including:

Platform: macOS
Command: `./script/test build`

```
[35/563] cd /Users/jonhui/jwhui/openthread/build/cmake/openthread-simulation-1.1/third_pa...ui/openthread/build/cmake/openthread-simulation-1.1/third_party/mbedtls/mbedtls-config.h"
In file included from /Users/jonhui/jwhui/openthread/third_party/mbedtls/mbedtls-config.h:36:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdio.h:64:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/_stdio.h:69:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/Availability.h:242:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:52097:2: error: #else after #else
#else
 ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:51388:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_13_4
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:51289:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_13_2
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:51194:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_13_1
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:51103:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_13
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:51016:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_12_4
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50933:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_12_2
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50854:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_12_1
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50779:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_12
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50708:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_11_4
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50641:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_11_3
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50578:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_11_2
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50519:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_11
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50464:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_10_3
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50413:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_10_2
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50366:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_10
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50323:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_9
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50284:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_8
         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/AvailabilityInternal.h:50249:10: error: unterminated conditional directive
        #if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_7
         ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
```

**Expected behavior** A clear and concise description of what you expected to happen.

No errors.

abtink, 2020-06-17:
I use `./tests/toranj/build.sh cmake` to build locally (on macbook or imac) and I get this error: 
```
In file included from src/core/common/instance.hpp:50:
In file included from src/core/common/random_manager.hpp:44:
third_party/mbedtls/repo/include/mbedtls/entropy.h:129:5: error: unknown type name 'mbedtls_sha256_context'; did you mean 'mbedtls_aes_context'?
    mbedtls_sha256_context  accumulator;
    ^~~~~~~~~~~~~~~~~~~~~~
    mbedtls_aes_context
third_party/mbedtls/repo/include/mbedtls/aes.h:99:1: note: 'mbedtls_aes_context' declared here
mbedtls_aes_context;
^
1 error generated.
```

Seems like after building with cmake the "third_party/mbedtls/mbedtls-config.h" somehow gets modified:
```
abtink@Abtins-iMac ~/GitHub/openthread (master) $ git status
On branch master
Your branch is ahead of 'origin/master' by 93 commits.
  (use "git push" to publish your local commits)

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   third_party/mbedtls/mbedtls-config.h

```
(To unblock others) ensure to `git stash` after this.

Moandor-y, 2020-06-18:
It seems to be an issue with `unifdefall`. Running
```
unifdefall -Isrc/core -Iexamples/platforms/simulation -Ietc/cmake -Iinclude -Ithird_party/mbedtls/repo/include third_party/mbedtls/mbedtls-config.h > /dev/null
```
results in the same errors. But these errors are harmless. The command output is correct. I think we can use redirections to hide these errors.

Regarding https://github.com/openthread/openthread/issues/5114#issuecomment-645659998, it is caused by in-source building. The build system will pre-process `<src>/third_party/mbedtls/mbedtls-config.h` and output to `<build>/third_party/mbedtls/mbedtls-config.h`. If `<src>` and `<build>` are the same, the input file will be truncated before processing, just like `cat mbedtls-config.h > mbedtls-config.h`. I'm considering changing the output path to somewhere like `<build>/intermediate/third_party/mbedtls/mbedtls-config.h` to prevent this.

Moandor-y, 2020-06-18:
It turns out that the `unifdef` tool shipped with macOS is too old. So changing to invoke `unifdef` only if its version is greater than or equal to `2.10`.

closed 2020-06-19 by jwhui