openthread/openthread #5121: Broadcast UDP Reliability and Performance

david-zencontrol, 2020-06-19:
Hi all.

We have been testing against a growing mesh of NRF52840 devices which are using a UDP protocol, sending commands to broadcast (FF03::1), RLOC and ML-EID addresses and in most cases we observe packets are being delivered reliably. 
However, when we send a broadcast query that generates multiple responses, we are receiving far fewer responses than expected. 

When our mesh had ~12 nodes, we were consistently receiving 10-12 responses but now that we have ~26 nodes we are still only receiving 10-12 responses and in many cases less than that. Responses are sent unicast to the source of the query, and we are evaluating using both RLOC and ML-EID as the source address. Sending unicast queries to all nodes generated a consistently higher response rate (> 20 responses).

We were initially testing against the pre-built libraries provided by Nordic in their SDK v4.1 (c6a258e3) however we observed major stalls in the Address Resolution API, where a node was seen to either not respond with an ADDR_NTF.ans (/a/an) for exactly 2 seconds, or not respond at all. When building from source (bad53a2e) the ADDR_NTF.ans responses seem to be sent correctly however the node that initiated the ADDR_QRY.qry is often seen to not then send its original UDP response. 

Once the mesh and node EID caches have stabilised, we reach our current issue of failing to receive responses.  I believe that I am not seeing the missed responses on a wireshark log, and that perhaps they are being lost in the brief period of high network traffic. We have increased our otLinkSetMaxFrameRetriesDirect to 8 and are contemplating 15, however we are concerned that further increasing network traffic will have adverse effects. 

Is what we are observing the expected behaviour for a thread mesh, or is there a issue that we are encountering?  If we are saturating the bandwidth of the mesh, is there a quick way for a node to detect this before scheduling a transmission to spread out the responses? We are evaluating sending mass unicast messages in lieu of broadcast but we are concerned that this approach may also not scale when we reach meshes with ~100 nodes.

Thanks for your time

jwhui, 2020-06-19:
@david-zencontrol , thanks for raising this issue. I think this is worth some additional investigation. Some initial questions:

1. Can you provide more information about your node topology? Are all the devices within communication range / single-hop? Or are the devices spread out over multiple hops?

1. What is the size of your address cache in your build (i.e. value of `OPENTHREAD_CONFIG_TMF_ADDRESS_CACHE_ENTRIES`)?

1. You mentioned testing with both RLOC and ML-EID as source. Are you seeing the same performance with both? Or is one performing better than the other? I would expect RLOC to perform better since it does not require an RLOC-to-EID lookup on the response.

1. Are you able to enable logging and provide the logs? It would be interesting to see whether any messages are getting dropped. For example, this log method:
https://github.com/openthread/openthread/blob/7574cde26484b8cea43f3ac64243395986c22521/src/core/thread/mesh_forwarder.cpp#L1540

david-zencontrol, 2020-06-19:
@jwhui thanks for your quick response.

1. 
This is the current graphical representation of our mesh. I would say most devices can see all other devices and are maintaining a large number of neighbors. I have spread them out as much as I can given our office, but this layout would be representative of our use case. 

[image]

The router table looks like:
`
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
|  0 | 0x0000 |       63 |         0 |     0 |      0 |  13 | 924152cb15aec292 |
|  4 | 0x1000 |       18 |         1 |     2 |      2 |  57 | b6a8831766357642 |
|  5 | 0x1400 |       18 |         1 |     2 |      2 |  24 | ce4242237119e9da |
|  7 | 0x1c00 |       18 |         1 |     3 |      3 |  14 | 666b20d986f5ecb1 |
|  8 | 0x2000 |       18 |         1 |     3 |      3 |  16 | f2619988b7f2c645 |
| 18 | 0x4800 |       28 |         1 |     3 |      3 |  16 | ea08c76bef3f5da0 |
| 23 | 0x5c00 |       28 |         1 |     3 |      3 |  20 | 5e5f8f01cec2f8d5 |
| 28 | 0x7000 |       18 |         1 |     3 |      3 |  14 | 127f8dafbdd6cf04 |
| 31 | 0x7c00 |       18 |         1 |     2 |      2 |  57 | 0ad584ca40b6423d |
| 34 | 0x8800 |       18 |         1 |     2 |      2 |  21 | 6e7213ab4c1f37e9 |
| 36 | 0x9000 |       28 |         1 |     3 |      3 |  27 | 668cfc264f357e39 |
| 40 | 0xa000 |       18 |         2 |     0 |      0 | 241 | 0000000000000000 |
| 50 | 0xc800 |       18 |         1 |     3 |      3 |  17 | 065015143a79c82d |
| 51 | 0xcc00 |       18 |         1 |     1 |      0 | 108 | c2c8a06a16097e0c |
| 54 | 0xd800 |       18 |         1 |     2 |      1 |  13 | bedb8b5fc21955fe |
| 58 | 0xe800 |       18 |         1 |     2 |      1 |  14 | 3a1689af00c13103 |
`

2. Cache size is 32, and we will be looking to increase this as part of our scale up.

3. Initially the RLOC is far more reliable as it can't cache miss, however once a devices cache is populated it seems comparable. Am I correct in my understanding that child nodes are more likely to time out of the cache in a way that routers do not?  

4. I will have to work on logging and SWD on this platform, however in the meantime perhaps the mac counters may be useful after sending 3x queries and responding 7,12,12 replies: 
[image]

david-zencontrol, 2020-06-19:
Attached is a wireshark of our node connecting to our mesh and sending the aforementioned 3x queries at pkt 244, 493, 564. We are (ipv6.src == ::1603:9b95:739d:94fd) and receive 9, 12, 12 responses respectively. 
[9_12_12.pcapng.zip](https://github.com/openthread/openthread/files/4802603/9_12_12.pcapng.zip)

david-zencontrol, 2020-06-19:
Here is a log of our device sending queries and receiving responses. Queries that resulted in 'Failed to send IPv6 UDP msg' logs still received ~12 replies.  

`
SEGGER J-Link V6.60c - Real time terminal output
J-Link OB-SAM3U128-V2-NordicSemi compiled Jan  7 2019 14:07:15 V1.0, SN=683160824
Process: JLinkGDBServerCLExe
[0000000454] [NOTE]-MLE-----: Role Disabled -> Detached
[0000000520] [WARN]-MLE-----: Failed to process Child Update Response: Security
[0000000525] [NOTE]-MLE-----: Role Detached -> Child
[0000000525] [WARN]-MLE-----: Failed to process Child Update Response: Drop
[0000000454] [NOTE]-MLE-----: Role Disabled -> Detached
[0000000864] [WARN]-MLE-----: Failed to process Child Update Response: Security
[0000000887] [NOTE]-MLE-----: Role Detached -> Child
[0000000887] [WARN]-MLE-----: Failed to process Child Update Response: Drop
[0000001462] [NOTE]-MLE-----: Different partition (peer:1830878328, local:0)
[0000001463] [WARN]-MLE-----: Failed to process Advertisement: LinkMarginLow
[000000190[0000020417] [WARN]-MLE-----: Failed to process Advertisement: NoRoute
[0000020999] [WARN]-MLE-----: Failed to process Advertisement: NoRoute
[0000025345] [WARN]-MLE-----: Failed to process Advertisement: NoRoute
[0000047983] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:e3dc:f008:feb6:6507, 0x4800 - snoop
[0000048094] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:f1fe:cb64:1aec:29b2, 0x1000 - snoop
[0000048281] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:87bd:c15b:dbe2:4bd0, 0x9001 - snoop
[0000048349] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:aef6:5f7e:81d0:9257, 0xc800 - snoop
[0000053112] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:85f8:9390:f470:3989, 0xe800 - snoop
[0000053237] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:9283:887f:38e6:1f2b, 0x9000 - snoop
[0000053344] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:78ad:d39e:fb0b:1dec, 0x2000 - snoop
[0000053486] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:aea7:4204:df5d:1e5f, 0x0401 - snoop
[0000053512] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:8878:5675:e62c:1b70, 0x9002 - snoop
[0000053535] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:8f16:a488:113:8d28, 0xc805 - snoop
[0000057627] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:3a5a:c7ba:5100:b942, 0x7c00 - snoop
[0000057643] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:5ba1:9107:1706:606a, 0x0400 - snoop
[0000061578] [NOTE]-MAC-----: Failed to send IPv6 UDP msg, len:90, chksum:6e61, to:0x4800, sec:yes, error:ChannelAcc[0000061607] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:d178:4816:7145:f97f, 0xec00 - snoop
[0000061732] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:a4c7:f9cd:5e76:84a2, 0x4840 - snoop
[0000063937] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:8afc:e692:2878:1d6e, 0xe803 - snoop
[0000075916] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:d1e0:d4f7:fc6c:675f, 0x5c00 - snoop
[0000075926] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:11f7:9442:5f48:35da, 0xec01 - snoop
[0000106865] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:579a:2e72:b3d:53cb, 0x484a - snoop
[0000113743] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:9234:ca54:ca61:22e5, 0xd800 - snoop
[0000195930] [WARN]-MLE-----: Failed to process Advertisement: NoRoute
[0000396378] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:56a6:4381:d883:5236, 0x4832 - snoop
[0000396415] [NOTE]-ARP-----: Cache entry added: fdcc:a23f:97:fb0d:8e82:5d82:abdd:5c6, 0x8800 - snoop
`

jwhui, 2020-06-20:
@david-zencontrol , thanks for providing additional data.

The MAC counters show a non-zero value for `mTxDirectMaxRetryExpiry`, which indicates frames are getting dropped at the MAC layer due to failed CCA or ACKs. I would suggest raising the max retry value to 15 - this is what we ship in our products today. Note that raising to 15 does not actually mean 15 real transmission on the air, since failures may be due to CCA.

In general, when scaling up, I think it would be a good idea to add some random delay when responding to a multicast request. Thread employs this technique in its control protocols, including both MLE and TMF, even for single-hop communication.

For the logs, it would help if you raise the debug level to DEBG and enable the MAC log region.

Hope that helps.

david-zencontrol, 2020-06-22:
@jwhui 
Thanks for your feedback. I implemented a delay which factors in the RLOC and the current CCA Failure rate to better distribute the responses and our response rate has increased significantly. Tuning will be needed as our mesh scales up.

Is there a standardised or recommended algorithm for calculating such a distributed delay, or will it be left to to application layers to implement their own delays?

jwhui, 2020-06-30:
@david-zencontrol , the typical solution involves having a random delay. Ideally, make the random window as long as you can tolerate the latency. The challenge is that the overall capacity depends on a number of factors, including topology, other traffic loads, and external environmental factors as well. You may consider using some kind of message transport that includes reliability, congestion control, and flow control. The benefit of such approaches is that they try to adapt to the underlying network capacity, rather than taking a conservative guess.

closed 2020-08-11 by jwhui