openthread/openthread #5133: posix DAEMON=1 build failed on macOS

simonlingoogle, 2020-06-20:
**Describe the bug**
Build on macOS failed with DAEMON=1.

**To Reproduce**

1. Git commit id: 0cf84da1006d70ac0a3a817406d702a1132eab63
2. IEEE 802.15.4 hardware platform: macOS, posix
3. Build steps
```bash
./bootstrap
make -f src/posix/Makefile-posix DAEMON=1
```

**Expected behavior**
Build successfully.

**Console/log output**
```bash
  CXX      libopenthread_posix_a-udp.o
  CXX      libopenthread_posix_a-virtual_time.o
/Users/simonlin/src/openthread/src/posix/../../src/posix/platform/udp.cpp:86:28: error: variable length arrays are a C99 feature [-Werror,-Wvla-extension]
                    control[OT_APPLE_IGNORE_GNU_FOLDING_CONSTANT(CMSG_SPACE(sizeof(struct in6_pktinfo)) + CMSG_SPACE(sizeof(int)))];
                           ^
/Users/simonlin/src/openthread/src/posix/../../src/posix/platform/udp.cpp:113:26: error: implicit conversion loses integer precision: 'unsigned long' to 'socklen_t' (aka 'unsigned int')
      [-Werror,-Wshorten-64-to-32]
    msg.msg_controllen = sizeof(control);
                       ~ ^~~~~~~~~~~~~~~
2 errors generated.
```

marvinroger, 2020-06-28:
I fixed a similar issues two weeks ago: https://github.com/openthread/openthread/pull/5101
I know nothing about C/C++, but isn't there an issue with the build flags on macOS? Why would a macOS build be stricter than a Linux one?

jwhui, 2020-06-29:
> I fixed a similar issues two weeks ago: #5101
> I know nothing about C/C++, but isn't there an issue with the build flags on macOS? Why would a macOS build be stricter than a Linux one?

@marvinroger , I think a couple issues are contributing to the differences:

1. Different compilers (e.g. clang vs. gcc).

1. Different platforms (e.g. macOS vs. Linux) may have different definitions (e.g. different networking-related structs).

When we find issues like this slip through, we should certainly plug the holes in our continuous integration.

PR #5839 (2020-12-15): [posix] fix build warnings/errors on macOS
fixes #5133
closed 2020-12-15 by jwhui