openthread/openthread #5138: Openthread Sleepy End Device Reboot when parent connection is lost.

bhumikaajain, 2020-06-22:
I have an openthread set-up with Rpi based Border Router and 2 SED nodes and 1 FTD node.
The set-up is such that, 1 SED is placed near to Border Router so it's always the direct neighbour and child of Border Router and another SED is placed far such that it connects to Border Router with 1 hop via FTD.
Set-up runs perfect for days if all devices are on.
Now when FTD is powered off for 7-8 hours, 2nd SED, which is direct child of FTD, gets rebooted once FTD is powered on again. Sometimes SED also backs off from the network.

My application is build over Nordic's NRF SDK for Thread and Zigbee Version 4.0.0.
Hardware used: nrf52840 SoC
Daemon Version: 0.08.00d (/345bf7d; Jun 27 2019 17:02:59)
NCP Version: "OPENTHREAD/20191113-00092-g8a1992e2; NRF52840; Apr  6 2020 22:33:55"

jwhui, 2020-06-22:
@bhumikaajain , thanks for raising this issue.

Can you indicate how you determined that the SED is "rebooting"? Does the device have any watchdog enabled by default?

/cc/ @LuDuda

bhumikaajain, 2020-06-23:
Yes, we have watchdog enabled by default.

jwhui, 2020-06-23:
> Yes, we have watchdog enabled by default.

@bhumikaajain , would it be possible for you to hook up a debugger and determine what is causing the watchdog reset?

bhumikaajain, 2020-06-24:
It is not feasible as the reset time is not fixed and I am using my SEDs as sensor nodes, remotely placed.

jwhui, 2020-06-24:
> It is not feasible as the reset time is not fixed and I am using my SEDs as sensor nodes, remotely placed.

@bhumikaajain , unfortunately, we have not yet experienced any similar behavior. If it is possible to replicate your setup, a debugger will help us identify the issue. If you disable the watchdog, spin-loops and asserts are easy to identify.

jwhui, 2020-07-15:
Closing stale issue. Please reopen if there are further updates.

closed 2020-07-15 by jwhui