openthread/openthread #5142: [meshcop] Nordic devices cannot join with external commissioner

wgtdkp, 2020-06-23:
**Describe the bug** A clear and concise description of what the bug is.

Thread Commissioning with nrf52840 devices failed because OTBR cannot forward `Client Hello` sent by the joiner.

**To Reproduce** Information to reproduce the behavior, including:
Follow the guide https://openthread.io/guides/border-router/external-commissioning?comm=ot-commissioner.

1. Git commit id: openthread master HEAD
2. IEEE 802.15.4 hardware platform: nrf52840
3. Build steps
4. Network topology: External Commissioner <-----> OTBR <------> Joiner

**Expected behavior** A clear and concise description of what you expected to happen.
The joining process shall success and "Join success" is printed by the joiner.

**Console/log output** If applicable, add console/log output to help explain your problem.
- The joining process failed with `Security` error:
    ```text
    > Join failed [Security]
    ```
- in OTBR log, `NoBufs` error is reported:
    ```text
    otbr-agent[8408]: [INFO]-PLAT----: processReceive: OK
    otbr-agent[8408]: [INFO]-MESH-CP-: JoinerRouter::HandleUdpReceive
    otbr-agent[8408]: [INFO]-MESH-CP-: Sent relay rx
    otbr-agent[8408]: [INFO]-PLAT----: processTransmit: OK
    otbr-agent[8408]: [INFO]-MESH-CP-: Sent to commissioner
    otbr-agent[8408]: [INFO]-MESH-CP-: Sent to commissioner on c/rx
    otbr-agent[8408]: [NOTE]-MESH-CP-: CoapSecure Transmit: NoBufs
    ```

**Additional context** Add any other context about the problem here.
The same commissioning process is executed with simulated devices (OTBR and Joiner) and it works well.  The traffic of the nrf52840 case shows that it is keep retransmiting the `Client Hello` message when including cookie. This is because the OTBR failed to forward this `Client Hello` message to the external commissioner (the `NoBufs` error). The `Client Hello` message is too long to be encoded within the `RLY_RX` message with a maximum length of `kApplicationDataMaxLength = 512` defined at:
https://github.com/openthread/openthread/blob/5c722a770c6f3a0627c93ec5ae3bb3e13e491bbe/src/core/meshcop/dtls.hpp#L74-L76

The simulated joiner works because it includes less support ECP group extension in `Client Hello` message:

`Client Hello` sent by nrf52840 joiner:
```text
Extension: supported_groups (len=20)
    Type: supported_groups (10)
    Length: 20
    Supported Groups List Length: 18
    Supported Groups (9 groups)
        Supported Group: secp521r1 (0x0019)
        Supported Group: secp384r1 (0x0018)
        Supported Group: secp256r1 (0x0017)
        Supported Group: secp256k1 (0x0016)
        Supported Group: brainpoolP256r1 (0x001a)
        Supported Group: secp224r1 (0x0015)
        Supported Group: secp224k1 (0x0014)
        Supported Group: secp192r1 (0x0013)
        Supported Group: secp192k1 (0x0012)
```

CLIENT HELLO sent by simulated joiner:
```text
Extension: supported_groups (len=4)
    Type: supported_groups (10)
    Length: 4
    Supported Groups List Length: 2
    Supported Groups (1 group)
        Supported Group: secp256r1 (0x0017)
```

The `Client Hello` extensions sent by the simulated joiner is expected because only `SECP256R1` is enabled in builtin mbedtls:
https://github.com/openthread/openthread/blob/5c722a770c6f3a0627c93ec5ae3bb3e13e491bbe/third_party/mbedtls/mbedtls-config.h#L54-L58

While nrf52840 uses nordic-mbedtls and it is only `SECP256R1` that is enabled in nordic-mbedtls:
https://github.com/openthread/openthread/blob/5c722a770c6f3a0627c93ec5ae3bb3e13e491bbe/third_party/NordicSemiconductor/libraries/nrf_security/config/nrf-config.h#L648-L668

This doesn't explain the captured traffic sent by the nrf52840 joiner. **One possible reason is that the [libraries](https://github.com/openthread/openthread/tree/master/third_party/NordicSemiconductor/libraries/nrf_security/lib) provided by nordic-mbedtls doesn't match the headers.** @LuDuda Could you help verify this?

Captured traffic:
[failed-nrf52840-external-commissioner.zip](https://github.com/openthread/openthread/files/4818193/failed-nrf52840-external-commissioner.zip)
[success-simulated-external-commissioner.zip](https://github.com/openthread/openthread/files/4818194/success-simulated-external-commissioner.zip)

LuDuda, 2020-06-23:
Hello @wgtdkp indeed we are aware about sending additional curves in Client Hello which we introduced with latest nrf_security library - and was tested against ot-commissioner. This is needed for next-higher layer. 

@piotrkoziar @edmont could you please take a look on this potential inconsistency in header files?

LuDuda, 2020-06-23:
@wgtdkp FYI, in our SDK we always keep CoAP Secure enabled which then increases the maximum datagram size to 1400 from 512.

wgtdkp, 2020-06-23:
> @wgtdkp FYI, in our SDK we always keep CoAP Secure enabled which then increases the maximum datagram size to 1400 from 512.

@LuDuda yes, enable `COAP_SECURE_API` works, but it seems the nrf52840 example in openthread repo is not built with this option by defult?

LuDuda, 2020-07-01:
Just the update @piotrkoziar works on this issue.

piotrkoziar, 2020-07-03:
Update: the inconsistency will be removed in the new nrf_security library. It may take some time - cooperation with some other team is required.

wgtdkp, 2020-07-03:
> Update: the inconsistency will be removed in the new nrf_security library. It may take some time - cooperation with some other team is required.

Thanks for the update.

piotrkoziar, 2020-07-03:
@wgtdkp It seems like there is an easier way - no need to update nrf_security. PR will be introduced soon.

wgtdkp, 2020-07-18:
@LuDuda @piotrkoziar @jwhui I think there is inconsistency between the nrf_security lib and headers, should we keep this issue open untill new nrf_security library is updated?

jwhui, 2020-07-20:
> I think there is inconsistency between the nrf_security lib and headers, should we keep this issue open untill the nrf_security library is updated?

@LuDuda @edmont , can you help answer the question above?

piotrkoziar, 2020-07-20:
@wgtdkp @jwhui I created #5265.
I think we can close the issue when #5265 is merged.

jwhui, 2020-07-20:
> @wgtdkp @jwhui I created #5265.
> I think we can close the issue when #5265 is merged.

Thanks!

PR #5265 (2020-07-22): [nrf_security] remove inconsistency from nrf_security config file
To avoid confusion, the additional supported in nrf_security library curves are now stated in the nrf-config.h.

Resolves #5142
closed 2020-07-22 by jwhui