openthread/openthread #5146: Network Time Sync, what does the offset represent and how to apply across nodes

firehills, 2020-06-24:
I’m trying to understand how Network Time Sync is designed to operate, what the offset represents, and how to correctly apply the offset across multiple nodes in order to synchronize time within the mesh. I am using some nrf52840 nodes as examples. 

The offset (internally TimeSync member mNetworkTimeOffset) is distributed across the network to all nodes. This gives a signed uS offset to apply to a node’s local/platform time when calling otNetworkTimeGet(). Over a given period the value is constant if the source does not differ by >OPENTHREAD_CONFIG_TIME_SYNC_JUMP_NOTIF_MIN_US  

If each node however has a different local time (simplest case just powered on with a few seconds between them) how is the shared offset value intended to resolve the differing time deltas? In an ideal world if all nodes call otNetworkTimeGet() at the same instance, it would return the same (synced) value. 

Is the intention that Time Sync enabled nodes require an RTC and/or perhaps an SNTP call initially resolves the local time offsets, and this gives a common base, in which case the offset would represent the local clock drift of the Leader? Even then local drift will differ between nodes?

Thanks in advance.

jwhui, 2020-06-24:
@firehills , thanks for raising this question.

Note that OpenThread's network time synchronization service is an experimental feature that is not included as part of the Thread Specification.

As for the implementation itself, the local time is simply a free-running clock local to the device. `mNetworkTimeOffset` represents the time offset between the local time and the network time. Given that the local time is local to each device, `mNetworkTimeOffset` will also differ for each device. The network time sync service does not require the local time on each device to be synchronized.

Hope that helps.

closed 2020-06-30 by jwhui

farhangj, 2021-07-12:
@jwhui  I tried to enable the time sync feature, using Nordic nRF NCS 1.6.0 - I have 1 router (FTD) in my test and 4 children (MTD/SED), I have CONFIG_OPENTHREAD_TIME_SYNC=y in all projects.  Using ot version = 1.1.
0- Understood that it is experimental, does it work or should I implement my own time sync feature? when is the expected release date for this?
1- if it works, I don't understand how I am supposed to set the time on the router. There seems to be no API for that.
2- The feature seems to get in the way of regular SED polls/check-ins with the router. e.g. i have poll period=5000ms on the SEDs, when I enable time sync, the SEDs are often in detached state and some security errors appear on my router. I wonder if time_sync is incompatible with CONFIG_OPENTHREAD_NRF_SECURITY_CHOICE.

jwhui, 2021-07-12:
> @jwhui I tried to enable the time sync feature, using Nordic nRF NCS 1.6.0 - I have 1 router (FTD) in my test and 4 children (MTD/SED), I have CONFIG_OPENTHREAD_TIME_SYNC=y in all projects. Using ot [...]

The time sync feature is not currently on any roadmap. It is not a part of the Thread Specification.

> 1- if it works, I don't understand how I am supposed to set the time on the router. There seems to be no API for that.

See [include/openthread/network_time.h](https://github.com/openthread/openthread/blob/main/include/openthread/network_time.h).

> 2- The feature seems to get in the way of regular SED polls/check-ins with the router. e.g. i have poll period=5000ms on the SEDs, when I enable time sync, the SEDs are often in detached state and [...]

The Time Sync feature should be compatible with SED data polling. However, I have not validated this feature in quite some time.