openthread/openthread #5149: ./script/bootstrap fails on Ubuntu 18.04 (64 bit)

iDarshan, 2020-06-25:
I am trying to build firmware for nrf52840 on Ubuntu 18.04 machine. I executed `./script/bootstrap` as a first step and it failed with following error:

```
  Using cached https://files.pythonhosted.org/packages/b2/34/bfcb43cc0ba81f527bc4f40ef41ba2ff4080e047acb0586b56b3d017ace4/zipp-3.1.0-py3-none-any.whl
Installing collected packages: zipp, importlib-metadata, markdown, tabulate, pygments, mdv
Successfully installed importlib-metadata-1.6.1 markdown-3.2.2 mdv-1.7.4 pygments-2.6.1 tabulate-0.8.7 zipp-3.1.0
./script/bootstrap: 69: ./script/bootstrap: brew: not found
Failed to install shfmt. Install it manually if you need.
darshan@Ubuntu18:~/OT/openthread$
```

It's searching for bew on Ubuntu! Seems like script is for Mac OS. What can be done to fix this?

bukepo, 2020-06-25:
Yes, brew also works on Linux. If not installed, you will have to install shfmt manually. You can find it on GitHub.

iDarshan, 2020-06-25:
But why to have brew when aptutils already there? Some packages already getting installed with apt, why some required to have brew?

jwhui, 2020-06-25:
@iDarshan , note that only the install of `shfmt` failed and `shfmt` is only needed if you plan to contribute shell scripts to the OpenThread project.

You can also install `shfmt` using `snap`: https://snapcraft.io/install/shfmt/ubuntu

I'm not aware of an official apt package for `shfmt`. Let us know if you can find one.

iDarshan, 2020-06-25:
@jwhui That means ./script/bootstrap is successfull and I can move forward with the build. I tried to build nrf52840 firmware and it fails with following errors (only partial log pasted). Seems like something broken in latest master branch.

```
/home/darshan/OT/openthread/examples/../src/core/common/debug.hpp:79:15: note: in definition of macro 'OT_ASSERT'
         if (!(cond))    \
               ^~~~
In file included from /home/darshan/OT/openthread/examples/../src/core/thread/thread_netif.hpp:40:0,
                 from /home/darshan/OT/openthread/examples/../src/core/common/instance.hpp:73,
                 from /home/darshan/OT/openthread/examples/../src/core/api/channel_manager_api.cpp:37:
/home/darshan/OT/openthread/examples/../src/core/coap/coap_secure.hpp:265:71: error: 'nullptr' was not declared in this scope
     otError SendMessage(Message &aMessage, ResponseHandler aHandler = nullptr, void *aContext = nullptr);
                                                                       ^~~~~~~
/home/darshan/OT/openthread/examples/../src/core/coap/coap_secure.hpp:265:97: error: 'nullptr' was not declared in this scope
     otError SendMessage(Message &aMessage, ResponseHandler aHandler = nullptr, void *aContext = nullptr);
                                                                                                 ^~~~~~~
/home/darshan/OT/openthread/examples/../src/core/coap/coap_secure.hpp:286:60: error: 'nullptr' was not declared in this scope
                         ResponseHandler         aHandler = nullptr,
                                                            ^~~~~~~
/home/darshan/OT/openthread/examples/../src/core/coap/coap_secure.hpp:287:60: error: 'nullptr' was not declared in this scope
                         void *                  aContext = nullptr);
                                                            ^~~~~~~
cc1plus: all warnings being treated as errors
In file included from /home/darshan/OT/openthread/examples/../src/core/meshcop/dataset_manager.hpp:44:0,
                 from /home/darshan/OT/openthread/examples/../src/core/thread/thread_netif.hpp:61,
                 from /home/darshan/OT/openthread/examples/../src/core/common/instance.hpp:73,
                 from /home/darshan/OT/openthread/examples/../src/core/api/channel_manager_api.cpp:37:
/home/darshan/OT/openthread/examples/../src/core/meshcop/dataset.hpp:332:81: error: 'nullptr' was not declared in this scope
     otError ApplyConfiguration(Instance &aInstance, bool *aIsMasterKeyUpdated = nullptr) const;
                                                                                 ^~~~~~~
Makefile:4252: recipe for target 'api/libopenthread_mtd_a-border_router_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-border_router_api.o] Error 1
make[4]: *** Waiting for unfinished jobs....
In file included from /home/darshan/OT/openthread/examples/../src/core/thread/thread_netif.hpp:76:0,
                 from /home/darshan/OT/openthread/examples/../src/core/common/instance.hpp:73,
                 from /home/darshan/OT/openthread/examples/../src/core/api/channel_manager_api.cpp:37:
/home/darshan/OT/openthread/examples/../src/core/thread/address_resolver.hpp:309:56: error: 'nullptr' was not declared in this scope
                              CacheEntryList *  aList = nullptr);
                                                        ^~~~~~~
In file included from /home/darshan/OT/openthread/examples/../src/core/thread/thread_netif.hpp:89:0,
                 from /home/darshan/OT/openthread/examples/../src/core/common/instance.hpp:73,
                 from /home/darshan/OT/openthread/examples/../src/core/api/channel_manager_api.cpp:37:
/home/darshan/OT/openthread/examples/../src/core/utils/child_supervision.hpp: In member function 'ot::Child* ot::Utils::ChildSupervisor::GetDestination(const ot::Message&) const':
/home/darshan/OT/openthread/examples/../src/core/utils/child_supervision.hpp:181:61: error: 'nullptr' was not declared in this scope
     Child *  GetDestination(const Message &) const { return nullptr; }
                                                             ^~~~~~~
cc1plus: all warnings being treated as errors
Makefile:4266: recipe for target 'api/libopenthread_mtd_a-channel_manager_api.o' failed
make[4]: *** [api/libopenthread_mtd_a-channel_manager_api.o] Error 1
Makefile:433: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:586: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:512: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-nrf52840:329: recipe for target 'do-build-cortex-m4' failed
make: *** [do-build-cortex-m4] Error 2
darshan@Ubuntu18:~/OT/openthread$
```

jwhui, 2020-06-25:
@iDarshan , it seems you may have some stale build files that need to be regenerated. Can you try the following:

```bash
git clean -xfd
./bootstrap
make -f examples/Makefile-nrf52840
```

Note that the `git clean` command will blow away any changes you have in your tree.

iDarshan, 2020-06-28:
@jwhui That worked. git clean was not needed though. Please add in README that in Linux if script/bootstrap shows error due to brew for shfmt, its ok to move ahead with build.

jwhui, 2020-06-29:
> @jwhui That worked. git clean was not needed though. Please add in README that in Linux if script/bootstrap shows error due to brew for shfmt, its ok to move ahead with build.

Submitted #5161. I opted to improve the output from the script itself in this PR. Please review.

PR #5161 (2020-06-30): [bootstrap] update bootstrap user messages
- Indicate that pretty tools are primarily used for code contributions.

- Change to bash.

Resolves #5149
closed 2020-06-30 by jwhui