openthread/openthread #5152: Router losing its role after reset when LONG_ROUTES is enabled

mwilniewiec, 2020-06-25:
My setup: 20x NRF52840 hardware nodes with LONG_ROUTES enabled (still Nordic's SDK 2.0: 41d8e4a with multiple different patches)
Note: This cannot be reproduced with 10 nodes.

1. enable LONG_ROUTES
2. take 20 nodes
3. take node N, make sure it is a router
4. reset node N
5. node N will be a child after startup (instead of keeping previous router RLOC)

I investigated the problem a bit.
1. Node N wakes up and sends Link Request to all routers as expected.
2. Neighbor routers answer with Link Accept providing Address16 TLV as expected
3. Because of big Route64 TLV (Length = 33) whole Link Accept does not fit into one message and is splitted into 2 messages
4. Node N do not receive those messages properly ( drops them ?, Link Accept is not called )

[image]

mwilniewiec, 2020-06-26:
I reproduced the issue on recent openthread from master on POSIX with default settings (no LONG_ROUTES).

1. Started ~30 posix nodes
2. Set router downgradethreshold to 32
3. state router on all nodes
4. Ive got 27 routers in my router table
5. reset any router, ifconfig up + thread start, check state - it is a child instead of comming back to router state

In that case Link Accept is Fragmented 
```
[Frame: 5185, payload: 0-127 (128 bytes)]
[Frame: 5187, payload: 128-148 (21 bytes)]
[Message fragment count: 2]
[Reassembled 6LoWPAN length: 149]
```
Route64 TLV Length: 36

jwhui, 2020-06-29:
@mwilniewiec , thanks for raising this issue.

It sounds like you have already found the root cause. The current Thread specification requires all MLE messages that are fragmented to be protected using IEEE 802.15.4 MAC frame security. Of course, when processing received frames with IEEE 802.15.4 MAC frame security, a link must already be established with the neighboring node. Thread requires enabling 802.15.4 MAC frame security to mitigate possible DoS attacks that can occur where an attacker might transmit frames indicating a large IPv6 datagram, causing the receiver to allocate memory. This causes a chicken-and-egg problem, where receiving the Link Accept is necessary to establish the link security itself.

Given that the "long routes" feature is experimental and not a part of the standard Thread specification, one path forward is to special case the scenario and possibly ignore 802.15.4 MAC Frame security when coming out of a reset. Another, more involved, possibility would require changing the MLE protocol to not send the Route64 TLV in the Link Accept and update the Route64 TLV after initially establishing the link security and further MLE message exchange.

mwilniewiec, 2020-06-30:
```
I reproduced the issue on recent openthread from master on POSIX with default settings (no LONG_ROUTES).
```
As I wrote in the second message, you actually don't need LONG_ROUTES to reproduce the issue. It's enough to have around 27 routers in the network, in that case Link Accept is also too big.

jwhui, 2020-06-30:
@mwilniewiec , thanks again for identifying and root-causing this issue.

Addressing this will require a change to the Thread Specification. I have submitted [SPEC-944](https://threadgroup.atlassian.net/browse/SPEC-944) to start the errata process within the Thread Group.

jwhui, 2020-07-17:
@mwilniewiec , I submitted one proposed solution with #5256. Please review.

jwhui, 2020-08-03:
@mwilniewiec , any thoughts on the proposed solution?

PR #5256 (2020-08-11): [mle] work around MLE Link Accept messages that are too large
This commit addresses an issue where the MLE Link Accept message may
be too large to fit within a single IEEE 802.15.4 frame. Thread
requires any MLE message transported using 6LoWPAN fragmentation to
also enable IEEE 802.15.4 MAC frame security. However, the purpose of
MLE Link Accept messages is to establish a secure link and cannot
itself be transported with IEEE 802.15.4 MAC frame security enabled.

An MLE Link Accept message may carry a Route64 TLV that has a length
dependent on the number of active routers in the network. When [... cut]
closed 2020-08-11 by jwhui