openthread/openthread #5153: [docs] description conflict in Thread Primer (en)

dsyx, 2020-06-26:
`A Minimal Thread Device does not subscribe to multicast traffic and forwards all messages to its Parent.` in [doc/site/en/guides/thread-primer/node-roles-and-types.md](https://github.com/openthread/openthread/blob/master/doc/site/en/guides/thread-primer/node-roles-and-types.md#minimal-thread-device)

and

`There is an additional unicast prefix-based multicast address used for All Thread Nodes, including SEDs.` in [doc/site/en/guides/thread-primer/ipv6-addressing.md](https://github.com/openthread/openthread/blob/master/doc/site/en/guides/thread-primer/ipv6-addressing.md#multicast)

These two sentences should be mutually exclusive.

jwhui, 2020-06-26:
As noted in https://github.com/openthread/openthread/pull/5106#discussion_r445141120:

I think the intended text is something like: "A Minimal Thread Device does not subscribe to the all-routers multicast address and forwards all messages to its Parent."

dsyx, 2020-06-27:
The multicast address described in the second sentence should be realm-local all-nodes multicast address (FF03::1). right?

jwhui, 2020-06-27:
> The multicast address described in the second sentence should be realm-local all-nodes multicast address (FF03::1). right?

For multicast, SEDs only implicitly subscribe to the All Thread Nodes multicast address (which is a unicast prefix-based multicast address). `ff03::1` cannot be used to address SEDs, unless the SED explicitly registers that address with its parent.

dsyx, 2020-06-27:
So, how should readers understand the second sentence?

jwhui, 2020-06-27:
> So, how should readers understand the second sentence?

I think you need to see the whole paragraph to make sense of it:

> You might notice that Sleepy End Devices (SEDs) are not included as a recipient in the multicast table above. There is an additional unicast prefix-based multicast address used for All Thread Nodes, including SEDs. This multicast address varies by Thread network, because it is built on the unicast Mesh-Local prefix.

The first sentence is saying that you cannot use `ff02::1`, `ff02::2`, `ff03::1`, and `ff03::2` to address SEDs, which is true in the normal case.

I agree that the following is an incorrect statement:
> A Minimal Thread Device does not subscribe to multicast traffic and forwards all messages to its Parent.

As I previously proposed, I think it is more proper to say that:
> A Minimal Thread Device does not subscribe to the all-routers multicast address and forwards all messages to its Parent.

Note that a non-sleepy MTD (i.e. Minimal End Device) is capable of receiving multicast for `ff02::1` and `ff03::`. However, a sleepy MTD (i.e. Sleepy End Device) is normally not capable of receiving such multicast messages.

dsyx, 2020-06-27:
```
There is an additional unicast prefix-based multicast address used for All Thread Nodes, including SEDs.
```

I mean: readers will know that SEDs will subscribe to an additional unicast prefix-based multicast address that all Thread nodes will subscribe to. But readers will want to know the name/type of this multicast address.

jwhui, 2020-06-27:
> ```
> There is an additional unicast prefix-based multicast address used for All Thread Nodes, including SEDs.
> ```
>
> I mean: readers will know that SEDs will subscribe to an additional unicast [...]

The name is "All Thread Nodes" and type is "unicast prefix-based multicast". I'm not sure what you are looking for - can you clarify or propose some alternative text?

dsyx, 2020-06-28:
I think the original text should mean: These IPv6 addresses (listed in the table) cannot be used to address SEDs, but SEDs can subscribe to these addresses.

According to the description of the Thread specification, there are two types of all-node multicast addresses: Link-Local and Realm-Local.

5.2.3.1 Link-Local Scope:

```
All Thread interfaces MUST subscribe to the Link-Local All Nodes multicast address (FF02::1).
```

5.2.3.2 Realm-Local Scope:

```
All Thread interfaces MUST subscribe to the realm-local all-nodes multicast address (FF03::1).
```

So the original text should be: `There are two additional unicast prefix-based multicast address used for All Thread Nodes.`

Right?

jwhui, 2020-06-28:
> I think the original text should mean: These IPv6 addresses (listed in the table) cannot be used to address SEDs, but SEDs can subscribe to additional multicast addresses (FF02::1 and FF03::1).

While it is technically possible, I don't think SEDs should subscribe to ff02::1 and ff03::1. I can't think of a use case where that would make practical sense. The original text states this with: "You might notice that Sleepy End Devices (SEDs) are not included as a recipient in the multicast table above."

> According to the description of the Thread specification, there are two types of all-node multicast addresses: Link-Local and Realm-Local.
>
> 5.2.3.1 Link-Local Scope:
>
> ```
> All Thread interfaces MUST [...]

I think this is close to the original text.  So how about the following modification:

BEFORE:
```
You might notice that Sleepy End Devices (SEDs) are not included as a recipient in the multicast table above. There is an additional unicast prefix-based multicast address used for All Thread Nodes, including SEDs. This multicast address varies by Thread network, because it is built on the unicast Mesh-Local prefix.
```

AFTER:
```
You might notice that Sleepy End Devices (SEDs) are not included as a recipient in the multicast table above. Instead, Thread defines link-local and realm-local scope unicast prefix-based IPv6 multicast address used for All Thread Nodes, including SEDs. These multicast addresses vary by Thread network, because it is built on the unicast Mesh-Local prefix (see RFC 3306 for more details on unicast-prefix-based IPv6 multicast addresses).
```

dsyx, 2020-06-28:
```
You might notice that Sleepy End Devices (SEDs) are not included as a recipient in the multicast table above. Instead, Thread defines link-local and realm-local scope unicast prefix-based IPv6 multicast address used for All Thread Nodes, including SEDs. These multicast addresses vary by Thread network, because it is built on the unicast Mesh-Local prefix (see RFC 3306 for more details on unicast-prefix-based IPv6 multicast addresses).
```

This looks great !

jwhui, 2020-06-29:
@dsyx , submitted #5160. Please review.

PR #5160 (2020-06-30): [devsite] add clarifications to multicast addressing primer
Resolves #5153.
closed 2020-06-30 by jwhui