openthread/openthread #5170: [github-actions] cli-ftd-otns failing in master

jwhui, 2020-06-30:
**Describe the bug** A clear and concise description of what the bug is.

cli-ftd-otns GitHub Action is failing in master

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id 5ce4046
2. IEEE 802.15.4 hardware platform N/A
3. Build steps N/A
4. Network topology N/A

**Expected behavior** A clear and concise description of what you expected to happen.

cli-ftd-otns GitHub Action passes.

**Console/log output** If applicable, add console/log output to help explain your problem.

https://github.com/openthread/openthread/runs/824271448

[logs_9382.zip](https://github.com/openthread/openthread/files/4854580/logs_9382.zip)

**Additional context** Add any other context about the problem here.

None.

ctan-g, 2020-06-30:
Looking into this

jwhui, 2020-06-30:
> Looking into this

Thanks, @ctan-g !

ctan-g, 2020-06-30:
@jwhui I found that the cause is my `OT_TOOL_WEAK` definition of `otPlatOtnsStatus` in `otns.cpp` isn't properly being overridden by the strong one in `otns.c` during linking when building the simulation images. Do you happen to know why this is happening?

simonlingoogle, 2020-07-01:
I wonder how did we pass the GitHub action checks.

simonlingoogle, 2020-07-01:
Turn out `cli-ftd-otns` GitHub action is not using the current openthread commit of the PR, but always using openthread master for checking. This was the reason why `cli-ftd-otns` check passed.

PR #5171 (2020-07-01): [otns] fix simulation to use strong otPlatOtnsStatus definition
This PR fixes #5170 

This PR also fixes the issue that OTNS tests failed to use openthread of the current PR, but used openthread master instead. 

@ctan-g could you verify if this fix works for Silk?
closed 2020-07-01 by jwhui