openthread/openthread #5175: [joiner] retry when not found

bukepo, 2020-07-01:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
Retry sending discovery request is a common use case for joiners. The joiner routers may fail to respond the request for either because the commissioner is not ready, or the commission didn't update steering data in time, or the discovery request packet was lost.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
introduce a feature to retry when the join attempt fails for no discover response is received. The retry may happen within a user given timeout or a user give times.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
* Let the application layer to implement retry by their own. I don't think this is good because every OpenThread users will have to implement this logic. And it takes an additional timer.

**Additional context** Add any other context or screenshots about the feature request here.

bukepo, 2020-07-01:
@abtink @jwhui Thoughts?

abtink, 2020-07-03:
Sounds good and reasonable to me.

jwhui, 2020-07-03:
I think that makes sense. Usual questions is what is the retry/backoff policy (linear vs exponential)? How much flexibility do we need to provide?

bukepo, 2020-07-03:
Currently, I'm considering a simple linear way. For example, users may request try at most 5 times with a fixed interval 5s.
Possible API:
```cpp
void Joiner::SetRetry(uint32_t aInterval, uint8_t aTimes);
```
If users don't want to use this feature, they can schedule their own retry in the callback functions which already exists, this will not harm the flexibility. Thought?

wgtdkp, 2020-07-03:
Can't this be implemented at application layer of the openthread core lib? For example, in ot-br-posix. In this case, we have the flexibility to easily change the retry strategy when product requirements changed which keep openthread joiner API simple. Thoughts?

wgtdkp, 2020-07-03:
In this case, we need to expose the discovery response event to user, which could be a burden of implementation. Then current propose also works for me.

simonlingoogle, 2020-07-03:
Retrying discovery requests for multiple times seems quite reasonable. What about using a configuration macro (e.g. `#define OPENTHREAD_CONFIG_DISCOVERY_REQUEST_RETRY 1`) instead of adding another OpenThread API.

bukepo, 2020-07-03:
@simonlingoogle chip vendors releases Thread prebuilt firmwares, which is often directly used by product vendors. I don't think fixed parameters can fit in different product's use cases.

simonlingoogle, 2020-07-03:
I think there is a difference between retries of Discovery Request and retries of the whole joining process.
My personal preference would be using configuration macro for the former and using openthread API for the latter.

What about using an API to configure the retry strategy for the whole joining process which uses exponential backoffs within a given interval range. 
```c++
void otJoinerSetRetries(uint8_t aTimes, uint32_t aMinInterval, uint32_t aMaxInterval); 
```

User can also set `aMinInterval` and `aMaxInterval` to a same duration to have linear backoffs.

wgtdkp, 2020-07-03:
Should we also introduce this into Thread specification?

jwhui, 2020-07-03:
> Should we also introduce this into Thread specification?

I think how many times and with what delay to send Discovery Request messages is a product-specific decision. The choice largely depends on the overall UX for the product and has no influence on the performance of the mesh network.

jwhui, 2025-01-07:
Closing stale issue.

closed 2025-01-07 by jwhui