openthread/openthread #5186: [cli] add a netstat command

bukepo, 2020-07-03:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
I hope to have a command to list all sockets, which can be helpful for debugging issues.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
 provide a simple implementation of netstat-like command, including bound address, port and status.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
simply list all open ports.

**Additional context** Add any other context or screenshots about the feature request here.

bukepo, 2020-07-03:
@jwhui @abtink what do you think?

jwhui, 2020-07-03:
I assume this is an OpenThread CLI? Sounds like a good idea to me.

bukepo, 2020-07-03:
Yes, an OpenThread CLI.

abtink, 2020-07-06:
Sounds great to me.

closed 2020-07-09 by jwhui