openthread/openthread #5204: Porting Guide reference to updating PRETTY_SUBDIRS appears to be inaccurate

tarpmaster, 2020-07-07:
**Describe the bug** A clear and concise description of what the bug is.

In the **_Porting Guide --> Set up the Build Environment --> GNU Automake --> Subdirectory Configuration_**, there is a note to update the **PRETTY_SUBDIRS** setting in **/openthread/examples/platforms/Makefile.am**.

However that setting does not appear to exist.  Seems likely the process was probably refactored at some point and the docs got out of sync.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
        f490ea658c82bc37c732310384a76c83fccdf387
2. IEEE 802.15.4 hardware platform
        n/a
3. Build steps
        n/a
4. Network topology
        n/a

**Expected behavior** A clear and concise description of what you expected to happen.
        n/a

**Console/log output** If applicable, add console/log output to help explain your problem.
        n/a

**Additional context** Add any other context about the problem here.
        n/a

jwhui, 2020-07-07:
@tarpmaster , thanks for raising this issue. The mention of `PRETTY_SUBDIRS` should be removed.

/cc/ @Vyrastas

jwhui, 2020-07-10:
https://openthread.io/guides/porting/set-up-the-build-environment has been updated, thanks to @Vyrastas !

closed 2020-07-10 by jwhui

Vyrastas, 2020-07-10:
Thanks for closing these, I forgot to 😄