openthread/openthread #5209: [cli] use compressed IPv6 addresses

Moandor-y, 2020-07-08:
**Is your feature request related to a problem? Please describe.**
Current OpenThread CLI has many places displaying IPv6 addresses without compression. E.g. `ipmaddr` gives something like `ff02:0:0:0:0:0:0:1` which is verbose.

**Describe the solution you'd like**
Change these representations to compressed IPv6 addresses. E.g. `ff02:0:0:0:0:0:0:1` -> `ff02::1`.

Moandor-y, 2020-07-08:
@jwhui @abtink what do you think?

bukepo, 2020-07-08:
SGTM:+1: If we don't want to increase the overhead to all builds, we may add this function in the cli library only. Thoughts?

simonlingoogle, 2020-07-08:
Be careful  that it might cause OpenThread THCI or other scripts to fail. @Moandor-y

Moandor-y, 2020-07-08:
> Be careful that it might cause OpenThread THCI or other scripts to fail. @Moandor-y

Yes. At least all scripts using regex to match IPv6 addresses need to be changed to something like [this](https://stackoverflow.com/questions/53497/regular-expression-that-matches-valid-ipv6-addresses).

jwhui, 2020-07-09:
Compressed representation sounds good to me.

jwhui, 2024-09-27:
Closing stale issue.

closed 2024-09-27 by jwhui