openthread/openthread #5211: Porting Guide reference to kThreadError_NotImplemented appears to be incorrect

tarpmaster, 2020-07-08:
**Describe the bug** A clear and concise description of what the bug is.

In **_https://openthread.io/guides/porting/implement-advanced-features_**, there is a reference to **kThreadError_NotImplemented** in the Auto Frame Pending description.  Probably is supposed to be **OT_ERROR_NOT_IMPLEMENTED**.

**To Reproduce** Information to reproduce the behavior, including:
n/a

1. Git commit id
n/a
2. IEEE 802.15.4 hardware platform
n/a
3. Build steps
n/a
4. Network topology
n/a

**Expected behavior** A clear and concise description of what you expected to happen.
n/a

**Console/log output** If applicable, add console/log output to help explain your problem.
n/a

**Additional context** Add any other context about the problem here.
n/a

jwhui, 2020-07-08:
@tarpmaster , thanks for raising this issue. Yes, we should replace `kThreadError_NotImplemented` with `OT_ERROR_NOT_IMPLEMENTED`.

/cc/ @Vyrastas

jwhui, 2020-07-10:
https://openthread.io/guides/porting/implement-advanced-features has been updated, thanks to @Vyrastas !

closed 2020-07-10 by jwhui