openthread/openthread #5218: [ci] "external-commissioner" test failures - dependency between ot-br and openthread repo PRs

abtink, 2020-07-10:
Noticed that in PR https://github.com/openthread/openthread/pull/5213 that the "external-commissioner" CI test (under simulation workflow and merged recently from https://github.com/openthread/openthread/pull/5187) is failing [[link]](https://github.com/openthread/openthread/pull/5213/checks?check_run_id=855929502).

This seems to be due to the test trying to build ot-br repo as well (with the new changes from OT PR pulled in):
```
FAILED: src/dbus/server/CMakeFiles/otbr-dbus-server.dir/dbus_thread_object.cpp.o 
../../src/dbus/server/dbus_thread_object.cpp: In member function ‘otError otbr::DBus::DBusThreadObject::SetMeshLocalPrefixHandler(DBusMessageIter&)’:
2126
../../src/dbus/server/dbus_thread_object.cpp:441:5: error: ‘otMeshLocalPrefix’ was not declared in this scope
2127
     otMeshLocalPrefix                         prefix;
2128
     ^~~~~~~~~~~~~~~~~
2129
compilation terminated due to -Wfatal-errors.
```

Note that this PR (per comments) is actually changing some of the public OT definitions (e.g., replacing`otMeshLocalPrefix` definition with `otIp6NetworkPrefix`). 

Looks like `ot-br` build is pulling in OT from PR (rather then the version of OT it is snapped to). This is adding dependency between repos (which we should avoid).

Digging a bit more into it, I think this recent merged PR https://github.com/openthread/ot-commissioner/pull/125 may be the cause of this.

@Moandor-y can you please help look into this? Thanks.

cc/ @bukepo

Moandor-y, 2020-07-10:
Yes. It will fail if there are any incompatible changes. This made the two repo depend on each other, and as discussed with @bukepo , @simonlingoogle and @wgtdkp , the only way is to change both repos at the same time. Since GitHub doesn't support PR dependencies, it has to be done manually. This does not only happen with `ot-br`, both `ot-commissioner` and `ot-br` are pulling `openthread` from the PR. Each time we encounter this we have to modify multiple repos and merge the PRs together (with some checks in failed state).
Or is there any better approach?

abtink, 2020-07-10:
I would highly advice against the approach to allow PRs to be merged with failed CI tests. 

I think we should avoid creating any such dependencies in the CI of openthread. This would block development of OT.

I would say, if any of OT CI test wants to build other repos, it should use fixed version (snapped to certain OT tag/commit that are known to build ok). After the change in main OT fork is merged, the dependent repos can be updated accordingly.

Moandor-y, 2020-07-10:
I agree that we should avoid dependencies. But if we use snapped OT commit, the coverage data won't be correct as the codes actually got run are not latest. So in https://github.com/openthread/openthread/pull/5187 I firstly set this job to allow failures (https://github.com/openthread/openthread/pull/5187#pullrequestreview-442285389), maybe we can consider this approach?

simonlingoogle, 2020-07-10:
I suggested against allowing CI tests to fail only to increase code coverage. 

However, after a second thought, I think checking `openthread PR` using `ot-br-posix` with errors ignored in CI do have benefits:

- It does not enforce `ot-br-posix` to work with `openthread PR`, thus introduces no dependency between repos
- The test results can give us a **hint** if this PR is going to break `ot-br-posix`. And if it does, we can be better notified and handle it sooner (either fix this `openthread` PR or enhance `ot-br-posix` later).
- It does give us more code coverage if tests pass

abtink, 2020-07-10:
Thanks guys. Allowing the "external commissioner" test to fail on main OT repo CI sounds reasonable to me.

jwhui, 2020-07-22:
Closing in favor of openthread/ot-commissioner#127

closed 2020-07-22 by jwhui