openthread/openthread #5224: Coap Request response not received on client end

bhumikaajain, 2020-07-10:
I am using Nordic's SDK for openthread and Zigbee v4.0.0  on nrf52840 platform.
In openthread coap client server model, I have created a bootstrap resource.
When client hit on that bootstrap resource, the request is received by server.
I am using **otCoapSendResponse** api to send the response back but the response is not  received on client end.
After sending the response, application reboots.

jwhui, 2020-07-11:
@bhumikaajain , we need more information to help diagnose. Are you able to share the code you are using?

You can also refer to [src/cli/cli_coap.cpp](https://github.com/openthread/openthread/blob/master/src/cli/cli_coap.cpp), which provides a demonstration of using CoAP APIs.

jwhui, 2020-07-28:
Closing stale issue.

closed 2020-07-28 by jwhui