openthread/openthread #5227: [joiner router] the discovery request should be accepted if extended pan id matches

bukepo, 2020-07-13:
**Describe the bug** A clear and concise description of what the bug is.
The discovery request should be accepted if extended pan id matches. However the current code is doing the opposite:
https://github.com/openthread/openthread/blame/master/src/core/thread/mle_router.cpp#L2840

**To Reproduce** Information to reproduce the behavior, including:

1. 4806fa4d0dd454d05452c1dbf219c7fb7ff2d3fa

**Expected behavior** A clear and concise description of what you expected to happen.
Accept the discovery request if the current extended panid matches any one of the attached in discovery request.
We should add a test for this.

wgtdkp, 2020-07-13:
It seems the behavior is intended:

```txt
On receipt of a Discovery Request message containing a Thread Discovery TLV, a Thread Router or REED MUST send a Discovery Response message except for when at least one of the two following situations occurs:
1.Joiner flag set and Joining not allowed:
    - The Discovery Request TLV Joiner flag is set to ‘1’ AND the Thread Network Partition of the Router/REED does not allow Joining (there is no Active Commissioner or the Active commissioner has disallowed all joiners via the Steering Data).
2. Extended PAN ID filter match:
    - The Discovery Request payload contains one or more valid Extended PAN ID TLVs AND the Router/REED Thread Network Extended PAN ID matches at least one of the Extended PAN ID values.

If either of the two cases above applies, or the Discovery Request does not contain a Thread Discovery TLV, the Router or REED SHOULD NOT prepare a reply and the Discovery Request
frame SHOULD be discarded.
```
chap 8.4.4.1.1.2 of Thread 1.1.

bukepo, 2020-07-13:
> It seems the behavior is intended:
> 
> ```
> On receipt of a Discovery Request message containing a Thread Discovery TLV, a Thread Router or REED MUST send a Discovery Response message except for when at least one of the two following situations occurs:
> 1.Joiner flag set and Joining not allowed:
>     - The Discovery Request TLV Joiner flag is set to ‘1’ AND the Thread Network Partition of the Router/REED does not allow Joining (there is no Active Commissioner or the Active commissioner has disallowed all joiners via the Steering Data).
> 2. Extended PAN ID filter match:
>     - The Discovery Request payload contains one or more valid Extended PAN ID TLVs AND the Router/REED Thread Network Extended PAN ID matches at least one of the Extended PAN ID values.
> 
> If either of the two cases above applies, or the Discovery Request does not contain a Thread Discovery TLV, the Router or REED SHOULD NOT prepare a reply and the Discovery Request
> frame SHOULD be discarded.
> ```
> 
> chap 8.4.4.1.1.2 of Thread 1.1.

I agree. I misunderstood.

closed 2020-07-13 by bukepo

abtink, 2020-07-13:
@bukepo  I also thought it was  mistake (a while ago) reading the same code  😄  (maybe we should add a comment in the code). 

It basically allows scanner to filter out some of the ext pan ids (scanner indicating it already knows about these PANs and does not need a response from any node matching the same ext PAN Id).

bukepo, 2020-07-13:
Yes:) And it seems this feature is not covered by our CI tests yet.