openthread/openthread #5234: [instance] elide otInstance from API?

bukepo, 2020-07-14:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.

There are seldom multiple instances use cases. How ever to support it, almost all OpenThread API were added the instance pointer. 

**Describe the solution you'd like** A clear and concise description of what you want to happen.
Elide the pointer from normal public APIs. But provide a API to select the instance, such as:

void otSelectInstance(otInstance *aInstance);

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
None.
**Additional context** Add any other context or screenshots about the feature request here.
#5231

simonlingoogle, 2020-07-14:
Maybe we should remove all `otInstance *aInstance` from APIs and not provide `otSelectInstance`.
There seems no real use case for `multiple OT instances`

wgtdkp, 2020-07-14:
> There seems no real use case for `multiple OT instances`

ot-esp32 project needs to dynamically allocate the OT instance.

simonlingoogle, 2020-07-14:
> > There seems no real use case for `multiple OT instances`
> 
> ot-esp32 project needs to dynamically allocate the OT instance.

Yes, you are right. We took advantage of the `multiple instances` so that the **single** OT instance is dynamically allocated on heap. But what `ot-esp32` needs is `dynamically allocated OT instance`, not `multiple instances`. We can easily have only one OT instance on heap without `multiple instances`.

abtink, 2020-07-14:
I have been thinking/suggesting removal of support for multi-instance feature for some time.

I also see no real use-case for it and it adds a lot of extra code and complexity.

It was added early on in the OT development. I think one idea/use for it was (at some point) to allow multiple netifs (different thread netifs or maybe some not tied to a thread network/radio) to be managed by OT. We don't support multiple `Netif` anymore.

jwhui, 2020-07-14:
I agree that there does not seem to be any real need today. However, I think the likelihood of needing to support multiple Thread networks will increase as deployment of Thread increases.

Silicon Labs EmberZNet stack has support for multiple Zigbee networks [AN724: Designing for Multiple Networks
on a Single Zigbee Chip](https://www.silabs.com/documents/public/application-notes/an724-multi-network.pdf).

I am fine to remove multiple instance support currently. However, I would not be surprised if we need to add it back when Thread is more widely deployed.

abtink, 2020-07-14:
If we think/see that there may be use-cases in future, then I'd suggest we keep it as is.

LuDuda, 2020-07-14:
If there is a real use-case foreseen, I support keeping the existing API rather than implementing `otSelectInstance()` API - which may complicate development in multi-threaded environment. Also re-introducing the instance to all the APIs will force current OT users to adjust APIs of their applications twice.

Can we predict what will be the gain of removing the instance from the API in case of e.g. RAM/Flash numbers?

simonlingoogle, 2020-07-15:
Agree that `otSelectInstance` is not convenient to use if there are indeed multiple OT instances. Agree that we should not alter current API if we might need multiple instances in the future. 

I wouldn't expect significant reduce in RAM/Flash because only the API adaption layer will be simpler, but the current OT core implementation won't be affected from the API simplification.

BTW, one use case I had in mind about multiple OT instances is that we can run multiple OT simulation instances in the same process of the simulator, which I would expect to perform much better than the current simulation mechanism that uses UDP messages for IPC.

bukepo, 2020-07-15:
If there are multiple Thread networks, are they going to running in multiple threads? lwIP supports multiple netifs, but it still runs in a single thread. And it has some global APIs to get information of the lower layers of the current packet being processed. So If we remove instance from APIs, we can still support multiple networks. For example, add otGetCurrentInstance() so in OpenThread callbacks, users can still tell which network triggered this event.

Besides, the current logging system doesn't fully support multiple instances. For example, if the log output is set to application, there's no parameter to tell logging to which instance.

Removing the parameter simplifies the APIs and can save resources both in single network situation and multiple-networks situations. I like the idea of @abtink that support multiple netifs instead of instances, so that more resources(heaps, mbedtls contexts, message buffers, logging) can be shared between networks.

jwhui, 2024-09-27:
Closing stale issue.

closed 2024-09-27 by jwhui