openthread/openthread #5252: Thread Harness Failed Test Cases

testter2, 2020-07-16:
Hello everybody,
I'm using Thread Test Harness software Release 2.0.0 and I'm integrating platform with open thread cli example downloaded from the ripo on 05/05/2020 and the following test case is failing when I run it:
! 8.2.1 (OMC Joining with JR any commissioner single correct) Commissioner : DUT, Joiner_1 : ARM , JoinerRouter : NXP
here is the report produced from running this test : 
[Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_16-13_46_55.pdf](https://github.com/openthread/openthread/files/4932472/Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_16-13_46_55.pdf)
and here is the thread console log : 
[ThreadConsoleLog_2020_07_16_13_26.log](https://github.com/openthread/openthread/files/4932486/ThreadConsoleLog_2020_07_16_13_26.log)
and this the sniffer capture : 
[Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_16-13_38_05_Ch-11.zip](https://github.com/openthread/openthread/files/4932540/Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_16-13_38_05_Ch-11.zip)

I am using the following configurations:
#define OPENTHREAD_CONFIG_COMMISSIONER_ENABLE 1
#define OPENTHREAD_CONFIG_DTLS_ENABLE 1
#define OPENTHREAD_CONFIG_JOINER_ENABLE 1
#define OPENTHREAD_CONFIG_COAP_API_ENABLE 1
#define OPENTHREAD_CONFIG_COAP_OBSERVE_API_ENABLE 0
#define OPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE 0

I need to know what could be the reason of this failure.

Thanks,

simonlingoogle, 2020-07-16:
You are testing the DUT in the `Commissioner` role, so you need to have the configuration `OPENTHREAD_CONFIG_COMMISSIONER_ENABLE=1`, which seems missing. 

Moreover, our most latest release is [Thread Reference 2019-11-13](https://github.com/openthread/openthread/releases/tag/thread-reference-20191113) which you can download from [Releases](https://github.com/openthread/openthread/releases).

testter2, 2020-07-19:
Thanks @simonlingoogle for your support.

I will try the pointed release and get back to you with the results.

testter2, 2020-07-19:
Hi @simonlingoogle ,

I've tried integrating with the [Thread Reference 2019-11-13](https://github.com/openthread/openthread/releases/tag/thread-reference-20191113) and I've used the following configurations:
#define OPENTHREAD_CONFIG_MAC_FILTER_ENABLE 1
#define OPENTHREAD_CONFIG_REFERENCE_DEVICE_ENABLE 1

and the test case !8.2.1(OMC Joining with JR any commissioner single correct) still fails with the following results:
1- Test report : 
[Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_19-16_21_53.pdf](https://github.com/openthread/openthread/files/4943993/Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_19-16_21_53.pdf)

2- Test log:
[ThreadConsoleLog_2020_07_19_16_00.log](https://github.com/openthread/openthread/files/4944000/ThreadConsoleLog_2020_07_19_16_00.log)

3- Sniffer Capture:
[Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_19-16_13_04_Ch-11.zip](https://github.com/openthread/openthread/files/4944003/Fail_ARC_8.2.1_Commissioner_R2.0.0_2020_07_19-16_13_04_Ch-11.zip)

Thanks.

testter2, 2020-07-20:
Hi @simonlingoogle,

After changing the topology and making both the JoinerRouter and Router_1 ARM devices the test case passed.

Is there an explanation for this?

also as you mentioned that the latest OpenThread release was in Nov-19 I want to know the date of the upcoming release?

Thanks.

simonlingoogle, 2020-07-21:
Hi @testter2 ,

Could you also share the passed result using ARM devices? 

The new test result shows that you are still using OT 20190621, rather than 20191113. 
```
DUT Firmware Version:
OPENTHREAD/20190621-gadecc057;
NONE; Jul 19 2020 17:18:26
```

I would also suggest you use TestHarness 2.0.1 (latest release), in which SiLabs firmware has been updated to `2.10.2 build 127` (you are currently using `2.7.1 build 246`).

If the result is still failing. I would suggest you run the test for multiple times because both OT and SiLabs reference devices uses software decryption for DTLS so they can be unreliable for commissioning cases. 

Thanks,

testter2, 2020-07-21:
Hi @simonlingoogle thanks for your support,

I replaced the SiLabs device and OpenThread device with ARM devices  and the test case is passing now the following results:

1- Test report:
[Pass_ARC_8.2.1_Commissioner_R2.0.0_2020_07_21-16_51_09.pdf](https://github.com/openthread/openthread/files/4954919/Pass_ARC_8.2.1_Commissioner_R2.0.0_2020_07_21-16_51_09.pdf)

2- Sniffer capture:
[Pass_ARC_8.2.1_Commissioner_R2.0.0_2020_07_21-16_41_30_Ch-11.zip](https://github.com/openthread/openthread/files/4954951/Pass_ARC_8.2.1_Commissioner_R2.0.0_2020_07_21-16_41_30_Ch-11.zip)

Thanks,

closed 2020-07-28 by jwhui