openthread/openthread #5258: Having BR as router/leader

karsev, 2020-07-17:
Hi,

This is a bit broader question/problem that might be interesting for Thread developers.

This topic popped-up, because we are having, more than usual, link errors when running ping tests in the installation,which has around 50 devices. We haven't yet optimised the radio driver - mac layer functionality, so let's keep it as a separate problem.

Shortly, the consecutive link errors cause links to be removed and this has the most biggest impact on the border router when it's role is end-node. This is because it's child RLOC address changes quite often causing often network data updates and thus more actions on nodes (e.g. address solicitations).

What would improve the situation with BR is to have it run as router only, but how to force it to be router?
One of the conditions in spec 1.1.1 in chapter 5.9.9 is that router can downgrade itself if ->

"Not be the sole Border Router for a particular Provisioning Domain (see Section 34 5.13.7, Provisioning Domain). 
If multiple Border Routers for the same Provisioning  Domain are eligible to downgrade they MAY use some external mechanism to select 36 which do so"

So once BR has become router it wouldn't downgrade itself for the reason having too many routers in the network.
I didn't find this rule implemented. Is this something that should be handled by our BR application and not by stack?
Or have I misunderstood something?

Anyway this wouldn't guarantee that BR is run in the router role always.
So the second question is that when the commercial extension source code is available?
Asking, because then it would be possible to have BR run always as a network leader/router by giving it more partition weight.

Thanks

jwhui, 2020-07-18:
@karsev , thanks for raising this question.

As you correctly observed, OpenThread does not provide a mechanism that supports the following statement from Section 5.9.9:

> If multiple Border Routers for the same Provisioning Domain are eligible to downgrade, they MAY use some external mechanism to select which do so."

One issue is that the statement above assumes you have some out-of-band channel between the border routers. In particular, what is the expected behavior when including border routers that are not under the same control?

As you mentioned, the above wouldn't guarantee a border router operates in the router role. It is fairly trivial to implement a mechanism that prevents a border router from downgrading to end device. The harder part is having a mechanism that can force a border router to upgrade from an end device to router. Thread does not currently provide any provisions to remotely force a router to downgrade.

Thread 1.2 does attempt to address this by using different partition weights and changing the partition selection rules. It doesn't completely solve the problem - for example, we would run into the same problem if there are many backbone routers in the network. However, the assumption is that the number of backbone routers will be limited - which could be enforced in a managed environment.

Commercial Extensions is still a work-in-progress, both in terms of spec and implementation. As a Thread Group member, if you are interested in accessing the prototype code we use in Thread Group interop events, feel free to reach out to me directly.

karsev, 2020-08-03:
Thanks @jwhui, this is fine!

closed 2020-08-03 by karsev