openthread/openthread #5259: OpenThread API: How to get child table using OpenThread API?

BusKetZz, 2020-07-17:
Hello Guys!
I am not sure whether it is a good place to ask such a question nevertheless I hope you will help :)
I have the OpenThread CLI application and I would like to create UDP datagram with the payload consisted of child table information.
So my question is: **how to get child table info and which API should I use?**
I have done some research, and I consider to play with **otThreadGetChildInfoByIndex()** placed in for/while loop. Does it make sense??

Best regards!

abtink, 2020-07-17:
@BusKetZz yes. That would work. You can use `otThreadGetMaxAllowedChildren()` to get max number of chidlren. 
The `ncp_base` code for `CHILD_TABLE` property getter can be good example:

```
template <> otError NcpBase::HandlePropertyGet<SPINEL_PROP_THREAD_CHILD_TABLE>(void)
{
    otError     error = OT_ERROR_NONE;
    otChildInfo childInfo;
    uint16_t    maxChildren;

    maxChildren = otThreadGetMaxAllowedChildren(mInstance);

    for (uint16_t index = 0; index < maxChildren; index++)
    {
        if ((otThreadGetChildInfoByIndex(mInstance, index, &childInfo) != OT_ERROR_NONE) || childInfo.mIsStateRestoring)
        {
            continue;
        }

        SuccessOrExit(error = mEncoder.OpenStruct());
        SuccessOrExit(error = EncodeChildInfo(childInfo));
        SuccessOrExit(error = mEncoder.CloseStruct());
    }

exit:
    return error;
}
```

Or you can check the CLI `Interpreter::ProcessChild()` implementation (almost the same pattern).

BusKetZz, 2020-07-17:
@abtink 
Thank you very much.
So this is my idea:

```c
static *otChildInfo get_child_info(otInstance *instance)
{
    uint16_t max_children = otThreadGetMaxAllowedChildren(instance);
    static otChildInfo child_info[max_children];

    for (uint16_t child_index = 0; child_index < max_children; child_index++) {
        if (otThreadGetChildInfoByIndex(instance, child_index, child_info[child_index]) == OT_ERROR_NOT_FOUND) {
              break;
        }
    }

    return child_info;
}
```

Would it work?

abtink, 2020-07-17:
I think you want to go through all entries (entire range of indexes) and skip over the ones that are not found, i.e., do cannot `break` when entry for an index is not in use as the entries may be freed in any order.

BusKetZz, 2020-07-17:
So let's modify it a little bit:

```c
static void get_child_table(otInstance *instance, otChildInfo *child_table, uint16_t max_children)
{
    for (uint16_t child_index = 0; child_index < max_children; child_index++) {
        otThreadGetChildInfoByIndex(instance, child_index, child_table[child_index])             
    }
}
```

This way, I can form UDP payload consisting of an array of `otChildInfo` type members.

abtink, 2020-07-17:
@BusKetZz , you want to follow the pattern I mentioned in the first reply, i.e. go through all indexes, skip over the ones that `otThreadGetChildInfoByIndex()` returns an error.

BusKetZz, 2020-07-18:
@abtink 
So my idea is that:

```c
static void get_child_table(otInstance *instance, otChildInfo *child_table, uint16_t max_children)
{
    otChildInfo child_info;    

    for (uint16_t child_index = 0; child_index < max_children; child_index++) {
        if (otThreadGetChildInfoByIndex(instance, child_index, child_info) == OT_ERROR_NONE) {
             *child_table = child_info;
              child_table++;
        }             
    }
}
```

Forgive me that I show and ask for help while prototyping, but I think this is the best place to do that :)

BusKetZz, 2020-07-20:
I have additional problem with the build process:
```
openthread/examples/apps/cli/main.c:200: undefined reference to `otThreadGetMaxAllowedChildren'
```

Should I add some files into build process?

jwhui, 2020-07-20:
@BusKetZz , the [`otThreadGetChildInfoByIndex()`](https://github.com/openthread/openthread/blob/434e6bbffc26a5eef1f5e4f5733b8bd536c250a6/src/core/api/thread_ftd_api.cpp#L265) is only available for FTD builds.

You should either:

1. Disable MTD and RCP in your build options, or

1. Wrap your child table code with `#if OPENTHREAD_FTD`.

Again, you can refer to the [CLI code](https://github.com/openthread/openthread/blob/434e6bbffc26a5eef1f5e4f5733b8bd536c250a6/src/cli/cli.cpp#L832) for example usage.

BusKetZz, 2020-07-21:
@jwhui 
Thank for your answer. Could you tell me how to disable MTD and RCP from build?

jwhui, 2020-07-21:
> @jwhui
> Thank for your answer. Could you tell me how to disable MTD and RCP from build?

@BusKetZz , if you are using the convenience makefiles in `examples/`, you can look for `configure_OPTIONS`:
https://github.com/openthread/openthread/blob/57ee45be21c2b6cbba0dc53cea235fca33772cbc/examples/Makefile-cc2538#L44-L52

Remove mentions of `--enable-mtd` and `--enable-radio-only`.

BusKetZz, 2020-07-23:
Thank you very much for all the information. I have solved my issue by adding `#ifdef OPENTHREAD_FTD` directive.

closed 2020-07-23 by BusKetZz