openthread/openthread #5274: Enhance size-report to include 1.2 features

simonlingoogle, 2020-07-22:
Continuing discussion here since the PR will be merged. 

Here are some examples by adding more 1.2 columns to `size-report`.  

### Example 1:

|  name  |  branch  | 1.1 text  | 1.1 data  | 1.1 bss  | 1.1 total | 1.2 text  | 1.2 data  | 1.2 bss  | 1.2 total |
| :----: | :------: | -----: | ----: | ---: | ----: |  ----: | ----: | ----: | ----: |
| libopenthread-ncp-ftd.a | master | 396720 | 736 | 72828 | 470284 | 396720 | 736 | 72828 | 470284 |
|  | #5268 | 396720 | 736 | 72828 | 470284 | 396720 | 736 | 72828 | 470284 |
|  | +/- | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| libopenthread-ncp-mtd.a| master | 319360 | 736 | 64180 | 384276 | 319360 | 736 | 64180 | 384276 |
|  | #5268 | 319360 | 736 | 64180 | 384276 | 319360 | 736 | 64180 | 384276 |
|  | +/- | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |

### Example 2 (Only add 1.1/1.2 for total column):
|  name  |  branch  | text  | data  | bss  | 1.1 total | text  | data  | bss  | 1.2 total |
| :----: | :------: | -----: | ----: | ---: | ----: |  ----: | ----: | ----: | ----: |
| libopenthread-ncp-ftd.a | master | 396720 | 736 | 72828 | 470284 | 396720 | 736 | 72828 | 470284 |
|  | #5268 | 396720 | 736 | 72828 | 470284 | 396720 | 736 | 72828 | 470284 |
|  | +/- | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |
| libopenthread-ncp-mtd.a | master | 319360 | 736 | 64180 | 384276 | 319360 | 736 | 64180 | 384276 |
|  | #5268 | 319360 | 736 | 64180 | 384276 | 319360 | 736 | 64180 | 384276 |
|  | +/- | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |

### Example 3 (removing `total`) :

|  name  |  branch  | 1.1 text  | 1.1 data  | 1.1 bss  | 1.2 text  | 1.2 data  | 1.2 bss  | 
| :----: | :------: | -----: | ----: | ---: | ----: |  ----: | ----: |
| libopenthread-ncp-ftd.a | master | 396720 | 736 | 72828  | 396720 | 736 | 72828 | 
|  | #5268 | 396720 | 736 | 72828  | 396720 | 736 | 72828 |
|  | +/- | 0 | 0 | 0 | 0 | 0 | 0 | 
| libopenthread-ncp-mtd.a | master | 319360 | 736 | 64180 |  319360 | 736 | 64180 |
|  | #5268 | 319360 | 736 | 64180 | 319360 | 736 | 64180 |
|  | +/- | 0 | 0 | 0 | 0 | 0 | 0 |

_Originally posted by @simonlingoogle in https://github.com/openthread/openthread/pull/5248#discussion_r458561842_

jwhui, 2020-07-22:
I think Example 3 is sufficient for me. I generally prefer to see the individual breakdown vs the total number.

simonlingoogle, 2020-07-29:
Hi @bukepo 
Would you give comments on what we need to enhance to achieve 1.2 size-report as in Example 3 ?

bukepo, 2020-07-29:
I hope we can print size diff of each symbols.

simonlingoogle, 2020-07-29:
> I hope we can print size diff of each symbols.

Agree that it's useful to have size diff of each symbol. But the result would be very long, so maybe we should have a link in the comment which can guide us to the detailed report? 

I think this is another topic that worth discussing. However it is unrelated to 1.2 features?

bukepo, 2020-08-11:
> > I hope we can print size diff of each symbols.
> 
> Agree that it's useful to have size diff of each symbol. But the result would be very long, so maybe we should have a link in the comment which can guide us to the detailed report?
> 
> I think this is another topic that worth discussing. However it is unrelated to size report for 1.2 features?

Agree. I think it's better to display the results of 1.1 and 1.2 separately in two sections, which also works for future versions.

simonlingoogle, 2021-05-19:
Close for now since we already have Thread 1.2 size-report.

closed 2021-05-19 by simonlingoogle