openthread/openthread #5285: NRFX_LOG_INFO() / NRF_LOG_INFO(): How to use them?

BusKetZz, 2020-07-23:
I have a some question. I would like to use one of the USB log macros like `NRFX_LOG_INFO()` and `NRF_LOG_INFO()`. But when I set `NRF_LOG_ENABLED 1` in `sdk_config.h` and include `nrf_log.h' into my file in `openthread/examples/platforms/nrf528xx/src/` I get the following error:

```
rf_log.h:55:10: fatal error: nrf_strerror.h: No such file or directory
 #include "nrf_strerror.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
```

What should I do to enable this functionality?

LuDuda, 2020-07-23:
Hello @BusKetZz,

Thank you for asking.

In OpenThread repository the NRF_LOG is not usable (only stubbed). For logging purpose, you can use either OpenThread's `otPlatLog` functionality in vanilla OpenThread, or download the `nRF5 SDK for Thread and Zigbee` which gives you full functionality of NRF Logger.

BusKetZz, 2020-07-23:
@LuDuda 
Thank You for your answer. I wanted to enable `NRF_LOG` within OpenThread.
To conclude: in OpenThread it is not possible to use `NRF_LOG` functionality.
May I know why is that?

LuDuda, 2020-07-23:
Simply because `NRF_LOG` is quite a complex module present in Nordic's SDK. We try to keep essential support for nRF58xx in vanilla OpenThread without bringing to much stuff from SDK.

As I mentioned, feel free to download and use `nRF5 SDK for T&Z` which includes OpenThread along with additional drivers, components (like NRF_LOG) and examples.

BusKetZz, 2020-07-23:
@LuDuda 
Again, thank you very much for valuable answer :)
My application is based on `OpenThread` repostory so switching to `nRF SDK` is not acceptable.
Anyway, is it possible to implement `NRF_LOG` functionality into OpenThread manually? Or maybe it is to complicated to perform?

LuDuda, 2020-07-24:
Hello @BusKetZz 

It is possible to integrate `NRF_LOG` with vanilla OpenThread, however it will require to include a couple of files and prepare right linker scripts. Why you can't just simply use `otPlatLog` instead, or map NRF_LOG to `otPlatLog`? Depending on your application, moving it to `nRF SDK` might be extremely easy and fast.

BusKetZz, 2020-07-25:
@LuDuda 
Again, thanks for your answer. I really appreciate that :)
I have asked about it just to make some research and get a few extra information.
To conclude, if necessary I will use `otPlagLog` or I will map `NRF_LOG` to `otPlatLog`, but I think the first option is the most reasonable.
Thank You very much once again!

closed 2020-07-25 by BusKetZz