openthread/openthread #5290: Slome methods of ot::Time can be error-prone to use

simonlingoogle, 2020-07-24:
**Describe the bug**
Some API of ot::Time is error-prone to use.

#### GetDistancePast
```c++
    /**
     * This method returns a new `Time` instance which is in distant past relative to current `Time` object.
     *
     * The returned distance past `Time` is guaranteed to be equal or before (as defined by comparison operator `>=`)
     * any other `Time` which is before this `Time` object, i.e., for any `t` for which we have `*this >= t`, it is
     * ensured that `t >= this->GetDistantPast()`.
     *
     * @returns A new `Time` in distance past relative to current `Time` object.
     *
     */
    Time GetDistantPast(void) const { return Time(mValue - kDistantFuture); }
```

I believe that GetDistantPast() actually returns the same value as GetDistantFuture(). 
Furthermore, even if we returned a correct value for GetDistantPast, it's going to become **Distant Future** in a few milliseconds. My suggestion is that we should remove this function since there is no use case of `GetDistantPast` in the current code base. 

#### operator -
```c++
    /**
     * This method calculates the time duration between two `Time` instances.
     *
     * @note Expression `(t1 - t2)` returns the duration of the interval starting from `t2` and ending at `t1`. When
     * calculating the duration, `t2 is assumed to be in the past relative to `t1`. The duration calculation correctly
     * takes into account the wrapping of numeric value of `Time` instances. The returned value can span the entire
     * range of the `uint32_t` type.
     *
     * @param[in]   aOther  A `Time` instance to subtract from.
     *
     * @returns The duration of interval from @p aOther to this `Time` object.
     *
     */
    uint32_t operator-(const Time &aOther) const { return mValue - aOther.mValue; }
```
I think requiring that `t2` is in the past relative to `t1` is too strict for such a method. My suggestion is that we should use `int32_t` for all duration values so that we won't have such restriction. I think all `uint32_t aDuration` arguments of methods of `Time` should be changed to `int32_t aDuration` for better flexibility. 

Thougths? @abtink

abtink, 2020-07-24:
> I believe that `GetDistantPast()` actually returns the same value as `GetDistantFuture()`.

Yes, it is ok for them to return same value (it's known and intentional). They serve different purposes though. Note that both are always defined *relative* to a given `Time` instance.
- `GetDistantFuture()` is ensured to be after all other times than are after the given `Time` instance (as if it is the maximum of all times that are ahead of the given `Time` instance).
- `GetDistantPast()` is ensured to be before all other times that are considered before the `Time` instance (sort of like minimum of all times before the `Time` instance).

The main use for them is when trying to find the earliest/latest time from a given list of time instances that are known to be after/before a reference `Time` (you can see examples of this in code).

> Furthermore, even if we returned a correct value for `GetDistantPast()` and `GetDistantFuture()`, the Distant Past is going to become Distant Future in a few milliseconds. My suggestion is that we should remove this function since there is no use case of GetDistantPast in the current code base.

The distant future/past times are always defined *relative* to some reference `Time` instance. Without a reference `Time` instance,  they are meaningless. So if we get current time `now = GetNow()`, then `now.GetDistantPast()` is only meaningful relative to its reference `now` value.  

I prefer we keep `GetDistantPast()` even though it may not be used currently as a mirror of `GetDistantFuture()`.

> I think requiring that t2 is in the past relative to t1 is too strict for such a method. My suggestion is that we should use int32_t for all duration values so that we won't have such restriction. I [...]

The fact that we require this and return `uint32_t` is indeed intentional. 
- We want to allow duration calculation between two time instance to be able to use the entire `32-bit` range of values.
- We do not want callers to use this method to do relative comparison of two time instances (i.e. is one after/before the other one) instead we want the caller to use the operators `<` , etc.

The main point of adding `Time` class (which is a practically a wrapper over a `uint32_t` value) is to handle the roll-over of the time value and provide correct relative time comparison (i.e. overloads of `<`, etc operators).
- Before this we used raw  `uint32_t` value directly, and it was easy/common to make mistake - it's natural to simply check two time values `t1`, `t2` with a `t1 < t2` (I am myself guilty of this) - and the dangerous part is that may work ok until the time values get close to the roll-over range.
- We provided a bunch of helper methods like `IsBefore()` or `IsStrictlyBefore()`, or `int32_t Diff()` etc, but still it was very error-prone, easy to make a mistake.
- So with the `Time` class it is  intentional that we limited the APIs (e.g., did not allow `int32_t operaor -`). This is to force the users towards a single (and hopefully safer) pattern of use.

simonlingoogle, 2020-07-27:
> The distant future/past times are always defined _relative_ to some reference `Time` instance. Without a reference `Time` instance, they are meaningless. So if we get current time `now = GetNow()`, then `now.GetDistantPast()` is only meaningful relative to its reference `now` value.

I can see why `GetDistantFuture` works now. But it's still very misleading. In the current implementation, for any value `t`, we have `t` < `t.GetDistantFuture()` and `t.GetDistantFuture() < t`. This is already confusing by itself. Moreover, `t <= t.GetDistantFuture()` and `t.GetDistantFuture() >= t` are both **false**, which I believe could surprise  most users. 

In my opinion, a better solution is that we should avoid using the current `DistantFuture` value, but redefine `kDistantFuture = ((1UL << 31) - 1)`. It will make the distant Time values for any time `t` to be more reasonable:

- `DistantFuture` and `DistantPast` are now different values
- Relationship of `t` and `DistantFuture (DF)` is straightforward (i.e. t < DF, t <= DF, DF > t, DF >= t)
- Relationship of `t` and `DistantPast (DP)` is straightforward (i.e. t > DP, t >= DP, DP < t, DP <= t)

Thoughts?

simonlingoogle, 2020-07-27:
> * We want to allow duration calculation between two time instance to be able to use the entire `32-bit` range of values.

I think allowing duration to use the entire `32-bit` range is actually not useful in most of time. For any time `now`, range [now+1, now+kDistantFuture) are recognized as **future** times, range (now-kDistantFuture, now-1] are recognized as **past** times. When we compare any time to `now`, we can never use the entire `32-bit` range but only a range of `kDistantFuture`. 

Furthermore, I believe using `uint32_t` type for `Time` has an implicit assumption that the distance between any two time values we handle in code **should never exceed** `kDistantFuture`, otherwise it's impossible to determine which is future and which is past. With such assumption, we should never expect time distances > kDistantFuture. 

> * Before this we used raw  `uint32_t` value directly, and it was easy/common to make mistake - it's natural to simply check two time values `t1`, `t2` with a `t1 < t2` (I am myself guilty of this) - [...]

Agree with you that comparing directly on `uint32_t` time values is wrong. But I think we can have `operator-` and `operator<` both work in a normal way without any restriction. For any time values `t1` and `t2`, if `t1` < `t2`, then we can have `t1 - t2 < 0` and `t2 - t1 > 0`. 

The only exception to this rule will be time values of distance (1<<31). For any time `t` and `t + (1<<31)`, there is no way to determine which is current. I think we can just not handle such case since it we should never have time values with such big time distance.

abtink, 2020-07-27:
> In my opinion, a better solution is that we should avoid using the current `DistantFuture` value, but redefine `kDistantFuture = ((1UL << 31) - 1)`. It will make the distant Time values for any time [...]

The reason for current model is we want DF relative to `t` to be the larger than any other value that considered larger than `t`.  The above definition with `kDistantFuture = ((1UL << 31) - 1)` would work but would then exclude the corner case of value half range `(1<<31)` away. Note that (as you pointed) the value of `(1<<31)`  is (intentionally) left as ambiguous, i.e., is treated as both larger and smaller than `t`.  A quick side-note on this: 
- I had a solution in mind (at the time of adding this) to make this unambiguous with the following rule: "when comparing `t1 < t2` , if they are exactly half span `(1<<31)` apart, then use actual raw value to decide on which is larger". 
- But at the end decided not to add this since it was increasing code size (and complexity) for a corner case that should not be used and is not really practical.

I feel the current model is fine, but I don't have any objection to your suggestion for this (we anyways need to treat the `(1 << 31)` as a corner case that should not be practically used). If you feel strongly about this, please feel free to go ahead and change it. :)

---- 

> Furthermore, I believe using uint32_t type for Time has an implicit assumption that the distance between any two time values we handle in code should never exceed kDistantFuture, otherwise it's impossible to determine which is future and which is past.

Initially in the design, we were considering two classes to ensure the notions are different and distinguished:
- Having ` Time` class which is does **not** provide any comparison operators and the subtraction of two time instances gives a  `uint32_t` value.
   - The idea was to use it as timestamp of past events (e.g., to later use it in a frame history feature, or history of radio state changes, etc.). 
  - In all these use-cases, the time instances would always be treated as if they are in the past,
  - And it would be useful to be able to use entire 32-bit range (specially with micro time where `2^32` micro sec is ~1.2 hour).
- Then having a subclass `TimeComparable` class which now has the notion of past/future and operator overloads for `<`, etc.

At the end, we combined them into same class assuming that then the developer would choose how to use it.

simonlingoogle, 2020-07-28:
Any inputs? @bukepo @jwhui

jwhui, 2021-01-12:
Closing stale issue.

closed 2021-01-12 by jwhui