openthread/openthread #5293: [mle] re-synchronization with FTD child

LuDuda, 2020-07-24:
Hello,

I'm currently on [this](https://github.com/openthread/openthread/commit/e653478c503d5b13207b01938fa1fa494a8b87d3) git commit using nRF52840.

I noticed, that sometimes FTD child after reset can't re-attach to the old parent quite fast. The reason for this is that Child drops several `MLE Child Update Response` from the parent, and finally after few retries it starts regular attachment using `MLE Parent Request`.

```
-MLE-----: Failed to process Child Update Response: Drop
```

The reason for this is that `HandleChildUpdateResponse` method calls `HandleLeaderData` which returns `OT_ERROR_DROP`.
This is because `mLeaderData` (local Leader Data) is empty (i.e. partitionId = 0, weighting=0, leaderId = 0 etc.), and so it is different from the Parent's Leader Data.

The problem does not occur when `MLE Advertisement` message is received somewhere in the re-attachment process - since `mLeaderData` is updated.

@jwhui Is this expected behaviour? If not, would it make sense to accept new Leader Data from the parent?

simonlingoogle, 2020-07-24:
I think this behavior was introduced in #4860 which modified `HandleLeaderData` of FTD to skip handling LeaderData of a different partition.

The related conversation was here: https://github.com/openthread/openthread/pull/4860#discussion_r410940029

The intention was to avoid switching to a different partition because `Data  Response` message does not have the RouteTLV, which is mandatory for an FTD to join the partition. 

I am not sure of the possible fix yet. 

Any thoughts? @jwhui @LuDuda

simonlingoogle, 2020-07-24:
Hi @LuDuda , @jwhui 

With some local test and reproducing, I came up with a workaround. 

Instead of just dropping the `Child Update Response` (original code shown below)
```c++
#if OPENTHREAD_FTD
            // An FTD skips handling LeaderData of a different partition.
            VerifyOrExit(!IsFullThreadDevice() || (leaderData.GetPartitionId() == mLeaderData.GetPartitionId() &&
                                                   leaderData.GetLeaderRouterId() == GetLeaderId()),
                         error = OT_ERROR_DROP);
#endif
```
, the FTD can try hard to accept this `Child Update Response` if it's restoring: **do not drop the message if its current partition ID is 0 and allocate Router for Leader if it's not allocated**. I tested the code below and it worked (FTD reattaches to parent instantly)
```c++
#if OPENTHREAD_FTD
            // An FTD skips handling LeaderData of a different partition.
            VerifyOrExit(!IsFullThreadDevice() || mLeaderData.GetPartitionId() == 0 ||
                             (leaderData.GetPartitionId() == mLeaderData.GetPartitionId() &&
                              leaderData.GetLeaderRouterId() == GetLeaderId()),
                         error = OT_ERROR_DROP);

            if (IsFullThreadDevice() && !Get<RouterTable>().IsAllocated(leaderData.GetLeaderRouterId()))
            {
                Get<RouterTable>().Allocate(leaderData.GetLeaderRouterId());
            }
#endif
```

Or maybe we can always allocate for the new Leader and needs not to drop this message at all?

Thoughts?

LuDuda, 2020-07-27:
@simonlingoogle thank you for your time on looking at this. I agree that we should accept new Leader data from our parent whenever we try to restore our parent.

Nice find with potential MLE Advertisement processing problem. I think that having a flag is a reasonable approach. I thought also that we could extend Request TLV to include Route64 TLV in response in Child Update Response as well, but I think this is prohibited by the spec.

simonlingoogle, 2020-07-27:
> Nice find with potential MLE Advertisement processing problem. I think that having a flag is a reasonable approach. I thought also that we could extend Request TLV to include Route64 TLV in response in Child Update Response as well, but I think this is prohibited by the spec.

Personally, I prefer to remove the code of checking `LeaderDataTLV` and dropping `Child Update Response`. This however will make it possible for `mRouterTable.GetLeader()` to return `nullptr`, causing a number of assertions to fail. So we also need to handle the `leader == nullptr` case safely everything we get a pointer to Leader. 

Adding the Leader to Router Table for Leader Data could solve this issue but I think it unnecessarily complicates the message processing logics. 

Thoughts? @LuDuda @jwhui

jwhui, 2020-07-27:
@LuDuda , thanks for raising this issue!

I agree with @simonlingoogle that always accepting Leader Data and properly handling the case where the Router Table does not include the current Leader ID is a promising approach. @simonlingoogle , would you like to submit a PR?

simonlingoogle, 2020-07-27:
> @LuDuda , thanks for raising this issue!
> 
> I agree with @simonlingoogle that always accepting Leader Data and properly handling the case where the Router Table does not include the current Leader ID is a promising approach. @simonlingoogle , would you like to submit a PR?

I will submit a PR lately.

simonlingoogle, 2020-07-28:
Submitted PR #5301 for fixing this bug. Please check. @LuDuda

LuDuda, 2020-07-28:
Thanks for fast help. Let me review and test it.

PR #5301 (2020-08-15): [mle] fix FTD unable to restore link to parent
This PR fixes #5293 

Testing if two `OT_ASSERT(leader != nullptr);` statements can be kept since they seems correct:
- OT Build: 
  - **FTD Leader Data dropping removed**
  - **`OT_ASSERT(leader != nullptr);` statements kept**
  - kMinDowngradeNeighbors = 4
  - OPENTHREAD_CONFIG_MLE_LONG_ROUTES_ENABLE = 1
- OTNS Stress Tests
  - [x] [Large Network Forming](https://github.com/openthread/ot-ns/blob/master/pylibs/stress_tests/large_network_forming.py) (simulated 100 days successfully)
  - [x] [MLEID Connectivity test](https://github.com/simonlingoog [... cut]
closed 2020-08-15 by jwhui