openthread/openthread #5300: [mac] hard-fault seen in ProcessReceiveSecurity

LuDuda, 2020-07-27:
On the build from the current master, I got the below hard fault.

Interestingly I am unable to reproduce it anymore. Do you have any idea what could be the issue here? I had just two boards and testing re-synchronization issue reported here #5293.

```
HardFault_Handler () at third_party/NordicSemiconductor/nrfx/mdk/gcc_startup_nrf52840.S:316
316	    b       .
(gdb) where
#0  HardFault_Handler () at third_party/NordicSemiconductor/nrfx/mdk/gcc_startup_nrf52840.S:316
#1  <signal handler called>
#2  0x000071b8 in ot::Mac::Mac::ProcessReceiveSecurity (this=this@entry=0x200069ac <ot::gInstanceRaw+23580>, aFrame=..., aSrcAddr=..., 
    aNeighbor=aNeighbor@entry=0x20006dcc <ot::gInstanceRaw+24636>) at src/core/mac/mac.cpp:1510
#3  0x00007d32 in ot::Mac::Mac::HandleReceivedFrame (this=0x200069ac <ot::gInstanceRaw+23580>, 
    aFrame=aFrame@entry=0x20009370 <sReceivedFrames>, aError=<optimized out>)
    at src/core/mac/mac.cpp:1651
#4  0x00009242 in ot::Mac::SubMac::Callbacks::ReceiveDone (this=<optimized out>, aFrame=aFrame@entry=0x20009370 <sReceivedFrames>, 
    aError=<optimized out>) at src/core/mac/sub_mac_callbacks.cpp:64
#5  0x00009028 in ot::Mac::SubMac::HandleReceiveDone (this=<optimized out>, aFrame=aFrame@entry=0x20009370 <sReceivedFrames>, 
    aError=<optimized out>) at src/core/mac/sub_mac.cpp:210
#6  0x000209ce in ot::Radio::Callbacks::HandleReceiveDone (this=this@entry=0x200015a3 <ot::gInstanceRaw+2067>, 
    aFrame=aFrame@entry=0x20009370 <sReceivedFrames>, aError=<optimized out>)
    at src/core/radio/radio_callbacks.cpp:43
#7  0x0000de40 in otPlatRadioReceiveDone (aInstance=aInstance@entry=0x20000d90 <ot::gInstanceRaw>, 
    aFrame=aFrame@entry=0x20009370 <sReceivedFrames>, aError=aError@entry=OT_ERROR_NONE)
    at src/core/radio/radio_platform.cpp:50
#8  0x00023a98 in nrf5RadioProcess (aInstance=aInstance@entry=0x20000d90 <ot::gInstanceRaw>)
    at examples/platforms/nrf528xx/src/radio.c:728
#9  0x00023e58 in otSysProcessDrivers (aInstance=aInstance@entry=0x20000d90 <ot::gInstanceRaw>)
    at examples/platforms/nrf528xx/src/system.c:143
#10 0x000002f8 in main (argc=0, argv=0x0 <__isr_vector>) at examples/apps/cli/main.c:114

(gdb) info locals
No locals.

(gdb) frame 2
#2  0x000071b8 in ot::Mac::Mac::ProcessReceiveSecurity (this=this@entry=0x200069ac <ot::gInstanceRaw+23580>, aFrame=..., aSrcAddr=..., 
    aNeighbor=aNeighbor@entry=0x20006dcc <ot::gInstanceRaw+24636>) at src/core/mac/mac.cpp:1510
1510	    aesCcm.Init(aFrame.GetHeaderLength(), aFrame.GetPayloadLength(), tagLength, nonce, sizeof(nonce));

(gdb) info locals
keyManager = <optimized out>
error = OT_ERROR_SECURITY
securityLevel = 5 '\005'
keyIdMode = 8 '\b'
frameCounter = 2000
nonce = "\":\234d\200%\216\273\000\000\a\320\005"
tag = "\000\000\000\000@\223\000 \000\000\000\000\000j\000 "
tagLength = 4 '\004'
keyid = 0 '\000'
keySequence = 0
macKey = <optimized out>
extAddress = <optimized out>
aesCcm = {mEcb = {mContext = {buffer = {buffer_cc310 = {537067520, 0, 872611181, 3223651, 986024414, 565478131, 2850422240, 4039041978, 
          128, 536871060, 536908656, 536899020, 0, 0, 0, 1, 0, 0, 1, 1, 4294902280, 537132760, 127, 76}, buffer_vanilla_mbedtls = {
          537067520, 0, 872611181, 3223651, 986024414, 565478131, 2850422240, 4039041978, 128, 536871060, 536908656, 536899020, 0, 0, 
          0, 1, 0, 0, 1, 1, 4294902280, 537132760, 127, 76, 4294902280, 537132776, 127, 536920302, 155245, 2, 155333, 8, 2, 536920302, 
          536911342, 4, 155431, 536920302, 537132748, 0, 536911506, 536920302, 536911342, 536920302, 536871680, 536871680, 0, 
          536908656, 536899020, 0, 16161, 536873485, 536899020, 0, 16161, 1128353069, 757935405, 1377843757, 1768252261, 543450486, 
          1835102822, 1919295589, 1931505007, 1953656680, 1684300064, 1936942450, 880291872, 3223651, 536920302, 4294967261}, 
        dummy = 537067520}, handle = 0x36fe8 <mbedtls_aes_cc310_backend_funcs>}}, 
  mBlock = "\354\235\000 \356\235\000 \377\000\000\000\023\204\002", mCtr = "\000\000\000\064\000\020\000@\000\360\001@\364\235\000 ", 
  mCtrPad = "\354\235\000 \356\235\000 \356\300\000 \223\201\002", mHeaderLength = 536897964, mHeaderCur = 16181, 
  mPlainTextLength = 536871680, mPlainTextCur = 537132912, mBlockLength = 793, mCtrLength = 0, mNonceLength = 241 '\361', 
  mTagLength = 22 '\026'}
(gdb) 

```

jwhui, 2020-07-27:
@LuDuda , thanks for raising this issue.

I could not find anything obvious in my initial review of the code. @abtink , thoughts?

abtink, 2020-07-28:
Nothing specific jumps out to me.

The fault is here:   
```
 aesCcm.Init(aFrame.GetHeaderLength(), aFrame.GetPayloadLength(), tagLength, nonce, sizeof(nonce));
```
One thing to look more into is whether `GetHeaderLength()` will return correct value (under all situations) - since the implementation uses `GetPayload()` which can return `nullptr` if frame cannot be parsed.
However, I know before we get to do security check on the received frame it must passes through `ValidatePsdu()`. 
I will check if there may be any situation that is not covered by `ValidatePsdu().

jwhui, 2020-11-12:
@LuDuda , have you run into this issue recently?

LuDuda, 2020-11-13:
@jwhui no. From my previous experience, such asserts popped out again right after I had closed GitHub issue, so give it a try 😄

closed 2020-11-13 by LuDuda