openthread/openthread #5303: [radio] Should ack be disabled when thread stops?

Irving-cl, 2020-07-28:
**Describe the bug** 
In current implementation, even after `thread stop` is called, the device would still be able to send acks for received frames because `Radio` is not actually disabled. This leads to some problems in some test cases when we try to simulate a real absence of a node.

**To Reproduce**
As an example, we can run a simple case:
```
#!/usr/bin/env python3

import unittest

import thread_cert

LEADER = 1
SED_1 = 2

class SED_Reattach(thread_cert.TestCase):
    TOPOLOGY = {
        LEADER: {
            'version': '1.2',
        },
        SED_1: {
            'version': '1.2',
            'mode': 's',
        },
    }
    """All nodes are created with default configurations"""

    def test(self):

        self.nodes[LEADER].start()
        self.simulator.go(5)
        self.assertEqual(self.nodes[LEADER].get_state(), 'leader')

        self.nodes[SED_1].start()
        self.nodes[SED_1].set_pollperiod(1000)
        self.simulator.go(7)
        self.assertEqual(self.nodes[SED_1].get_state(), 'child')

        self.simulator.go(5)
        self.nodes[LEADER].stop()

        self.simulator.go(100)

if __name__ == '__main__':
    unittest.main()

```
We can find that in the pcap that, even after `LEADER` stops, the data requests sent by `SED` still have acks.

**Expected behavior** 
Shall we also disable the radio so that the device won't have any acks when thread stops?

@abtink @jwhui @bukepo @librasungirl @simonlingoogle

jwhui, 2020-07-28:
@Irving-cl , OpenThread makes a distinction between the interface being up and Thread protocols (e.g. MLE and TMF) being enabled. When the interface is up, the 802.15.4 radio is still functional and should be able to send ACKs, even if Thread is disabled.

However, I believe that `self.nodes[LEADER].stop()` should also bring both Thread and interface down. In this case, the Leader should not be sending ACKs.

Looking at the code, `ThreadNetif::Down()` calls `Get<Mac::Mac>().SetRxOnWhenIdle(false)`, which should put the radio to sleep. Will require some more investigation to see why ACKs are still being sent.

simonlingoogle, 2020-07-28:
Maybe it's related to this comment: https://github.com/openthread/openthread/pull/4903#issuecomment-620755900 ?

jwhui, 2020-07-28:
FWIW - I just did a simple test and the `ifconfig down` command does indeed put the simulated radio into the sleep state.

Irving-cl, 2020-07-28:
Thanks! It turned out to be a problem in my code.

closed 2020-07-28 by Irving-cl