openthread/openthread #5311: Can not discover Thread Network on NRF52840DK

oubotong, 2020-07-30:
**Describe the bug** :
After setting up Rpi3 as Border Router with NRF52840-dongle as RCP, I used the `ot-ctl` to create a Thread network. However, after flashing a simple `ot-cli-ftd.hex` image to another NRF52840DK board and type `scan` in the shell, no network is discovered.
**To Reproduce**:

**For the raspberry pi 3 with Stretch Lite OS**:
```
git clone https://github.com/openthread/ot-br-posix.git
cd ot-br-posix
./script/bootstrap
./script/setup
```
Then modify the `/etc/default/otbr-agent`:
```
OTBR_AGENT_OPTS="-I wpan0 spinel+hdlc+uart:///dev/ttyACM0" #I checked that this is the port for my dongle
```
**For the dongle**:
```
git clone https://github.com/openthread/openthread.git
cd openthread
./script/bootstrap
./bootstrap
make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 TMF_PROXY=1 DHCP6_SERVER=1 DHCP6_CLIENT=1 DISABLE_SPI=1 SNTP_CLIENT=1 DISABLE_BUILTIN_MBEDTLS=1 USB=1 BOOTLOADER=USB
cd output/nrf52840/bin
arm-none-eabi-objcopy -O ihex ot-rcp ot-rcp.hex
nrfutil pkg generate --hw-version 52 --sd-req 0x0 --application-version 1 --application ot-rcp.hex out.zip
sudo nrfutil dfu usb-serial -pkg out.zip -p /dev/ttyACM0 -b 115200
```
Then reboot the raspberry pi 3 and I can see there are three services running:
```
avahi-daemon.service
otbr-agent.service
otbr-web.service
```

**For the other NRF52840DK board**:
```
cd openthread
make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 
cd output/nrf52840/bin
arm-none-eabi-objcopy -O ihex ot-cli-ftd ot-cli-ftd.hex
sudo nrfjprog -f nrf52 -s 683704924(This_is_my_serial_number) --chiperase --program t-cli-ftd.hex --reset
```
Then connect the NRF52840DK to the raspberry pi 3. At this time, both of the NRF52840-dongle and NRF52840DK are connected to the raspberry pi and dongle is at `/dev/ttyACM0` and NRF52840DK is at `/dev/ttyACM1`

**Time to create the network and discover it**:
On the Raspberry Pi:
```
sudo ot-ctl dataset init new
sudo ot-ctl dataset 
**
Active Timestamp: 1
Channel: 21
Channel Mask: 07fff800
Ext PAN ID: 4426336fa9d052ce
Mesh Local Prefix: fd0c:d895:c65c:3c26/64
Master Key: 7a2ee40141b9f0fadfb45b510be99bf2
Network Name: OpenThread-e42f
PAN ID: 0xe42f
PSKc: 98c43de72c16fd8fa9943298b8f8b1d2
Security Policy: 0, onrcb
Done
**
sudo ot-ctl dataset commit active
sudo ot-ctl ifconfig up
sudo ot-ctl thread start
```
Then we need to get the shell of NRF52840DK:
```
screen /dev/ttyACM1 115200
```
After the shell pop(by pressing ENTER), we scan:
```
(shell) scan
**
Nothing was discovered
**
```

**Expected behavior** 
The NRF52840DK should discover the thread network created on the Raspberry Pi 3 and then join it.

oubotong, 2020-08-01:
Any help? Still can't scan out the Thread network

oubotong, 2020-08-03:
The only thing I can discover on the NRF52840DK is this:
```
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm |
+---+------------------+------------------+------+------------------+----+-----+
| 0 |                  | 0000000000000000 | f815 | 00d02dfffe0250e6 | 2|  32 | 
```

jwhui, 2020-08-04:
@oubotong , I just tried this with latest master on openthread/openthread (3cd5f49) and openthread/ot-br-posix ([ecc2570](https://github.com/openthread/ot-br-posix/commit/ecc25701117fc99c6630584dc98294527a145f7d)) and everything worked as expected. I was able to scan for the newly created network:
```
> scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 | OpenThread-89bc  | f11620d326f0c247 | 89bc | daba25a6e7c8eace | 14 | -36 | 228 |
Done
```

What git commit IDs are you using?

oubotong, 2020-08-04:
@jwhui, Thanks for your check. I have made it work by replacing the NRF52840-dongle and NRF52840DK with new ones. I guess OpenThread needs to provide more specific ways to verify that all the required components have the correct functionalities.  
Anyway, I have learned a lot when fixing the problem :)

jwhui, 2020-08-04:
> @jwhui, Thanks for your check. I have made it work by replacing the NRF52840-dongle and NRF52840DK with new ones. I guess OpenThread needs to provide more specific ways to verify that all the required components have the correct functionalities.

@jwhui , my test was with two nRF52840 USB Dongles, which should have all of the capabilities required for OpenThread. Glad you finally have a working setup.

closed 2020-08-11 by jwhui