openthread/openthread #5336: Overflow leads to invalid error

d1ee, 2020-08-04:
https://github.com/openthread/openthread/blob/481f69703b6478d461590ac6b036c7c7d5b6a6b8/src/core/coap/coap.cpp#L974

For number `mMaxRetransmit >= 22` and for any number `mAckTimeout >= 1000`, the multiplication leads to overflow and returns a `false` value for this method.

It wasn't trivial to find out why `IsValid()` method was returning `false` value. Maybe it would help to increase the `tmp`'s size to `uint64_t`.

jwhui, 2020-08-04:
RFC 7252 [Section 4.8.2](https://tools.ietf.org/html/rfc7252#section-4.8.2) defines `MAX_TRANSMIT_SPAN` as follows:

```
      MAX_TRANSMIT_SPAN is the maximum time from the first transmission
      of a Confirmable message to its last retransmission.  For the
      default transmission parameters, the value is (2+4+8+16)*1.5 = 45
      seconds, or more generally:

         ACK_TIMEOUT * ((2 ** MAX_RETRANSMIT) - 1) * ACK_RANDOM_FACTOR
```

With:

- ACK_TIMEOUT = 1000
- MAX_RETRANSMIT = 22
- ACK_RANDOM_FACTOR = 1.5

Results in:

- 1000 * ((2 ** 22) - 1) * 1.5 = 6291454500.0 milliseconds

Or about ~72 years. Do you really have a use case that requires such large values?

d1ee, 2020-08-04:
@jwhui Actually no, the MAX_RETRANSMIT was allowed up to 30 in the code so I tried out 

- ACK_TIMEOUT = 2000
- MAX_RETRANSMIT = 30
- ACK_RANDOM_FACTOR = 1

but I was surprised that lead to an error, which looked ok for me. And this was harder to find because the result of a computation.

I wouldn't say it's a problem or a bug, but it's misleading because I had no idea why MAX_RETRANSMIT=30 didn't work.

jwhui, 2020-08-05:
@d1ee , I just submitted #5338 as a proposed fix. Please review.

d1ee, 2020-08-05:
@jwhui Thanks :+1:

PR #5338 (2020-08-06): [coap] reduce OT_COAP_MAX_RETRANSMIT to 20
Also add a note to clarify the set of valid CoAP transmission
parameters.

Resolves #5336
closed 2020-08-05 by d1ee