openthread/openthread #5347: Question - OpenThread - Sending "multicast" UDP messages and setting retransmits

farhangj, 2020-08-06:
Hello there,

I am using openthread CLI example on Nordic nRF52840 devkits to demo mesh networking.
Using the openthread/udp lib to send periodic multicast messages ( every 5s from one of the devices), and sending ACKs back from the devices that receive messages straight back to the **sender** of the original message, by storing the "peer address" of the sender which i get from the original message and sending the ACK message back to that address.

I have reduced the gain and spread my devices out so that I can see the "Mesh" actually working - meaning messages should propagate by hopping to adjacent nodes. They're all FTD.
I am noticing a 20% message loss in some nodes farther away from the sender but within reach of other FTD nodes. I am suspecting the retransmits/hops aren't enabled or not set right.

**Questions:**
Does the underlying openthread link layer has hops/retransmits built in? 
If so, how do i control the number of retransmits (upon reception of a multicast message) of each node ? There are some settings around "Direct" vs "indirect" msg retransmits. I am not using indirect (polling). Direct retransmit is set to 3, does that mean each node that receives a message will in turn retransmit the message 3x as well? or is this only applied to the original sender of a message and not to other nodes?
If not, does going to upper layers such as Coap provide the true meshing and retransmit features?

jwhui, 2020-08-06:
FTDs always support mesh routing as that is required by the Thread protocol specification.

OpenThread currently implements 3 MAC retransmissions, as specified in Thread 1.x. However, in practice, we have found a value of 15 to be more appropriate. You can configure this value with the [OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT](https://github.com/openthread/openthread/blob/master/src/core/config/mac.h) build configuration.

If you require end-to-end reliability, then transport/application-layer reliability mechanisms are also strongly encouraged.

Hope that helps.

farhangj, 2020-08-06:
Thanks Jonathan for the quick reply.
OK, does 15 retries apply to the original message sender MAC only or to the all the nodes that receive the message too?
e.g. say i have 3 nodes, A, B and C. A only sees B. C only sees B. And B sees both A and C.

Questions:
If A sends a message, it will retry sending it R times (determined by the OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT), is that correct? Or does it get an ACK at the MAC layer from other devices and doesn't retry that many times?
Does B also retry sending the message **R times** upon receiving it?
Cranking R to 15, even with a handful of nodes (eg 4) could lead to 60 msg retries, do you think RF transmit over-running each other could happen? or are there systems in place in openthread to avoid that?

jwhui, 2020-08-06:
> OK, does 15 retries apply to the original message sender MAC only or to the all the nodes that receive the message too?
> e.g. say i have 3 nodes, A, B and C. A only sees B. C only sees B. And B sees both A and C.

Since the retry value is at the MAC layer, only the nodes that have implemented 15 retries will retransmit 15 retries.

> Questions:
> If A sends a message, it will retry sending it R times (determined by the OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT), is that correct? Or does it get an ACK at the MAC layer from other devices and doesn't retry that many times?

A will retransmit until it receives an ACK from the intended receiver or it has already tried `OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT` times.

> Does B also retry sending the message **R times** upon receiving it?

B will also retry sending the message `OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT` times, whatever that is set to on B.

> Cranking R to 15, even with a handful of nodes (eg 4) could lead to 60 msg retries, do you think RF transmit over-running each other could happen? or are there systems in place in openthread to avoid that?

In a very good RF environment (i.e. with no message loss), you will only see 1 transmission per device. In a very poor RF environment, you could certainly see up to 15 retries per device. When you have a poor RF environment you basically have to choose between retransmitting many times or not have connectivity.

Hope that helps.

farhangj, 2020-08-06:
Very helpful, thanks Jonathan.

In the scenario of a multicast UDP message, do you know what the condition(s) is/are for retries to stop? Is it an ACK from adjacent node? i.e. in the example above does node "A" stop as soon as it hears and ACK from B, relying on B to propagate the message to C? Or does it retry up to OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT  to hear from C also?

side note: looks like with otLinkGetTxDirectRetrySuccessHistogram() i can get insight into how well the retransmits are working.

jwhui, 2020-08-06:
> In the scenario of a multicast UDP message, do you know what the condition(s) is/are for retries to stop? Is it an ACK from adjacent node? i.e. in the example above does node "A" stop as soon as it hears an ACK from B, relying on B to propagate the message to C? Or does it retry up to OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT to hear from C also?

IPv6 multicast translates to IEEE 802.15.4 broadcast. In the broadcast case, ACKs are not sent in response. Current default is a single transmission per broadcast message. However, you can experiment with larger values using [OPENTHREAD_CONFIG_MAC_TX_NUM_BCAST](https://github.com/openthread/openthread/blob/e810357adb1f31e7e8fb55042317c451e87535a6/src/core/config/mac.h#L172).

> side note: looks like with otLinkGetTxDirectRetrySuccessHistogram() i can get insight into how well the retransmits are working.

Yes.

farhangj, 2020-08-06:
> IPv6 multicast translates to IEEE 802.15.4 broadcast. In the broadcast case, ACKs are not sent in response.

Does this mean that a IEEE 802.15.4 broadcast is not taking advantage of the mesh feature? i.e. messages won't propagate through the network?

Is there a better alternative layer to use for what I am trying to do that includes ACKing to ensure messages are received? All i need to achieve is sending messages from any node to all others. Any node can send the message, and they're all FTD. and i need the Mesh feature to work (hence why I am using OpenThread).

jwhui, 2020-08-06:
> Does this mean that a IEEE 802.15.4 broadcast is not taking advantage of the mesh feature? i.e. messages won't propagate through the network?

Link-local IPv6 multicast translates to a single IEEE 802.15.4 broadcast.

Non-link-local scope IPv6 multicast (e.g. realm-local or larger scope) is forwarded using [RFC 7731](https://tools.ietf.org/html/rfc7731), which operates at the network layer, not link layer. MPL adds its own redundancy by broadcasting each message 2-3 times on each router. You can try using larger values by changing [kMplRouterDataMessageTimerExpirations](https://github.com/openthread/openthread/blob/e810357adb1f31e7e8fb55042317c451e87535a6/src/core/thread/mle_types.hpp#L193).

> Is there a better alternative layer to use for what I am trying to do that includes ACKing to ensure messages are received? All i need to achieve is sending messages from any node to all others. Any [...]

Reliable multicast is a difficult networking problem that is not specific to Thread. I realize now from your original message you are mostly concerned about multicast messaging. The primary challenge is dealing with the potential flood of ACKs coming back.

farhangj, 2020-08-06:
> Link-local IPv6 multicast translates to a single IEEE 802.15.4 broadcast.

Jonathan, thanks for all the details.

If UDP multicast messages are translated to a single IEEE 802.15.4 broadcast, then I am not really taking advantage of the "mesh" features of OpenThread.

Do you have any recommendation on what's a better approach or layer to use? Or is it that I have to develop application on top of Thread to ensure message propagation throughout the network, regardless of what layer i am calling in my application? Currently i'm calling UDP API, but i can go up to CoAP if that helps?
is there no readily-available multicast protocol in OpenThread that I can use? Multicast seems such a generic use-case, I am surprised if it doesn't support multicast with propagation (retransmit via all FTDs) feature built in.

jwhui, 2020-08-06:
> If UDP multicast messages are translated to a single IEEE 802.15.4 broadcast, then I am not really taking advantage of the "mesh" features of OpenThread.
>
> Do you have any recommendation on what's a [...]

As I tried to convey in https://github.com/openthread/openthread/issues/5347#issuecomment-670196678, realm-local and higher scope multicast in Thread uses MPL for forwarding over multiple hops. Thread's implementation of MPL works by having routers broadcast each newly-received message multiple times. It is a "flooding" protocol, but over a dynamically set of elected routers.

You also asked about reliable multicast delivery. MPL is best effort in that there are no acknowledgments. Reliable multicast delivery is a difficult problem because it typically involves bidirectional communication between the source and each group member.

farhangj, 2020-08-06:
I guess i don't fully understand the difference between Link-Local IPv6 and non-link-local.

There's no connectivity to the outside world out of my OpenThread network (e.g. no access from the internet is required) - 1 single PANid.
I am opening a UDP socket on all devices (every node is FTD), and i open a specific port number on all.
Then from the sender device I send a message to IPv6 address "FF03::1" which defines multicast to every node from what I understand.
Is that considered Local-Link or non-local-link?

Address is populated via : otIp6AddressFromString("FF03::1", &messageInfo.mPeerAddr);
Message is created using : otUdpNewMessage(thread_ot_instance_get(), NULL);
and finally sent to the socket via : otUdpSend(&socket, test_Message, &messageInfo);

all radio modes are THREAD_RADIO_MODE_RX_ON_WHEN_IDLE
auto-commissioning and auto-start are both on.

jwhui, 2020-08-06:
> I guess i don't fully understand the difference between Link-Local IPv6 and non-link-local.

Thread devices link-local scope as devices that are reachable with a single radio transmission. If you want to communicate over multiple hops, then you need to use realm-local or larger scope. See [Thread Primer - IPv6 Addressing](https://openthread.io/guides/thread-primer/ipv6-addressing) for more information.

farhangj, 2020-08-17:
Sorry I was away for a week. Picking this back up again..
I read the primer - In that document there's no mention of "realm-local" or "larger scope".
I assume those refer to what the Thread Primer calls "Mesh Local" and its IPV6 address is "ff03::1" which I was using in my code.
Therefore I assume the Thread network was already using MPL retransmits that you mentioned. Set to 2 by default. I have roughly 80% success rate on multicast messages, nodes closer to the initiator of the multicast message had higher success rates.
I will increase that to 4 and see if my lossy network gets better success rate across the system. In theory it should.

Is there any way that I can observe or get metrics that show MPL retransmits are working? e.g. Via the CLI on each node?  I have intentionally reduced the RF output power to mimic a lossy network.

jwhui, 2020-08-18:
Unfortunately, the IPv6 MPL module does not currently expose any of its state via APIs or CLIs.

OpenThread does have some [MAC-level counters](https://github.com/openthread/openthread/blob/master/src/cli/README.md#counters) that display the number of broadcast messages sent/received, but that will include any MAC frame that is broadcast, not just MPL message.

You could also enable [logging](https://github.com/openthread/openthread/blob/962c06f8e778a6a55358154b5365bcf5611d230d/src/core/thread/mesh_forwarder.cpp#L1434) to observe what messages are sent/received.

If you wanted to just extract the MPL information, you could look at adding log messages in [src/core/net/ip6_mpl.cpp](https://github.com/openthread/openthread/blob/master/src/core/net/ip6_mpl.cpp). Interesting areas to log would be:

1. When a new IPv6 multicast message is received ([here](https://github.com/openthread/openthread/blob/962c06f8e778a6a55358154b5365bcf5611d230d/src/core/net/ip6_mpl.cpp#L102))

1. When a buffered IPv6 multicast message is retransmitted ([here](https://github.com/openthread/openthread/blob/962c06f8e778a6a55358154b5365bcf5611d230d/src/core/net/ip6_mpl.cpp#L383) and [here](https://github.com/openthread/openthread/blob/962c06f8e778a6a55358154b5365bcf5611d230d/src/core/net/ip6_mpl.cpp#L406))

Hope that helps.

closed 2020-09-01 by jwhui