openthread/openthread #5410: [docker] codelab documentation problem

Irving-cl, 2020-08-17:
I'm trying to use docker to run openthread simulation. I found that in our codelab [Simulating a Thread network using OpenThread in Docker](https://codelabs.developers.google.com/codelabs/openthread-simulation/#1), in step 2 - **Set up Docker**, the command to run docker is:
```
$ docker run --name codelab_otsim_ctnr -it --rm \
   --sysctl net.ipv6.conf.all.disable_ipv6=0 \
   --cap-add=net_admin openthread/codelab_otsim_test bash
```
But the docker's name should be `codelab_otsim`, instead of `codelab_otsim_test`, since the previous installing command is:
```
docker pull openthread/codelab_otsim:latest
```
If I run with `codelab_otsim_test`, I got an error of `Unable to find image 'openthread/codelab_otsim_test:latest' locally`.

jwhui, 2020-08-17:
@Irving-cl , on [this page](https://codelabs.developers.google.com/codelabs/openthread-simulation/#2), under "Create the newtork" -> "1. Start Node 1", I see:
```
$ docker run --name codelab_otsim_ctnr -it --rm \
   --sysctl net.ipv6.conf.all.disable_ipv6=0 \
   --cap-add=net_admin openthread/codelab_otsim bash
```
I could not find any mention of `codelab_otsim_test`. Can you clarify where you see that?

Irving-cl, 2020-08-17:
It's on the previous page, **2. Set up Docker - Pull the Docker image**:
[image]

jwhui, 2020-08-17:
@Irving-cl , yes I see it now. Assigning to @Vyrastas .

Vyrastas, 2020-08-17:
Fixed!

closed 2020-08-17 by Vyrastas