openthread/openthread #5414: [message] potential issue with checksum calculation

abtink, 2020-08-17:
I think we may have a potential issue with checksum calculation.

The checksum uses `uint16` values as it updates the value (padding zero at the end if total number of bytes is odd).
We have different flavors of `Message::UpdateChecksum()`

1. A static method `UpdatChecksum(value) adding `uint16_t` value.
1. A static method `UpdateChecksum(void *aBuffer, uint16_t aLength)` adding an array of bytes (given as buffer and length)
1. A `Message` method `UpdateChecksum(uint16_t offset, uint16_t aLength)` adding content of `message` from given offset up to specified length. 

The problem is with the third method. 
- The `Message` content is internally stored in chunks of data in a sequence of linked-listed allocated `Buffer` instances. 
  - The first `Buffer` stores the `Metada` as well, so it can store smaller chunk of the message data.
- The `UpdateCheckum()` implementation goes through all data chunks from different `Buffer` instances and updates the checksum for each data chunk (calling second flavor of `UpdateChecksum` on each data chunk).
- The issue arises if any of the data chunks happen to have an odd length, then when we add the next chunk its first byte will be treated as new `uint16_t` rather than MSB of the previous `uint16_t`.

I don't think this is critical. I think it all current place where we calculate checksum the offsets are  even and also the metda data length (which impact the data in first `Buffer`) is also even.

jwhui, 2020-08-19:
@abtink , thanks for flagging this issue. As you mentioned, the current implementation assumes that each check has even offset and even length. Maybe it would be sufficient to add some static checks in rather than updating the implementation to support odd offset/length?

abtink, 2020-08-19:
@jwhui , I have a refactoring and simplification of all checksum related method in the works (adding a `Checksum` class/module that provides all related functionality). It would also address this.

The challenge I see is if something is added to message metadata to make it odd sized, we may face an issue (the first chunk can be of odd length).  Also buffer size itself can be customized by project config (and can be odd value), so I think it may be better be safe in all cases.

jwhui, 2020-08-19:
> The challenge I see is if something is added to message metadata to make it odd sized, we may face an issue (the first chunk can be of odd length). Also buffer size itself can be customized by project config (and can be odd value), so I think it may be better to be safe in all cases.

I think the offset would generally be OK - the structures are not declared as packed, so I assume there would be padding due to word alignment.

As for buffer size itself, I think we could have a compile-time check to ensure it is not odd. That way we do not incur and code overhead for dealing with the odd size case.

All that said, if you have a way to address the odd offset/size case without little to no overhead, I'm fine with that as well.

jwhui, 2020-09-01:
Resolved by #5434.

closed 2020-09-01 by jwhui