openthread/openthread #5447: [tests] unexpected fails

librasungirl, 2020-08-24:
**Describe the bug** 
Some tests fail even if the PR doesn't have any logical changes.

Besides manually rerun, are there anything we can do to enhance the stability of tests?

The typo fixes in #5446  should not cause any fails, however below fails are observed:

- [toranj-ncp: exit status 11](https://github.com/openthread/openthread/pull/5446/checks?check_run_id=1019610573#step:4:611)

- [packet-verification Cert_5_5_02_LeaderReboot](https://github.com/openthread/openthread/pull/5446/checks?check_run_id=1019610815#step:5:812) 

- [cli-ftd test_router_reattach](https://github.com/openthread/openthread/pull/5446/checks?check_run_id=1019610603#step:5:446)

**Expected behavior** 
Consistently pass if no bugs newly introduced in the PR.

librasungirl, 2020-08-24:
@jwhui @abtink @simonlingoogle @FiveDimensions

simonlingoogle, 2020-08-24:
Another packet-verification 5.5.2 fail: https://github.com/openthread/openthread/pull/5425/checks?check_run_id=1021503850

Could you take a look? @FiveDimensions

simonlingoogle, 2020-08-24:
```
Traceback (most recent call last):
------------------------------------------------------------------------------------------------------------------------
  File "test-001-get-set.py", line 47, in <module>
Starting 'test-001-get-set'
    wpan.Node.init_all_nodes()
Took too long to init node Node (index=1, interface_name=wpan1) (15.0565950871>15 sec)
```
Should toranj increase the timeout? @abtink

jwhui, 2020-08-26:
@simonlingoogle @FiveDimensions @abtink , I am experiencing an increase in transient failures. Any thoughts on what the issue might be?

abtink, 2020-08-26:
@librasungirl , @simonlingoogle , regarding the `toranj` failure, we have seen this issue and have spent a lot of time on trying figure it out in past (when we were using travis CI). Based on the info this seems to be due to an issue (bad install) in dbus library itself (we have seen similar situation on actual devices as well). Increasing test timeout or increasing number test attempts won't impact and help in this case. A re-run (which re-builds and re-installs everything) seems to be the main option here.  

I agree that we want to minimize the CI unreliable failures, but based on my observations the `tornaj` workflow is not failing that often compared to other transient failures.  For reference, the CI actions on my fork of OT, for the past ~20 branches/commits (~last week), I had to re-run `simulation` workflow about ~7-8 times, `build` workflow ~2-3 times but do not recall any re-kick of `toranj`. It may be good if we can collect info/stat about failures (on main fork, or may be on developer's fork (where we can have more control on tracking failures and re-kicks of CI workflows)).

simonlingoogle, 2020-09-01:
I have submitted #5479 so that it is possible to investigate these issues using the failed log and PCAP.
Please take a look when you have time.

FiveDimensions, 2020-09-05:
#5479 is a great idea to archive the test artifacts. Checked the case 5.5.2 test logs, failed in Step 9 Leader reattach and MLE Parent Request packet can not be found caused by packets order, enhanced this case in #5490 and rerun more than 5 times and no related failure found. Will keep on rerunning > 10 times then open to review

PR #5517 (2020-09-11): [scripts] fix test_router_reattach
Fix `test_router_attach.py` failing by a small chance. 

**Root cause: `routerdowngradethreshold` restored back to default (23) after node reset, so the test had 5/120 chance to downgrade. We should set it back to 32 before starting it.**

This PR also enhance the test for a little bit:
- Make sure the node reattached as Router within 1 second
- Make sure the node does not downgrade after Router Selection Jitter (3s)

Fixes #5447
closed 2020-09-08 by jwhui